nugget/python-insteonplm #3: Add Support for 6-Button Keypad Dimmer (2334-292)

larizzo, 2017-03-01:
Was wondering if you are looking at or at willing to add support for 6-Button Keypad Dimmer (2334-292). Let me know what I could do to help. I'm more then happy to provide logs/debugging information.

I linked one button from my Keypad with my USB PLM and when I press it I see this in the homeassistant.log:
```
17-02-28 20:15:46 INFO (MainThread) [insteonplm.protocol] Processing message: b'02503d8ff8000003cf1300'
17-02-28 20:15:46 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback <bound method PLM._parse_insteon_standard of <insteonplm.protocol.PLM object at 0x2b73b84449b0>>
Traceback (most recent call last):
  File "uvloop/cbhandles.pyx", line 49, in uvloop.loop.Handle._run (uvloop/loop.c:48380)
  File "/usr/local/lib/python3.5/site-packages/insteonplm/protocol.py", line 492, in _parse_insteon_standard
    device = self.devices[msg.address.hex]
  File "/usr/local/lib/python3.5/site-packages/insteonplm/protocol.py", line 40, in __getitem__
    raise KeyError
KeyError
```

nugget, 2017-03-01:
Definitely, I'll take a look and see if there's any logs you can provide that would help.

larizzo, 2017-03-01:
I have it setup through openhab currently so I can grab logs from it also if that helps.

larizzo, 2017-03-01:
It seems to require sending a group ID from what I can tell from openhab code. Here's an example of my config. I'm not sure if the keypadbuttonA part is necessary or not. Also when I was trying to find the groups I just did trial and error till what I was trying to trigger worked...
```
{insteonplm="3D.8F.F8:F00.00.15#keypadbuttonA,group=7"}
```
Here's a debug log for openhab when I pressed the keypadbuttonA:
```
2017-02-28 22:07:06 DEBUG o.o.b.i.internal.driver.Port[:382]- writing (500): OUT:Cmd:0x62|toAddress:3D.8F.F8|messageFlags:0x0F=DIRECT:3:3|command1:0x19|command2:0x01|
2017-02-28 22:07:07 DEBUG o.o.b.i.internal.driver.Port[:382]- writing (500): OUT:Cmd:0x62|toAddress:3D.8F.F8|messageFlags:0x0F=DIRECT:3:3|command1:0x19|command2:0x00|
2017-02-28 22:07:07 DEBUG o.o.b.i.InsteonPLMActiveBinding[:581]- got msg: IN:Cmd:0x50|fromAddress:3D.8F.F8|toAddress:34.1F.4A|messageFlags:0x2B=ACK_OF_DIRECT:3:2|command1:0x0C|command2:0xE4|
2017-02-28 22:07:07 DEBUG o.o.b.i.i.device.DeviceFeature[:264]- 3D.8F.F8:GenericLastTime publishing: 2017-02-28T22:07:07
2017-02-28 22:07:09 DEBUG o.o.b.i.InsteonPLMActiveBinding[:581]- got msg: IN:Cmd:0x50|fromAddress:3D.8F.F8|toAddress:34.1F.4A|messageFlags:0x2B=ACK_OF_DIRECT:3:2|command1:0x0C|command2:0x00|
2017-02-28 22:07:09 DEBUG o.o.b.i.i.device.DeviceFeature[:264]- 3D.8F.F8:GenericLastTime publishing: 2017-02-28T22:07:09
2017-02-28 22:07:15 DEBUG o.o.b.i.i.device.InsteonDevice[:379]- qe taken off direct: KeyPadButtonGroup(0:0:0) OUT:Cmd:0x62|toAddress:3D.8F.F8|messageFlags:0x0F=DIRECT:3:3|command1:0x19|command2:0x01|
2017-02-28 22:07:17 DEBUG o.o.b.i.i.device.InsteonDevice[:370]- still waiting for query reply from 3D.8F.F8 for another 4500 usec
2017-02-28 22:07:19 DEBUG o.o.b.i.i.device.InsteonDevice[:370]- still waiting for query reply from 3D.8F.F8 for another 2500 usec
2017-02-28 22:07:21 DEBUG o.o.b.i.i.device.InsteonDevice[:370]- still waiting for query reply from 3D.8F.F8 for another 500 usec
2017-02-28 22:07:22 DEBUG o.o.b.i.internal.driver.Port[:382]- writing (500): OUT:Cmd:0x62|toAddress:3D.8F.F8|messageFlags:0x0F=DIRECT:3:3|command1:0x19|command2:0x01|
2017-02-28 22:07:23 DEBUG o.o.b.i.i.device.InsteonDevice[:374]- gave up waiting for query reply from device 3D.8F.F8
2017-02-28 22:07:23 DEBUG o.o.b.i.i.device.InsteonDevice[:379]- qe taken off direct: LoadSwitchButton(1:1:6) OUT:Cmd:0x62|toAddress:3D.8F.F8|messageFlags:0x0F=DIRECT:3:3|command1:0x19|command2:0x00|
2017-02-28 22:07:23 DEBUG o.o.b.i.i.d.RequestQueueManager[:132]- device queue for 3D.8F.F8 is empty!
2017-02-28 22:07:23 DEBUG o.o.b.i.internal.driver.Port[:382]- writing (500): OUT:Cmd:0x62|toAddress:3D.8F.F8|messageFlags:0x0F=DIRECT:3:3|command1:0x19|command2:0x00|
2017-02-28 22:07:24 DEBUG o.o.b.i.InsteonPLMActiveBinding[:581]- got msg: IN:Cmd:0x50|fromAddress:3D.8F.F8|toAddress:34.1F.4A|messageFlags:0x2B=ACK_OF_DIRECT:3:2|command1:0x0C|command2:0xE4|
2017-02-28 22:07:24 DEBUG o.o.b.i.i.d.MessageDispatcher[:134]- 3D.8F.F8:LoadSwitchButton DIRECT_ACK: q:QUERY_PENDING cmd: 80
2017-02-28 22:07:24 DEBUG o.o.b.i.i.d.MessageDispatcher[:139]- changing key to 0x19 for msg IN:Cmd:0x50|fromAddress:3D.8F.F8|toAddress:34.1F.4A|messageFlags:0x2B=ACK_OF_DIRECT:3:2|command1:0x0C|command2:0xE4|
2017-02-28 22:07:24 INFO  o.o.b.i.i.d.MessageHandler[:468]- SwitchRequestReplyHandler: dev 3D.8F.F8 button 1 switched to OFF
2017-02-28 22:07:24 DEBUG o.o.b.i.i.device.DeviceFeature[:264]- 3D.8F.F8:LoadSwitchButton publishing: OFF
2017-02-28 22:07:24 DEBUG o.o.b.i.i.d.MessageDispatcher[:159]- defdisp: 3D.8F.F8:LoadSwitchButton set status to: QUERY_ANSWERED
2017-02-28 22:07:24 DEBUG o.o.b.i.i.device.DeviceFeature[:264]- 3D.8F.F8:GenericLastTime publishing: 2017-02-28T22:07:24
2017-02-28 22:07:25 DEBUG o.o.b.i.InsteonPLMActiveBinding[:581]- got msg: IN:Cmd:0x50|fromAddress:3D.8F.F8|toAddress:34.1F.4A|messageFlags:0x2B=ACK_OF_DIRECT:3:2|command1:0x0C|command2:0x00|
2017-02-28 22:07:25 DEBUG o.o.b.i.i.device.DeviceFeature[:264]- 3D.8F.F8:GenericLastTime publishing: 2017-02-28T22:07:25
2017-02-28 22:07:38 DEBUG o.o.b.i.internal.driver.Port[:382]- writing (500): OUT:Cmd:0x62|toAddress:3D.8F.F8|messageFlags:0x0F=DIRECT:3:3|command1:0x19|command2:0x01|
2017-02-28 22:07:39 DEBUG o.o.b.i.internal.driver.Port[:382]- writing (500): OUT:Cmd:0x62|toAddress:3D.8F.F8|messageFlags:0x0F=DIRECT:3:3|command1:0x19|command2:0x00|
2017-02-28 22:07:40 DEBUG o.o.b.i.InsteonPLMActiveBinding[:581]- got msg: IN:Cmd:0x50|fromAddress:3D.8F.F8|toAddress:34.1F.4A|messageFlags:0x2B=ACK_OF_DIRECT:3:2|command1:0x0C|command2:0xE4|
2017-02-28 22:07:40 DEBUG o.o.b.i.i.device.DeviceFeature[:264]- 3D.8F.F8:GenericLastTime publishing: 2017-02-28T22:07:40
2017-02-28 22:07:41 DEBUG o.o.b.i.InsteonPLMActiveBinding[:581]- got msg: IN:Cmd:0x50|fromAddress:3D.8F.F8|toAddress:34.1F.4A|messageFlags:0x2B=ACK_OF_DIRECT:3:2|command1:0x0C|command2:0x00|
2017-02-28 22:07:41 DEBUG o.o.b.i.i.device.DeviceFeature[:264]- 3D.8F.F8:GenericLastTime publishing: 2017-02-28T22:07:41
```
This is from triggering the switch from openhab:
```
2017-02-28 22:11:59 INFO  o.o.b.i.i.d.CommandHandler[:161]- LightOnOffCommandHandler: sent msg to switch 3D.8F.F8 to on
2017-02-28 22:11:59 INFO  o.o.b.i.i.d.CommandHandler[:180]- Sending message to 3D.8F.F8
2017-02-28 22:11:59 DEBUG o.o.b.i.i.d.RequestQueueManager[:132]- device queue for 3D.8F.F8 is empty!
[... cut ...]
2017-02-28 22:12:18 DEBUG o.o.b.i.i.d.MessageDispatcher[:139]- changing key to 0x19 for msg IN:Cmd:0x50|fromAddress:3D.8F.F8|toAddress:34.1F.4A|messageFlags:0x2B=ACK_OF_DIRECT:3:2|command1:0x0C|command2:0x00|
2017-02-28 22:12:18 INFO  o.o.b.i.i.d.MessageHandler[:468]- SwitchRequestReplyHandler: dev 3D.8F.F8 button 1 switched to OFF
2017-02-28 22:12:18 DEBUG o.o.b.i.i.device.DeviceFeature[:264]- 3D.8F.F8:LoadSwitchButton publishing: OFF
2017-02-28 22:12:18 DEBUG o.o.b.i.i.d.MessageDispatcher[:159]- defdisp: 3D.8F.F8:LoadSwitchButton set status to: QUERY_ANSWERED
2017-02-28 22:12:18 DEBUG o.o.b.i.i.device.DeviceFeature[:264]- 3D.8F.F8:GenericLastTime publishing: 2017-02-28T22:12:18
```

teharris1, 2018-01-10:
@larizzo I have been looking at this and I want to make sure I understand how this device works. If I read the documentation correction, for a 6 button device there is an "On" button, "Off" button and 4 "scene" or "group" buttons, correct? The "On" / "Off" are pretty easy to understand from an HA view (i.e. have one switch that sets the device to either on or off or dimmed) The scene buttons are a bit harder. From what I can tell you cannot set the state of them so they are controllers only (i.e. they control other devices or scenes but it is not possible to control them). Is that how I should look at this?

majortom64, 2018-01-29:
@teharris1 The 6-button and 8-button KeyPadLincs are basically the same device with two different configurations. They are really a set of buttons and LEDs, and a dimming or non-dimming load controller in the same box. By default, in 6-button mode buttons 1-2 are linked to the load's on command and buttons 7-8 are linked to its off command. In 8-button mode, button 1 is linked to the load. However, all units after 2013 should also support detached-load mode that allows one to change with button (if any) controls the local load.

This is really useful in cases where one has multiple keypads in a space (say various entries to a room) and wants to have a consistent experience (button 1 always turns on the overhead light, button 2 controls the sconces). In that case, assuming keypad A's load is the overhead lights and keypad B's load is the sconces, one would set keypad B into detached load mode and link its button 1 to the main load on keypad A.

Here is a great explanation of the KeyPadLinc: http://wiki.indigodomo.com/doku.php?id=inst_keypadlinc_buttons

As for the LEDs on the buttons, it is possible to control them directly, and one can put them in groups to have them all match state. If the load is controlled by the PLM, then the linked buttons do not change state (so they need to be sent an on or off directly). A nice approach is to create an Insteon group for all the buttons on all the KeyPadLincs that should be controlling the same thing and then send that group an on or off. (I explained some more of this in my issue on Insteon link management.)

http://www.madreporite.com/insteon/commands.htm is a great resource for Insteon commands.

The buttons themselves have two modes: toggle and non-toggle. In toggle mode, the button sends on and off with alternate presses. In non-toggle mode, the button can be set to always send either an on or an off. It is also possible to set buttons into an exclusive configuration so that when one is on, all the others are off.

teharris1, 2018-01-30:
@majortom64 great info. Thanks. Focusing on minimum viable product, I propose that we create the ability to use the buttons to control other devices (or it's own load depending on that setting). As I see it there are really only two devices, an On/Off device and a dimmable device. When I think about the critical function for a home automation system, the pressing of a button should trigger some other automation such as pressing button C unlocks an Z-Wave lock on the front door. This is not so much of a switch from the home automation system's point of view it is more of an event. It could even be considered a sensor so that the sensor value changes (ie. from on to off or from off to some level) and the home automation system tells some other event to occur. 

So, bottom line, if the keypad showed up in Home Assistant as a sensor where you could manage automation based on the button press of the keypad?

majortom64, 2018-01-30:
@teharris1 Your view is close. KeyPadLincs are really two types of three bundled devices: one load controller - dimmable or on/off, and a set of buttons and a set of lights. The problem is that they way people tend to use KPLs is by linking them together so that all linked buttons automatically maintain state. In other words, when I press button 3 (labeled c) on one KPL, it turns on the load on another and turns on the LEDs on the other two KPLs that also control the same light.

In other words, there is a load (may or may not be controlled locally - see detached load), a sensor and an indicator (led for button state).

The problem is that when the load on a KPL is turned on or off using a message from the PLM, the buttons on the other KPLs are not updated (meaning their state is wrong and that to change the state I have to press the button twice - once to get the button into the same state as the load, and the second time to actually send the right signal (if the button thinks the load is on, it will send an off). This means that to work properly in this situation, the PLM also need to be able send the codes to control the button light state.

Here is an example using all three: http://wiki.indigodomo.com/doku.php?id=garage_door_kpl_control

One can open the garage door by pressing the garage door opener button or by pressing one of the buttons on one of KeyPadLincs, but its status (open/closed) is displayed on that button no matter how it is opened or closed.

teharris1, 2018-01-31:
Got it. Great examples. Disconnected load is something on the backlog so not going to do that for this device only but the LEDs are interested. 

Not going to be able to get to this one for a few weeks. I want to get the refactored code into HA first. Also, I need to think about the number of moving parts and how they translate themselves into the user interface.

majortom64, 2018-01-31:
@teharris1 Glad they make sense. Insteon/SmartLabs took a long time to add detached loads to the firmware despite being asked for years, so I can wait a bit for it. Until you support link management, it really has little impact. Just know that one can have 9 buttons on a KeyPadLinc (the 9th is the detached load), and it is reasonable to have two links between a button on one device and another device (one link is from the button on the on KeyPadLinc to the button on the other, and the second link is from the button to the detached load on the other)

http://forums.indigodomo.com/download/file.php?id=5314

majortom64, 2018-02-10:
@teharris1 I found some useful correspondence from Jonathan Dale, the keeper of MadReporite about creating self-links (a link from the KeyPadLinc button to the KPL itself (needed to allow the KPL to be a part of a scene controlled by one of its own buttons, or to allow a button on the KPL to control the detached load):

Still… if you want to edit the link database, assuming your devices aren’t super old (this is for i2 devices), you would use the Get/Set Database command (page 16 of the file you linked) which my site has under the name Read/Write All-Link Database (0x2F) but the details are mostly the same. Don’t worry about all the other commands for putting the device into linking mode, you don’t need it, also it probably won’t work linking the device to itself that way. The only command you need is Get/Set Database.
 
First read the database to figure out what’s there, either D1-D13 all 0x00 (to get all the records starting with 0) or step through the records with the requested address in D3-D4 (stepping by 8’s) and any value other than 0x00 in D5 (your file says D5 should have the number of records you are requesting, but still 0x00 for all of them).
 
The response will have D1 unused, D2 0x01, D3-D4 address, D5 0x00, D6-D13: 8-byte record. If returning multiple records, the address is decremented by 8 for each record (since each database entry is 8 bytes).
 
Because this is a long response (many responses), the chances for it to get interrupted can be high, especially for a device with a lot of links (like a KeypadLinc). I found sometimes I needed to try multiple times.
 
To write a record, same command (your file calls this Set Database), with D1 unused, D2 0x02, D3-D4 address, D5 number of bytes (0x01-0x08), D6-D13 data to write.
 
Also see the Memory Map section of that file (page 19).

The file to which he refers is: http://cache.insteon.com/developer/2334-xxx2486xdev-042013-en.pdf

teharris1, 2018-03-03:
@larizzo @majortom64 I am taking this one on now that the refactored code has gotten into the 0.65.0 release. I need to dummy the first version of this down and hear is what I am thinking.

1. Assume the load is attached (critical assumption because otherwise development is 6 months out:)
2. Receive on/off and set level messages from buttons as sensors. (Hide them in HA to avoid cluttering up your interface but they can be used for automations. i.e. press button 'C' and have HA turn on the outside lights.'
3. LEDs will look like switches / lights in HA (switches for on/of and lights for dimmable).  You will be able to control them and use them to control other HA entities. 

Would this be useful and is it the right way to think about the device?

Future enhancements will allow triggering of scenes and attached/detach load (which has a lot of issues with it such as the need for a GUI)

teharris1, 2018-03-03:
OK, so after a little more reading, I think there is a fourth assumption

4. Button bit masks are same as button' LED (i.e. LED on button 3 has bit mask `0000 0100`) 

So then the question is, if button 3 is pressed should I assume LED 3 is on?

teharris1, 2018-03-05:
@majortom64 @larizzo I have a VERY experimental version available in my repo.  Download the insteonplm and home-assistant repos and install both. What you will see are three HA entities for each button on your KeypadLinc.  So if your KPL is a dimmer and the address 1A.2B.3C you will see:

- light.1a3b3c
- light.1a2b3c_led
- light.1a2b3c_scene
- light.1a2b3c_2
- light.1a2b3c_2_led
- light.1a2b3c_2_scene
- etc.
If your KPL is an on/off device light will be replaced with switch. Here is what I THINK will happen. light.1a2b3c will control the load (99% sure of that one).  light.1a2b3c_2 will control the B button. light.1a2b3c_2_led will control the B button LED on or off.  The one that should be really interesting is light.1a2b3c_2_scene. This should mimic the actual pressing of the button.  

As I said, this is very experimental and I am looking for feedback.

teharris1, 2018-03-05:
Here are the links to the two repos:
[insteonplm](https://github.com/teharris1/python-insteonplm)
[home-assistant](https://github.com/teharris1/home-assistant)

asktimfisher, 2018-06-26:
Been chatting with you in the HA Community (https://community.home-assistant.io/t/insteon-switch-button-automations/57761) - just wanted to chime in and welcome tagging me here as well for any and all testing you might be after!

teharris1, 2018-07-18:
@majortom64 @larizzo @asktimfisher I am back on this one. There are a few design decisions that I could use input on. Given the complexity of the device's functionality may come in different releases. The first release, I believe should focus on:
1. Responding to button presses on the device (button 1 works currently but 2 - 8 do not)
2. Capture LED status and display that status to the UI
3. Trigger a scene on the device 

Technically the device functions such that the buttons trigger scenes. The reverse then should be true where the UI triggers the scene on the device. This would mean you never actually trigger a button, you trigger a scene. Also, the LEDs are separate from the button technically. In other words, a scene can turn on an LED on a button but that button may not be associated with that scene. This makes for a very confusing UI requirement. Going to start experimenting but wanted to give you all an update.

majortom64, 2018-07-20:
@teharris1 I think that the buttons can either trigger groups or scenes. Most of my KeyPad Lincs use grouped buttons to keep the LED and button state in sync with some light (meaning that the buttons only control one light, not scenes with groups of lights).

For your list, I would modify 2 and add 4 for your first release:

2. Mirror LED status between the UI and the device.

This means that when I turn off a light using the UI all the linked KeyPad Lincs should also turn off the LED. 

Which leads to adding:

4. Be able to change the state of an LED on a keypad based on some HASS condition.

For example:  I use an led as an indicator for garage door open status (from my I/O Linc garage door sensor) and the button to control the door (see [here](https://wiki.indigodomo.com/doku.php?id=garage_door_kpl_control)). The LED flashes while the door is changing state and then is on when the door is open and off when closed.

As far as scene control, I have always had two problems with the direct linking of a button to a multi item scene. I press button 3 to activate scene "dinner". 

That scene sets:

1. Kitchen SwitchLinc to 50%.
2. Dining room sconce SwitchLinc to 50%
3. Dining room table chandelier SwitchLinc to 75%

and the LED on button 3 turns on.

If I then turn off the Kitchen SwitchLinc, the button 3 LED does not go off, even though the scene is no longer active (this would be possible if HASS understood the state of a scene and updated the LED accordingly).

The second is that it is not clear what "turning off" a scene should do. Should it turn off every item in the scene? Should it reset them to their previous states? Should it just be a different scene that the creator thinks of as the "off" state for the scene?

Also, if one is using a KeyPad Linc in a HASS system, it would be nice to be able to create scenes that were a mix between Insteon groups/scenes and HASS scenes.

asktimfisher, 2018-07-22:
Just confirming everything you're both saying as an ideal MVP. #1 on your list is clearly the biggie. :)

teharris1, 2018-07-27:
Do any of you have the KPL Dimmer? I do and it is the most frustrating device. A few observations:
1) You cannot set one LED at a time. You must set every LED in one command. This means that rapid-fire changes are VERY unreliable. 
2) The device is SLOW to respond to status requests and very often does not respond at all. You get an acknowledgment that the command was received but then nothing. Frustrating. 
3) There are separate settings for the LED on/off and the button on/off/level. However, the button value appears to be meaningless, which is why when you link the dimmer to a dimmable responder, the dimming function is unpredictable.

In general, I think this is a poorly designed device but I digress. 

So, looking at a lot of other code from other developers, I am going to turn the LEDs on and off but not bother with setting the button. Nobody seems to be setting the button on level, they only seem to be setting the LED.  Here is now the important question:

**_Does it matter if the buttons are dimmable or would on/off buttons in Home Assistant be fine?_**

The implication is they would show up as `switch` rather than `light` and would not have a dimming feature even if it is a dimming device.

teharris1, 2018-07-28:
@majortom64 @asktimfisher Testing time!!!!

Download the following files.:
[insteonplm](https://github.com/teharris1/python-insteonplm/archive/keypadlinc.tar.gz)
[Home Assistant](https://github.com/teharris1/home-assistant/archive/KeypadLinc.tar.gz)

Uninstall the current version of `insteonplm` (`sudo pip3 uninstall insteonplm`)
Install the downloaded version of `insteonplm` (from the directory that you untar it in `sudo pip3 install .`)
Copy the following files to your Home Assistant instance:
- `homeassistant/components/insteon_plm/__init__.py`
- `homeassistant/components/switch/insteon_plm.py`

majortom64, 2018-07-28:
All (or almost all) my KeyPadLincs are dimmers and most of them are linked to dimmable SwitchLinc switches (I only have a small number of on/off switches in my whole environment). Setting the LED state is most important, as that matters when I use Siri to turn off a device and that is linked to other KeyPadLinc buttons. I would argue that switching the button state when one changes the LED state is also important as it creates a confusing experience where one has to push the button twice to get the desired result. Finally, I am less concerned about setting the level through the interface, but making all the buttons on/off only would be a problem (as I often dim the connected lights via linked buttons).

teharris1, 2018-07-28:
This is why the device is so frustrating for me. If I set the LED to on, even if the underlying button is technically off, when I push the button it sends an off command. The button appears to follow the LED more than it follows the `on level` of the button. This makes sense from a user experience view but technically it is difficult to understand. why they did not just have the button turn on if you turn on the LED or turn on the LED if you turn on the button. Give this code a try and tell me what use cases you would want to enable.

majortom64, 2018-07-28:
Will do!

asktimfisher, 2018-07-28:
I'm using the non-USB Insteon Hub (https://www.amazon.com/gp/product/B00VG5DB86/). Do I need one of the devices listed in the PLM (here: https://www.home-assistant.io/components/insteon_plm/) before I can test for you?

teharris1, 2018-07-28:
For this device, yes. But I have been looking for testers for the hub. There is another issue open for that. Can you help there?

teharris1, 2018-07-29:
Yes, same overall goals. I added the KeypadLinc code to the hub code so the KPL should work in that code too. You can find that thread here #20.  Post any comments about the hub in that thread. If you see specific things about the KPL you can post them in either thread.

teharris1, 2018-08-03:
@majortom64 @asktimfisher any feedback yet?

asktimfisher, 2018-08-26:
Sorry for being AWOL on this one! Rebuilding my whole platform from Hassio/Pi to Ubuntu(Docker)/NUC. Not quite done but will be pulling this in and testing as soon as it's up.

asktimfisher, 2018-10-11:
Finally got HA up and running last night and should be able to test this out this weekend. May reach out for some assistance (via the HA forums) since this will be my first time testing new code in HA.

closed 2019-07-30 by teharris1

asktimfisher, 2019-07-30:
> @asktimfisher How are you doing? Any progress?

Sorry about the lack of communication! For completely unrelated reasons to HA support, I decided to move on from Insteon.