nugget/python-insteonplm #4: Add support for X10 powerline devices.

rstanley75, 2017-03-13:
Yup, I said it.  X10.

My house was built in 1937 so not everything is exactly what you would call up-to-code.  As such, there are light switches that are still stuck in no-neutral-wire hell (and likely will be for quite some time).  I have a few lingering X10 devices in my house.  While it would be wonderful to purge my life of these awful things, for the time being I seem to be stuck with them.  As much as I despise these devices it would be really nice if I had a way to control them.

Potential difficulties as I see them:

- A whole different set of weird communication parameters.  For example, two messages must be transmitted to send one command.
- Few X10 devices support two-way communication, meaning we can't rely on the these devices to tell us whether or not they've changed state or respond to any requests as to their state.
- We can't get a list of X10 devices from the PLM so the device addresses will need to be added manually in the config file.

glyndon, 2018-04-04:
Same here. 
Wanted to switch to HASS but had to stick with OpenHAB because of too many X10 devices I still need to keep in service.
Would be happy to help make that feature work, but not sure where to begin.

teharris1, 2018-04-05:
Ok, so this is not that hard of a change if I understand X10 correctly. My main questions are:
1. How should the user interface behave? X10 devices don't give feedback so I assume optimistic status is required. Is a simple on/off switch sufficient or is a variable level required (i.e. dimmable switch)?
2. What basic commands would you want supported? On, off, what else?
3. Can you help me find detailed docs on the protocol?

Not sure when I can get to this one but if we keep it simple and build up it should be relatively easy.

glyndon, 2018-04-05:
In my case, the X10 devices are a mix of dimmers and switches, but mostly just on/off switches. 
On/off/bright/dim should handle most cases.
The OpenHAB add-in for the InsteonPLM already handles X-10 pretty well, maybe that code would be helpful?
https://github.com/openhab/openhab1-addons/tree/master/bundles/binding/org.openhab.binding.insteonplm

glyndon, 2018-04-09:
Here are 3 files [I'd kept from years ago and just found] which explain the X-10 protocol as it was to be transacted through a CM11A, an RS-232 PLM for X-10. These docs are all describing the same thing, so they mostly overlap. The docs on X10 were known to have some odd errors, so having various versions to compare may help.

[CM11A programming protocol - cm11a_protocol.txt](https://github.com/nugget/python-insteonplm/files/1891589/CM11A.programming.protocol.-.cm11a_protocol.txt)
[Cm11a-ver0.5v.zip](https://github.com/nugget/python-insteonplm/files/1891590/Cm11a-ver0.5v.zip)
[rf-protocol.txt](https://github.com/nugget/python-insteonplm/files/1891591/rf-protocol.txt)

glyndon, 2018-04-14:
In the meantime, I'm going to give the following a try, since I still think my old CM11A may be in working order.
It'll be ugly, but so is X10 ;-)
https://www.home-assistant.io/components/light.x10/

glyndon, 2018-04-21:
FWIW, the CM11-HEYU-MQTT stack is working for me so far. I had to modify the heyu-mqtt.pl script I found so it would also pass the AllOff, Bright, Dim, LightsOn keypad buttons, and now it does.
I'd still prefer to have this module do the X11 directly for HA, but this workaround will let me fully use HA in the meantime.

teharris1, 2018-05-08:
FYI, I have started working on this feature. Hoping to be at a working version by the end of this week. The first pass is going to use the command line interface `insteonplm_interactive` that comes with the newer versions. Once we show it is working in that tool I will wire up Home Assistant. I will need your help to test for sure since I don't have any X10 devices.

glyndon, 2018-05-08:
Happy to test. On vacation for the week in case I respond late.

teharris1, 2018-05-14:
Let the testing begin!

You can download the test code here: [Insteon PLM X10](https://github.com/teharris1/python-insteonplm/archive/x10.tar.gz)

Untar and install with pip (i.e. change directory to the python-insteonplm and use `sudo pip install .`) Don't foget to uninstall older versions first (`sudo pip uninstall insteonplm`).

Once installed run the `insteonplm_interactive` command to start the interactive command shell.  Inside the command shell enter the following commands:

`set_device /dev/<some device>`
`set_workdir /home/<username>`   # This step is optional but recommended for future startups
`connect`
`add_x10_device A 1 OnOff`
`on_off_test x10A01`  # the format of this is x10 <housecode><unit code> as 2 digits 01 - 16

If you have not used the insteonplm library lately, the `connect` command may take a LONG time. I have 30 devices and the first time it ran it took about 10 minutes. If you use the `set_workdir` command above, subsiquent startups will be much faster. 

This is just a first step and right now only On/Off devices are supported. Next I will make Dimmable supported. After that I will add it to Home Assistant. If either of you have a motion sensor or other binary sensor I can work on that too.

glyndon, 2018-05-14:
Ok, I'll give it a shot (probably not until tomorrow).
I have dimmable and on/off X10 devices, and also have 'sensors' in the form of keypads which send commands (we can treat them as sensors, since they send what the actual legacy X10 physical sensors sent).

teharris1, 2018-05-14:
Assuming you only have one PLM this needs to have exclusive access to the device/port. So, no you need to shut down HASS. But if you already have HASS running then when you run the `set_workdir` command, use your Home Assistant config directory. In that directory there should be a file called `insteon_plm_device_info.dat`. That is the file that the insteonplm module created in the `workdir`. Using the config directory for the workdir will improve the start up time a lot.

When we get the On/Off function working, dimmables are easy. As for sensors, it would help if I can capture a log of the messages. In the tool you can use `set_log_level v` which will put the logging level into "verbose"  mode. From there trigger the 'sensor' and you should see a few lines that say 
`Processing message: {'code': 0x52, 'rawX10': ...`
Copy any lines like that and post them so I can see what is happening.

glyndon, 2018-05-14:
On/Off did not succeed on the X10 device (mine is set to G3). No action observed.
The same verb did work when directed at an Insteon device.
Lot of that test attached.
Keypad messages intercepted and logged, also attached.
The keypad did activate the G3 device.

[x10_keypad_message_exercise.txt](https://github.com/nugget/python-insteonplm/files/2001168/x10_keypad_message_exercise.txt)
[x10_on_off_test_log.txt](https://github.com/nugget/python-insteonplm/files/2001169/x10_on_off_test_log.txt)

teharris1, 2018-05-14:
So looking at the keypad file, it is obvious that the housecode translation should be G = hex D = 1101. I will have to check my house code translations.

teharris1, 2018-05-14:
Hmmm, this is what OpenHab has:
        s_houseCodeToString.put(0x6, "A");
        s_unitCodeToInt.put(0x6, 1);
        s_houseCodeToString.put(0xe, "B");
        s_unitCodeToInt.put(0xe, 2);
        s_houseCodeToString.put(0x2, "C");
        s_unitCodeToInt.put(0x2, 3);
        s_houseCodeToString.put(0xa, "D");
        s_unitCodeToInt.put(0xa, 4);
        s_houseCodeToString.put(0x1, "E");
        s_unitCodeToInt.put(0x1, 5);
        s_houseCodeToString.put(0x9, "F");
        s_unitCodeToInt.put(0x9, 6);
        s_houseCodeToString.put(0x5, "G");
        s_unitCodeToInt.put(0x5, 7);
        s_houseCodeToString.put(0xd, "H");
        s_unitCodeToInt.put(0xd, 8);
        s_houseCodeToString.put(0x7, "I");
        s_unitCodeToInt.put(0x7, 9);
        s_houseCodeToString.put(0xf, "J");
        s_unitCodeToInt.put(0xf, 10);
        s_houseCodeToString.put(0x3, "K");
        s_unitCodeToInt.put(0x3, 11);
        s_houseCodeToString.put(0xb, "L");
        s_unitCodeToInt.put(0xb, 12);
        s_houseCodeToString.put(0x0, "M");
        s_unitCodeToInt.put(0x0, 13);
        s_houseCodeToString.put(0x8, "N");
        s_unitCodeToInt.put(0x8, 14);
        s_houseCodeToString.put(0x4, "O");
        s_unitCodeToInt.put(0x4, 15);
        s_houseCodeToString.put(0xc, "P");
        s_unitCodeToInt.put(0xc, 16);
Not sure how easy this is for you to read but it has G as 0x05 also. (technically 0x50 but that is a detail in the implementation). 0x0D is H. Are you sure you did not press the 'H3' button not the 'G3' button?

glyndon, 2018-05-14:
My bad... My VERY VERY bad, and I'm thus VERY Sorry. You are correct. I had the keypad and device set to H, not G. (I was *sure* that I'd set G). I'll repeat the test using H and forward results.

glyndon, 2018-05-14:
Success. 
Attached are new logs, showing the result when I sent the housecode that the devices are actually set to ('H').
[x10_keypad_message_exercise(2).txt](https://github.com/nugget/python-insteonplm/files/2001795/x10_keypad_message_exercise.2.txt)
[x10_on_off_test_log(2).txt](https://github.com/nugget/python-insteonplm/files/2001796/x10_on_off_test_log.2.txt)

teharris1, 2018-05-14:
Excellent. Thanks for testing. It should go pretty quickly from here. I am going to do the following:
1. Add dimming
2. Wire up Home Assistant
Should have something to test in a couple of days

teharris1, 2018-05-19:
HA is ready for testing. Download [Home Assistant](https://github.com/teharris1/home-assistant/archive/x10.tar.gz) and [insteonplm ](https://github.com/teharris1/python-insteonplm/archive/x10.tar.gz) from my repo. From Home Assistant you can install directly from that repo or you can use your current install and replace the following files:
home-assistant/homeassistant/components/insteon_plm  (THIS IS A DIRECTORY with 2 files in it)
home-assistant/homeassistant/components/light/insteon_plm.py
home-assistant/homeassistant/components/switch/insteon_plm.py

You can also replace the insteon_plm.py files under binary_sensor, sensor and fan if you want. If you have a file `home-assistant/homeassistant/components/insteon_plm.py` may sure to delete it.

In your configuration.yaml file add something like the following:
```
insteon_plm:
  port: /dev/insteon
  x10_devices:
    - housecode: c
      unitcode: 12
      platform: switch
    - housecode: b
      unitcode: 9
      platform: light
      dim_steps: 64
```
platform is either `light` for dimmable devices or `switch` for on/off only devices. Dimmable devices have a variable number of steps to dim from full on to full off. Normally this is 22, which is the default. If your device has a different number of steps then use the `dim_steps` attribute otherwise you can ignore it.

glyndon, 2018-05-20:
Very cool. Will test ASAP.
Any word on adding incoming X10 events (keypad buttons, and switch actions)?
X10 switches (the in-wall kind, at least) send their address and the action command (e.g., on/off) when operated manually (so other same-coded devices can act as if in unison), it might be handy in HA automations to be able to react to them.

teharris1, 2018-05-20:
I should already work.

teharris1, 2018-05-21:
@glyndon I just double checked and inbound dim and bright should work too.

glyndon, 2018-05-21:
Ok. I'll check it out. I saw in the code that you're handling messages bi-directionally, so just may need to experiment a bit with HA (still finding my way around in there) to see how it manifests and how to react to it in automation. 
Thanks for this work!!!

teharris1, 2018-05-22:
Any progress?

glyndon, 2018-05-23:
Not yet. Because I'll be testing it in my live system, I am waiting until I have a large-enough block of time wherein I can backup the existing filespace, and to become confident that I can reverse the install instructions if I need to rollback.
Should be within a day or so.
Meanwhile, can you remind me how I'll be able to detect events from the keypad buttons that don't correspond to a device (e.g. 'Dim', 'AllOff', etc)?

glyndon, 2018-05-23:
Success - I think.
It's controlling an appliance module (switch) successfully. (I don't have any dimmers in use at the moment, but will get one and try it.)

This floated by the systemctl journal as it was starting up, but didn't seem to be fatal:

`May 23 05:51:06 rosie hass[1781]: 2018-05-23 05:51:06 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback IM._handle_get_next_all_link_record_nak(<insteonplm.m...at 0xa6911470>)
May 23 05:51:06 rosie hass[1781]: Traceback (most recent call last):
May 23 05:51:06 rosie hass[1781]:   File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
May 23 05:51:06 rosie hass[1781]:     self._callback(*self._args)
May 23 05:51:06 rosie hass[1781]:   File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/plm.py", line 536, in _handle_get_next_all_link_record_nak
May 23 05:51:06 rosie hass[1781]:     self.devices.save_device_info()
May 23 05:51:06 rosie hass[1781]:   File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/linkedDevices.py", line 209, in save_device_info
May 23 05:51:06 rosie hass[1781]:     'cat': device.cat,
May 23 05:51:06 rosie hass[1781]: AttributeError: 'X10OnOff' object has no attribute 'cat'
May 23 05:51:07 rosie hass[1781]: 2018-05-23 05:51:07 INFO (MainThread) [homeassistant.core] Bus:Handling <Event system_log_event[L]: source=/srv/homeassistant/lib/python3.5/site-packages/homeassistant/core.py, exception=Traceback (most recent call last):
May 23 05:51:07 rosie hass[1781]:   File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
May 23 05:51:07 rosie hass[1781]:     self._callback(*self._args)
May 23 05:51:07 rosie hass[1781]:   File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/plm.py", line 536, in _handle_get_next_all_link_record_nak
May 23 05:51:07 rosie hass[1781]:     self.devices.save_device_info()
May 23 05:51:07 rosie hass[1781]:   File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/linkedDevices.py", line 209, in save_device_info
May 23 05:51:07 rosie hass[1781]:     'cat': device.cat,
May 23 05:51:07 rosie hass[1781]: AttributeError: 'X10OnOff' object has no attribute 'cat'
May 23 05:51:07 rosie hass[1781]: , timestamp=1527069066.748606, message=Error doing job: Exception in callback IM._handle_get_next_all_link_record_nak(<insteonplm.m...at 0xa6911470>), level=ERROR>
`

teharris1, 2018-05-23:
This error does not impact the functionality of the devices. Not a fatal error but this means your `insteon_plm_device_info.dat` file is wiped. The impact is that future start ups will take much longer. Easy fix but not until tomorrow most likely.

glyndon, 2018-05-23:
That explains the lengthier-than-usual journal traffic when starting, even though I didn't notice it taking much longer time. I'm running on a Pi1Bplus, so am used to it taking a couple minutes to start (a good time to go make/get coffee). Once running and stabilized, it uses all of 3% CPU so as long as it remains happy on the 1B, I'm happy.

glyndon, 2018-05-25:
Anecdotal, not well-verified, but consistent here is that after working successfully this morning, the x10 sensors seem now to be working only sporadically. 
I've power-cycled the PLM, and restarted HASS, but the symptom remains consistent: Intermittent response to incoming X10 signals. 
While I can't rule out a phy problem in my premises, I can say that if I use a keypad to directly signal a device, it works reliably. But the PLM doesn't seem to be noticing these and passing them to the sensor entities reliably.

teharris1, 2018-05-28:
@glyndon I saw this in an email from you but I don't see it in the thread:

`Anecdotal, not well-verified, but consistent here is that after working successfully this morning, the x10 sensors seem now to be working only sporadically.
I've power-cycled the PLM, and restarted HASS, but the symptom remains consistent: Intermittent response to incoming X10 signals.
While I can't rule out a phy problem in my premises, I can say that if I use a keypad to directly signal a device, it works reliably. But the PLM doesn't seem to be noticing these and passing them to the sensor entities reliably.
`
Not sure if you deleted the comment or not but if you did not and somehow it was lost, can you do the following:

1. restart the system
2. tail the log file with the following command `tail -f home-assistant.log | grep -E 'Processing'`
3. Let me know if all tests result in some type of "Processing message..." line

glyndon, 2018-05-28:
I deleted the comment because I later found it to be again working fine with no action (ie. didn't do anything to HA) on my part. So, I wrote the problem off to the relatively rare but not unheard-of problem here with something transient interfering with X10 signals. Happens only about once a year, and always passes (so I tend to forget about it). Given the coincidence of happening to be testing the new code, I'd initially thought maybe the code had a mem/socket leak or something that was manifesting after a few hours running, and wanted to report it. But after I left it running undisturbed, and the next morning x10 was fine again, I realized the problem had to be external to HA.

glyndon, 2018-05-28:
Just one lingering question: How do I define a sensor for the 4 'meta' buttons on the keypad - the ones that send codes to all devices (AllOn, AllLigtsOff) or the device last changed (bright/dim)?

geek65535, 2018-05-28:
Are you going to maintain this in your own fork, or do a PR back to nugget's repo?
I'm looking forward to this code going mainstream, especially I just switched from OpenHAB to find that HA doesn't support X10 on Insteon (which OpenHAB does).

One thing: I grabbed the files you posted above and copied them into place, but I'm still getting an "invalid configuration" error.
or during setup of component insteon_plm

Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/setup.py", line 142, in _async_setup_component
    result = await component.async_setup(hass, processed_config)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/insteon_plm/__init__.py", line 100, in async_setup
    ipdb = IPDB()
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/insteon_plm/__init__.py", line 249, in __init__
    from insteonplm.states.x10 import (X10DimmableSwitch,
ImportError: No module named 'insteonplm.states.x10'

glyndon, 2018-05-29:
In addition to copying over the files from the HA repository, you need also to do a remove/install of the insteonplm module itself, per the instructions several comments above.

teharris1, 2018-05-29:
@geek65535, yes @glyndon is correct. That is the issue you are seeing.

@glyndon Let's discuss the use case for the All-On/All-Off switches. First, can you confirm that All-On and All-Off are working in Home Assistant? They should be. If they are not then I need to review the code. Assuming they work, the inbound All-On/All-Off is sent to all X10 devices to turn them on/off. 

My plan for All-On/All-Off is to make them services in Home Assistant. This would allow them to be used in automations so HA could send an All-Off or All-On command. In other words, All-On and All Off are commands to send, not devices. Does that work?

geek65535, 2018-05-29:
@glyndon I installed the insteonplm module, but I didn't see any instructions on removing the module first. I tried 'pip3 uninstall insteonplm' then 'cd python-insteonplm-x10; pip3 install .'
I'm still getting errors like this when running insteonplm_monitor and it receives x10 messages:
ERROR:asyncio:Exception in callback IM._handle_x10_send_receive(<insteonplm.m...at 0x76a58f90>)
handle: <Handle IM._handle_x10_send_receive(<insteonplm.m...at 0x76a58f90>)>
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/home/homeassistant/.local/lib/python3.5/site-packages/insteonplm/plm.py", line 574, in _handle_x10_send_receive
    self.devices[self._x10_address.id].receive_message(msg)
AttributeError: 'NoneType' object has no attribute 'receive_message'
ERROR:asyncio:Exception in callback IM._handle_x10_send_receive(<insteonplm.m...at 0x763ba1f0>)
handle: <Handle IM._handle_x10_send_receive(<insteonplm.m...at 0x763ba1f0>)>
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/home/homeassistant/.local/lib/python3.5/site-packages/insteonplm/plm.py", line 576, in _handle_x10_send_receive
    self._x10_command_to_device(housecode, unit_command_byte, msg)
  File "/home/homeassistant/.local/lib/python3.5/site-packages/insteonplm/plm.py", line 582, in _x10_command_to_device
    self.devices[self._x10_address.id].receive_message(msg)
AttributeError: 'NoneType' object has no attribute 'receive_message'

This is on a Pi 3B running Hassbian, if that helps. Speaking of help, thanks for any you can give me.

teharris1, 2018-05-29:
@geek65535 my repository is just a development repo. I merge it into the main repo when features are ready. So, yes it will be part of the base install. Right now we are on HA release 0.70.0. I would expect the X10 functionality to make it into 0.71.0 in about 2-3 weeks.

geek65535, 2018-05-29:
@teharris1 Thanks! I appreciate the work you're doing here very much.

teharris1, 2018-05-29:
@geek65535 the commands you are using to uninstall and install the new files looks correct. Check your Home Assistant config directory. I often find a `deps` directory in there for some unknown reason. It contains dependency modules. If it is there see if there is a latent install of insteonplm.

If it is not there, check to see if there are other insteonplm installs. The easiest way to do this is with find:
`cd /`
`find . -name insteonplm`

teharris1, 2018-05-29:
@geek65535 Just re-read your last post. The issue you have is not an install issue.

teharris1, 2018-05-29:
@geek65535 OK, you found a bug, so thanks. :) 

However, insteonplm_monitor is not intended to work with X10 since you cannot add an X10 device in monitor mode. Give `insteonplm_interactive` a try.  Per the following:

> Once installed run the insteonplm_interactive command to start the interactive command shell. Inside the command shell enter the following commands:
set_device /dev/<some device>
set_workdir /home/<username> # This step is optional but recommended for future startups
connect
add_x10_device A 1 OnOff
on_off_test x10A01 # the format of this is x10 as 2 digits 01 - 16

teharris1, 2018-05-29:
@glyndon, @rstanley75, @geek65535 Input needed.

Looking at the code and the `All Units On` and `All Lights Off` X10 commands should be working but the 'All Lights On` command should not be. I will 'fix' that but is there a real difference between `All Units Off` and `All Lights Off`?  How should I think about those two commands?

teharris1, 2018-05-29:
I have now added the following services to HA:
`insteon_plm.x10_all_units_off`
`insteon_plm.x10_all_lights_on`
`insteon_plm.x10_all_lights_off`

These would be used in automations as follows:

`- id: X10_ALL_LIGHTS_OFF_EXAMPLE'
  alias: X10 All Lights Off
  trigger:
  - at: '23:00'
    platform: time
  condition: []
  action:
  - data:
      housecode: a
    service: insteon_plm.x10_all_lights_off
`

teharris1, 2018-05-29:
Also, `binary_sensor` now works as a device type in HA. Use this for your keypads. So now your X10 keypad should be able to trigger events in HA (i.e. "lock the front door")

glyndon, 2018-05-29:
To your question about the AllOn (et al) buttons:

I use them as inputs - sensors - in HA, in order to programatically define their actions (e.g., as a trigger leading to some automation or perhaps to activate a scene). Having them act like services may be unnecessary since a scene can do that already (just list all the X10 devices in it) and is more in keeping with how HA supports broad actions on other types of devices.

You are correct in that - in a legacy X10 environment - they don't send device-specific verbs, but rather their codes are respected by all X10 devices, according to the type of device.

Dim and Bright normally affect only the last device(s) addressed (this is a characteristic built into the devices "if I am a lamp module and was last spoken to with an On/Off, I will now act on this dim/bright command").
AllOn/AllOff are also non-device specific, and all X10 devices will react to their messages (except that non-Lamp modules ignore AllLightsOn, and only react to AllUnitsOff).

But in my use, I just treat all 4 of them as generic 'function keys' which I can react to in HA as I wish, since I keep the keypad on a different housecode as the devices, so HA is the only path between the keypad and my devices. That mode may be a handy option for some users.
I'm using them in this way today via my Heyu/MQTT bridge. HA is subscribed to their MQTT topic as the triggers in a set of automations. So if they could also be defined in HA as a sensor - or even as a switch as are the other buttons - that would be sufficient to have HA know when a user presses one of them.

glyndon, 2018-05-29:
When calling the `insteon_plm.x10_all_units_off` service, I'm getting the following in the log (and it's not successful at controlling the devices).

`2018-05-29 09:11:21 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback IM._handle_x10_send_receive(<insteonplm.m...at 0xb0a90fb0>)
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/plm.py", line 600, in _handle_x10_send_receive
    self.devices[self._x10_address.id].receive_message(msg)
AttributeError: 'NoneType' object has no attribute 'receive_message'
`

and

`2018-05-29 09:12:11 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/core.py", line 988, in _event_to_service_call
    service_data = service_handler.schema(service_data)
TypeError: 'str' object is not callable
`

teharris1, 2018-05-29:
@glyndon I see the issue with the second problem and I just posted a fix. The issue was on the Home Assistant side so that is the code to replace.

Regarding the first issue, I see how to prevent the error message but I want to make sure I am seeing the full problem. The error is saying there is no device registered with the X10 address the message is directed to. In other words, a message is coming for X10 device C 12 but there is no device C 12 registered with Home Assistant. Can you do two things for me:

1. Confirm your configuration.yaml settings for the X10 devices
2. Send me the log file line that says "Processing message: ..." that came in right before the error message.

glyndon, 2018-05-29:
I haven't changed the config.yaml settings since they were working yesterday, but I'll include them here, in case I'm using the wrong platform now.

`insteon_plm:
  port: /dev/USBinsteon
  x10_devices:
    - housecode: g
      unitcode: 7
      platform: switch
    - housecode: h
      unitcode: 1
      platform: sensor
    - housecode: h
      unitcode: 2
      platform: sensor
    - housecode: h
      unitcode: 3
      platform: sensor
    - housecode: h
      unitcode: 4
      platform: sensor
    - housecode: h
      unitcode: 7
      platform: sensor`
(G7 is an actual device to which I send on/off in scenes. There are no devices on 'H', those are virtual devices I'm using as sensors to react to the keypad)

Re: the first issue, a log line with "Processing message" never appears. However, *any* X10 traffic, even to devices not defined in HA and no matter which keypad button is pressed, causes the following in the log.

`2018-05-29 11:43:21 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback IM._handle_x10_send_receive(<insteonplm.m...at 0xb0a3cc50>)
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/plm.py", line 600, in _handle_x10_send_receive
    self.devices[self._x10_address.id].receive_message(msg)
AttributeError: 'NoneType' object has no attribute 'receive_message'
`

teharris1, 2018-05-29:
By the way, i should tell you that I am running a 2448A7 USB stick PLM. It will not even send X10 messages. So I have no way of testing anything. I am sorry this is such a PTA and I appreciate the support in testing.

glyndon, 2018-05-29:
insteonplm worked. Followed your test instructions above (May 13) and it seems fine.

glyndon, 2018-05-29:
> BTW, thanks again for the description above regarding AllOn/AllOff and
> based on that I may have an issue with the code. If you press the dim
> button 5 times does it send:
>
>    1. 10 messages (one with the unitcode and one with the command code
>    for each dim command) or
>    2. 6 messages (one unitcode message and 5 dim messages)
>
> The solution is designed around number 1 but can easily change to option
> 2. The extension to this question is if a unit code message is sent and 5
> command codes are sent, does the same unit process all 5 commands?

It actually only ever sends: housecode+’Dim’ (or ‘Bright’) each time. It’s
up to the devices to know if they should act on it (which they only do if
they were the last one specifically commanded on/off by their address).
On my keypad, if I hold it down for a few seconds, it sends a
housecode+’Dim’ and includes some kind of count meaning “how many”.
I can see this via the CM11a and heyu. That’s not necessarily a core
feature though, and may be anextended feature of this keypad. For my use, I
just want to know the button was pressed, and don’t carehow long it was
held.

glyndon, 2018-05-29:
To your question about sensor vs binary_sensor: 
I don't see a use for 'sensor' at all, in that I'm not aware of any x10 sending devices that send anything but 'on' or 'off' codes. They may exist, but I'm not aware.

glyndon, 2018-05-29:
Success: binary_sensors work (for device-coded keypad buttons, haven't ventured to the others yet), and control of OnOff (switch) and dimmable (light) devices works!
Now, I'm on the way to figuring out how I'll use the other 4 keypad 'binary_sensors'...

glyndon, 2018-05-29:
One odd quirk: Sending commands like dim/bright to an x10 light results in the plm 'stalling' for up to a minute. As if everything (including Insteon device commands) gets queued up, while it works out how to send the x10 commands. The eventually do happen, and in the order they were issued.
Not a show-stopper, but intriguing.

glyndon, 2018-05-29:
I'm still unclear on how to define binary_sensors for the 4 non-device keypad buttons. What would I call them? 
e.g., platform (binary_sensor) and housecode would be predictable, but what would I use as unitcode ?

geek65535, 2018-05-30:
So, I lost count of how many times I cloned a repo but forgot to switch branches, copied the code to the wrong folder, or forgot to restart the service--but I finally got it working! Thanks a bunch, @teharris1. It sounds like @glyndon is doing a good job of testing, but if you need anyone else to test anything (I've got a 2413U, if it makes a difference), I'll be happy to help.

teharris1, 2018-05-30:
@geek65535 glad to hear you got it working. The more testing the better.

teharris1, 2018-05-30:
@glyndon you said:
>One odd quirk: Sending commands like dim/bright to an x10 light results in the plm 'stalling' for up to a minute. As if everything (including Insteon device commands) gets queued up while it works out how to send the x10 commands. The eventually do happen, and in the order they were issued.

This is a direct result of a conservative approach to serial commands that were taken with this library. The delay is likely caused by a forced pause to wait for an acknowledgement from the PLM before sending the next command. I do queue up all the messages and then send them one by one and am careful of the order but now that I know that I only need to send one unit message for each dim/bright command series I think I can make it faster by removing the unit messages in between.

teharris1, 2018-05-30:
@glyndon Regarding the all on/off messages as devices/entities in HA:
I am open to the idea only because I don't have a better one so let me make sure i understand the use case:
1. You have a controller that has 4 buttons (All Units Off, All Lights On, All Lights Off, no idea what the 4th one is:)
2. When you press one of these 4 buttons you want HA to be able to react and use the signal as a trigger for an action. 
Did I get the use case correct?

glyndon, 2018-05-30:
There are 4 'special' buttons on the typical X10 keypad off to the right of the 8 unit-specific ones.
- Bright
- Dim
- All Lights On
- All Units Off

The last two labels aren't typo's. In a pure X10 environment they cause different results because the codes they send trigger different kinds of devices. I presume this is because they designed the keypads for most-common use cases. E.g., when you hear a strange sound at 2am and you want "All Lights On" you don't also want it to turn the coffeemaker on. Likewise, when you want everything off, you really want *everything* off (including the coffeemaker). 

But I don't need that functionality, as I can do it using HA Scenes. I just want to be able to deal with them as binary_sensor, so that I can use them to trigger automations.
In one of the files I attached a while back, I included the data from pressing each of these.
[https://github.com/nugget/python-insteonplm/files/2001168/x10_keypad_message_exercise.txt](url)

glyndon, 2018-05-30:
> NOW I get it. Feeling a little slow on the uptake. I also found a way to
> use events in HA that I think are the right thing to do here. Unfortunately
> it is a lot of work but is also applicable to other Insteon features like
> Double Tap. have a look at this:
> https://www.home-assistant.io/docs/automation/trigger/
>
> Conceptually this would be the implementation:
>
> automation:
>   trigger:
>     platform: event
>     event_type: insteonplm.x10_all_units_on
>     event_data:
>       housecode: a
>   action:
>     service: light.turn_on
>     data:
>       brightness: 150
>       rgb_color: [255, 0, 0]
>       entity_id:
>         - light.kitchen
>         - light.living_room
>
> How would that work for you? While I think this is the right way to do it,
> it will take me a while and is not in the pipeline.

That would work. Would it be easier just to add the buttons’ codes to the
routine that parses them as binary_sensor?
I think either way would work.

[22 more replies]

closed 2018-06-01 by teharris1