nugget/python-insteonplm #5: Add support for 240v Load Controler (2477SA1) and ON/OFF Switch (2477S)

larizzo, 2017-04-22:
I'm trying to add support for 240v Load Controler 2477SA1 and do a PR but something odd is happening when I test.

I added this line to the ipdb.py:
`Product(0x09, 0x0a, 0x000041, '240v Load Controler ', '2477SA1', ['switch']),`

I end up with both the 2477SA1 and the 2477S identifying as 240v Load Controler.

Logs for 2477SA1:
```
17-04-22 09:30:04 INFO (MainThread) [insteonplm.protocol] ALL-Link Record for 2E.8B.AF: flags:a2 group:01 data:00/00/00
17-04-22 09:30:04 INFO (MainThread) [insteonplm.protocol] ALL-Link Record for 2E.8B.AF: flags:e2 group:01 data:09/0a/41
17-04-22 09:30:04 INFO (MainThread) [insteonplm.protocol] New INSTEON Device 2E.8B.AF: 240v Load Controler (09:0a)
17-04-22 09:30:04 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 2E.8B.AF (2e8baf) ['switch']
17-04-22 09:30:04 INFO (MainThread) [homeassistant.core] Bus:Handling <Event platform_discovered[L]: platform=insteon_plm, discovered=[{'description': '240v Load Controler', 'subcat': 10, 'model': '2477SA1', 'address_hex': '2e8baf', 'address': '2E.8B.AF', 'firmware': 65, 'capabilities': ['switch'], 'product_key': 65, 'cat': 9}], service=load_platform.switch>
17-04-22 09:30:04 INFO (MainThread) [homeassistant.components.switch.insteon_plm] Registered 2E.8B.AF with switch platform.
17-04-22 09:30:04 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: old_state=None, new_state=<state switch.2e8baf=off; friendly_name=2E.8B.AF, Firmware=0x41, Cagegory=0x9, Model=2477SA1, Subcategory=0xa, Description=240v Load Controler, Product Key=65, INSTEON Address=2E.8B.AF @ 2017-04-22T09:30:04.882958-04:00>, entity_id=switch.2e8baf>
17-04-22 09:30:04 INFO (MainThread) [insteonplm.protocol] Requesting status for 2E.8B.AF
17-04-22 09:30:06 INFO (MainThread) [insteonplm.protocol] INSTEON device status 2E.8B.AF is at level 0xff
17-04-22 09:30:06 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: old_state=<state switch.2e8baf=off; friendly_name=2E.8B.AF, Firmware=0x41, Cagegory=0x9, Model=2477SA1, Subcategory=0xa, Description=240v Load Controler, Product Key=65, INSTEON Address=2E.8B.AF @ 2017-04-22T09:30:04.882958-04:00>, new_state=<state switch.2e8baf=on; friendly_name=2E.8B.AF, Firmware=0x41, Cagegory=0x9, Model=2477SA1, Subcategory=0xa, Description=240v Load Controler, Product Key=65, INSTEON Address=2E.8B.AF @ 2017-04-22T09:30:06.617403-04:00>, entity_id=switch.2e8baf>
```
Logs for 2477S:
```
17-04-22 09:30:04 INFO (MainThread) [insteonplm.protocol] ALL-Link Record for 3D.25.74: flags:a2 group:01 data:09/0a/41
17-04-22 09:30:04 INFO (MainThread) [insteonplm.protocol] New INSTEON Device 3D.25.74: 240v Load Controler (09:0a)
17-04-22 09:30:04 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 3D.25.74 (3d2574) ['switch']
17-04-22 09:30:04 INFO (MainThread) [homeassistant.core] Bus:Handling <Event platform_discovered[L]: platform=insteon_plm, discovered=[{'description': '240v Load Controler', 'subcat': 10, 'model': '2477SA1', 'address_hex': '3d2574', 'address': '3D.25.74', 'firmware': 65, 'capabilities': ['switch'], 'product_key': 65, 'cat': 9}], service=load_platform.switch>
17-04-22 09:30:04 INFO (MainThread) [homeassistant.components.switch.insteon_plm] Registered 3D.25.74 with switch platform.
17-04-22 09:30:04 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: old_state=None, new_state=<state switch.3d2574=off; friendly_name=3D.25.74, Firmware=0x41, Cagegory=0x9, Model=2477SA1, Subcategory=0xa, Description=240v Load Controler, Product Key=65, INSTEON Address=3D.25.74 @ 2017-04-22T09:30:04.913668-04:00>, entity_id=switch.3d2574>
17-04-22 09:30:04 INFO (MainThread) [insteonplm.protocol] ALL-Link Record for 3D.25.74: flags:e2 group:01 data:02/2a/45
17-04-22 09:30:04 INFO (MainThread) [insteonplm.protocol] Requesting status for 3D.25.74
17-04-22 09:30:06 INFO (MainThread) [insteonplm.protocol] INSTEON device status 3D.25.74 is at level 0x0
17-04-22 09:30:06 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: old_state=<state switch.3d2574=off; friendly_name=3D.25.74, Firmware=0x41, Cagegory=0x9, Model=2477SA1, Subcategory=0xa, Description=240v Load Controler, Product Key=65, INSTEON Address=3D.25.74 @ 2017-04-22T09:30:04.913668-04:00>, new_state=<state switch.3d2574=off; friendly_name=3D.25.74, Firmware=0x45, Cagegory=0x2, Model=2477SA1, Subcategory=0x2a, Description=240v Load Controler, Product Key=65, INSTEON Address=3D.25.74 @ 2017-04-22T09:30:04.913668-04:00>, entity_id=switch.3d2574>
```
Also I see this entry:
`Product(0x02, 0x2a, None, 'SwitchLinc Switch', '2477S', ['light']),`
Which I think should match the 2477S but does not match it as it was undetected before I added the config for the 2477SA1.

If you can let me know what I did wrong I'll test and create a PR for you. 

Thanks, Shawn

nugget, 2017-04-22:
That is really strange.  I wonder if it's possible the the ALL-Link database in your PLM has somehow gotten the record for your 2477S confused with the wrong cat/subcat of your 2477SA1.  That is where the initial cat/subcat/product data come from, not from the device itself.  It's not a behavior I've observed, but it seems possible.

If you run `insteonplm_monitor -v` (with --device as needed) you can see the raw traffic back and forth.  Additionally you could edit `tools.py` and add calls to `conn.protocol.product_data_request('3d2574')` and `conn.protocol.product_data_request('se8baf')` to see if the product data from the devices agrees with the product data in the PLM's internal database.

larizzo, 2017-04-23:
You're right, I unlinked and linked the two devices again and they're showing correctly now:
```
17-04-23 08:53:49 INFO (MainThread) [insteonplm.protocol] ALL-Link Record for 2E.8B.AF: flags:e2 group:01 data:09/0a/41
17-04-23 08:53:50 INFO (MainThread) [insteonplm.protocol] ALL-Link Record for 2E.8B.AF: flags:a2 group:01 data:09/0a/41
17-04-23 08:53:50 INFO (MainThread) [insteonplm.protocol] ALL-Link Record for 3D.25.74: flags:a2 group:01 data:02/2a/45
17-04-23 08:53:50 INFO (MainThread) [insteonplm.protocol] ALL-Link Record for 3D.25.74: flags:e2 group:01 data:02/2a/45
```
Should I be putting none or 0x000041? I see most of the time you put none but sometimes you put a value. I'm not sure what that does though.

`Product(0x09, 0x0a, 0x000041, '240v Load Controler', '2477SA1', ['switch']),`
or 
`Product(0x09, 0x0a, None, '240v Load Controler', '2477SA1', ['switch']),`

PR #8 (2017-05-01): Add support for 220/240V 30A Load Controller NO (2477SA1) and 220/240V 30A Load Controller NC (2477SA2) 
Closes #5.
closed 2017-05-01 by nugget