nugget/python-insteonplm #6: insteonplm hangs while doing initial discovery

tendiian, 2017-04-25:
I'm attaching 3 separate attempts to run Home Assistant.  All of them end in the same way hanging on initial device polling.  

The log settings were as follows:
```
logger:
  logs:
    aiohttp.server: critical
    insteonplm: debug
    homeassistant: critical
```
Configuration:
```
insteon_plm:
  port: /dev/ttyUSB0

```
Not sure what else to add at this point.  Please don't hesitate to ask for more info.

[run1.txt](https://github.com/nugget/python-insteonplm/files/954132/run1.txt)
[run2.txt](https://github.com/nugget/python-insteonplm/files/954131/run2.txt)
[run3.txt](https://github.com/nugget/python-insteonplm/files/954133/run3.txt)

tendiian, 2017-05-27:
Updated plm works now

closed 2017-05-27 by tendiian

elJosho, 2017-06-13:
can you expand on what you did to 'update plm'?  I'm pulling my hair out trying to get things working and I'm running into very similar errors.  the PLM works fine on another platform (altsteon) and has a populated ALDB, and I've tried HASS as well as the standalone version of insteonplm_monitor and I get the same errors.  thanks!

tendiian, 2017-06-17:
I bought a new one.  It's working about 70% of the time.  Seems to hang up
after 24-30 hours.  Sometimes have to restart multiple times to get it
through init.