nugget/python-insteonplm #9: What would it take...

lanrongwen, 2017-05-01:
I like the work that has been done so far, but I see many of my devices not showing up.  I'd like to help even if I have limited time and my Python is a bit rusty...  I have thermostats, open close sensors, dimmers, on/off switches that I can test against...

nugget, 2017-05-01:
For nearly any simple device it should just be a matter of addign the appropriate lines to https://github.com/nugget/python-insteonplm/blob/master/insteonplm/ipdb.py and then running insteon_plm_monitor to ensure that the messages make sense when you work with the device.

For device types which are significantly different than what's currently supported it's likely we'd want to add another callback method to suit that particular device.

lanrongwen, 2017-05-04:
K, I'll start working on it, I see a few devices I have that aren't in your list.  I'll have to learn up on the plm protocol...

lanrongwen, 2017-05-21:
I noticed in the ipdb.py file that some lights are marked 'dimmable' and others 'dimmer'.  In HomeAssistant it looks like the insteion_plm.py is looking for dimmable.  So I think I'm missing where the 'dimmer' comes into play.  Can you point me in the right direction?  Thanks.

teharris1, 2018-01-05:
@lanrongwen should we close this issue?

teharris1, 2018-02-08:
@lanrongwen I am closing this issue in favor of having separate issues for each device type we want to create. BTW, the issue about dimmer vs dimmable is fixed in the current release.

closed 2018-02-08 by teharris1