nugget/python-insteonplm #11: IPDB not Identifying Device Types

lanrongwen, 2017-05-27:
In my testing I have found the IPDB isn't correctly identifying my devices.  From what I can see it's basing it off the All_Link_Record lookup, but it always returns a generic 010000 code.  Looks like we could possibly change it to do a lookup to get a list of devices, then send that list through a loop or something to do a Product Data Request on each one (which is supposed to return the device class) and then add them to self.devices array...  Still learning my way around the code, so I'm not 100% sure if that would work.

nugget, 2017-05-27:
I actually had the code behave exactly the way you describe earlier during development but I backed off that approach because I was disappointed to see that a large portion of my devices simply do not respond to the product data requests at all.  For battery-powered devices like the 2845-222 hidden door sensor this makes sense, since they're not actively listening in order to conserve power.  Still, even some devices I would expect to be responsive were not.  For the majority of other devices, the response from the product data request ended up just being redundant and duplicated the information already received from the all-link database contents.

There's a little bit of vestigial code [here](https://github.com/nugget/python-insteonplm/blob/master/insteonplm/protocol.py#L340) which will still make the request if the all-link database returns a cat of 00 or no category at all.  It's possible we can resolve the problem of a corrupt or incomplete all link record by broadening the scope of that if/else to make it more aggressive about sending the request if things look incomplete or too generic.

There's also an extended status request that might be desirable/useful for some classes of devices, but I haven't explored.  It's supported but not invoked normally.

Example: https://github.com/nugget/python-insteonplm/blob/master/insteonplm/tools.py#L67

lanrongwen, 2017-05-29:
Hmmm...  So far it isn't working great for me, granted I haven't really made all the tweaks I was thinking about (still learning my way around the code).  I'd love to some how enhance the auto discover, but I have a feeling that most tools use the link process to capture the model data and then store it in a db or cfg file (such as the insteon house linc software).  Trying to think of what would make it best for users (such as in home-assistant).  Maybe the override could describe the class/subclass as well?  But would that actually add anything more than the simple override such as switch/light/etc.?  Maybe the other option is something where one could list all their devices (and their classes) and then the module would 1) either use the ones as listed in the all-link-db, or 2) add/link them if missing...

lanrongwen, 2017-06-16:
So I have a rough proto-type where I can define my devices and pass them in.  So far it seems to be working, just some more testing needed.  It should be backwards compatible with the current HA override setup...

nugget, 2017-07-30:
Assuming this was PR #15 I'm going to close the issue.  I just merged your PR and it's included in the 0.7.5 release I'm about to push to PyPi.

closed 2017-07-30 by nugget