nugget/python-insteonplm #12: Trouble getting things working

jfbloom22, 2017-06-05:
I am having a lot of fun telling Alexa to turn on and off my lights.  Thanks for putting this library together.  The only issue is that things stop working randomly.  Things work great and then I will turn off a light and I see it the toggle in Home Assistant turn off, then back on (the physical light turns off).  Once that happens then I know it is downhill.  I can not turn the light back on because the interface toggle thinks it is currently on.
I did some troubleshooting by shutting down Home Assistant and running `insteonplm_monitor --device /dev/ttyUSB0`.   I modified the tools.py file to simply turn on / or turn off the three lights I have connected.  Sometimes it would work, but most of the time it would only make it through one or two lights.  It seems like everything works reliably until I add a 3rd module.  

Below is the console output from insteonplm_monitor.

### A few things I have tried:
Factory Reset my 2413U as well as the 3 Insteon Dimmer modules.
Ensured that the modules are in the same room.
Re-Setup and tested my 2 Insteon Access Points.
I also briefly tested this out on my Raspberry Pi B+ thinking maybe the Model A is too slow.  But I had the same issue.  

### Setup:
Raspberry Pi Model A
Clean Install of Hassbian (Linux hassbian 4.9.24+ #993 Wed Apr 26 17:56:54 BST 2017 armv6l GNU/Linux)
Insteon 2413U
Python 3.4.2
Master branch of python-insteonplm

### Console output
I modified this some so that it would turn the module on, wait two seconds, then request the status of the module.
The main question I have is do you have any ideas why I am not consistently receiving the response to the status request while it is polling each device? This seems to be the first issue I see (hopefully close to the root cause). I restarted the Pi and re-ran this several times.  Sometimes in the section where it is polling all devices in ALDB, I would receive the status for all 3, but most of the time I would only receive the status for the first one or the second.  I suspect there is some buffer that is filling up.  Any suggestions?

```
pi@hassbian:~ $ insteonplm_monitor --device /dev/ttyUSB0
INFO:insteonplm.tools:Connecting to Insteon PLM at /dev/ttyUSB0
INFO:insteonplm.connection:Connecting to PLM on /dev/ttyUSB0
INFO:insteonplm.protocol:New callback <function console.<locals>.async_insteonplm_light_callback at 0xb62b0150> with {} (0 items already in list)
INFO:insteonplm.protocol:Connection established to PLM
INFO:insteonplm.protocol:Requesting PLM Info
INFO:insteonplm.protocol:Requesting First ALL-Link Record
INFO:insteonplm.protocol:PLM Info from 13.AC.33: category:03 subcat:15 firmware:92
INFO:insteonplm.protocol:ALL-Link Record for 3E.94.10: flags:e2 group:01 data:01/0e/43
INFO:insteonplm.protocol:New INSTEON Device 3E.94.10: LampLinc Dual Band 2-pin (01:0e)
WARNING:insteonplm.tools:New Device: {'model': '2457D2', 'cat': 1, 'subcat': 14, 'product_key': None, 'address_hex': '3e9410', 'address': '3E.94.10', 'firmware': 67, 'capabilities': ['light', 'dimmable'], 'description': 'LampLinc Dual Band 2-pin'}
INFO:insteonplm.protocol:Requesting Next ALL-Link Record
INFO:insteonplm.protocol:ALL-Link Record for 3E.98.2D: flags:e2 group:01 data:01/0e/43
INFO:insteonplm.protocol:New INSTEON Device 3E.98.2D: LampLinc Dual Band 2-pin (01:0e)
WARNING:insteonplm.tools:New Device: {'model': '2457D2', 'cat': 1, 'subcat': 14, 'product_key': None, 'address_hex': '3e982d', 'address': '3E.98.2D', 'firmware': 67, 'capabilities': ['light', 'dimmable'], 'description': 'LampLinc Dual Band 2-pin'}
INFO:insteonplm.protocol:Requesting Next ALL-Link Record
INFO:insteonplm.protocol:ALL-Link Record for 01.89.44: flags:e2 group:01 data:01/07/37
INFO:insteonplm.protocol:New INSTEON Device 01.89.44: LampLinc Dimmer V2 2-pin (01:07)
WARNING:insteonplm.tools:New Device: {'model': '2856D2', 'cat': 1, 'subcat': 7, 'product_key': None, 'address_hex': '018944', 'address': '01.89.44', 'firmware': 55, 'capabilities': ['light', 'dimmable'], 'description': 'LampLinc Dimmer V2 2-pin'}
INFO:insteonplm.protocol:Requesting Next ALL-Link Record
INFO:insteonplm.protocol:ALL-Link database dump is complete
INFO:insteonplm.protocol:Polling all devices in ALDB
INFO:insteonplm.protocol:Requesting status for 01.89.44
INFO:insteonplm.protocol:Requesting status for 3E.94.10
INFO:insteonplm.protocol:Requesting status for 3E.98.2D
INFO:insteonplm.protocol:INSTEON device status 01.89.44 is at level 0x0
INFO:insteonplm.protocol:Device 018944.onlevel changed: None->0"
INFO:insteonplm.protocol:INSTEON device status 3E.94.10 is at level 0x0
INFO:insteonplm.protocol:Device 3e9410.onlevel changed: None->0"
INFO:insteonplm.tools:Jonathan, Are you ready to rumble?
INFO:insteonplm.tools:Relay Status
INFO:insteonplm.protocol:Requesting status for 3E.94.10
INFO:insteonplm.tools:Jonathan, Are you ready to rumble?
INFO:insteonplm.tools:Relay Status
INFO:insteonplm.protocol:Requesting status for 01.89.44
INFO:insteonplm.tools:Jonathan, Are you ready to rumble?
INFO:insteonplm.tools:Relay Status
INFO:insteonplm.protocol:Requesting status for 3E.98.2D
```

teharris1, 2018-04-05:
@jfbloom22 have you tried the updated version? It has been out about a month.

jfbloom22, 2018-04-19:
@teharris1 thanks for the follow up.  I sorta gave up on this and setup an Insteon Hub.   It also sometimes has trouble communicating with the devices.   I'll fire up the Raspberry and give this a try over the weekend.

jfbloom22, 2018-05-03:
@teharris1 thanks again for following up on this.  I did not have a chance to test this out.  It looks like I am the only one experiencing this issue, so I am closing it.

closed 2018-05-03 by jfbloom22