nugget/python-insteonplm #13: Status of lights and switches does not update after manual change

rdagger, 2017-06-09:
The Insteon devices all work great when controlled from within HA, but when I manually turn a device on or off at the wall switch, the HA status does not update.
Tested with the following:

- Fresh install of the latest version of Hassbian image on RPi2
- Insteon 2413U USB PLM
- Insteon 2477D SwitchLinc Dimmer Switch
- Insteon 2477S SwitchLinc On/Off Switch

I've made no changes to the yaml configuration file other than the following:
```
  customize:
    light.4xxxx4:
      friendly_name: Stairs Light
      icon: mdi:stairs
    light.4xxxxe:
      friendly_name: Entry Light
    light.4xxxx9:
      friendly_name: Hallway Light
      hidden: true
    light.4xxxx1:
      friendly_name: Porch Light
    light.3xxxxe:
      friendly_name: Yard Lights
      icon: mdi:spotlight-beam
    switch.3xxxxa:
      friendly_name: Front Fountain
      icon: mdi:water-pump
    switch.3xxxxc:
      friendly_name: Back Fountain
      icon: mdi:water-pump
    light.3xxxx8:
      friendly_name: Garage Sconces
      icon: mdi:garage

insteon_plm:
  port: /dev/ttyUSB0
  device_override:
    - address: 3xxxxa  # Front fountain
      platform: switch  # Make fountain a switch instead of a light
```

nugget, 2017-06-09:
Did you make sure to link the devices in both directions?   http://www.insteon.com/support-knowledgebase/2015/1/28/understanding-linking

rdagger, 2017-06-11:
Thanks that helped but I'm still having an issue with switches that are linked to other switches.
The stairs and hallway wall switches are linked together so either one turns on/off both.
HA can turn on/off the stairs which also hits the hallway.  HA also see manual switch presses of stairs.
However, HA cannot turn on/off the hallway and does not see manual presses of the hallway switch.
Here are the link tables for the stairs and hallway:

```
>>> plm.printLinkTable('Stairs')
000: AA 01 3x xx x6 FE 1C 01 : <PLM>
001: E2 01 4x xx x9 03 1C 01 : (Hallway)
002: AA 01 4x xx x9 FE 1C 01 : (Hallway)
003: E2 01 3x xx x6 03 1C 01 : <PLM>

>>> plm.printLinkTable('Hallway')
000: AA 01 3x xx x6 FE 1C 01 : <PLM>
001: AA 01 4x xx x4 FE 1C 01 : (Stairs)
002: E2 01 4x xx x4 03 1C 01 : (Stairs)
003: E2 01 3x xx x6 03 1C 01 : <PLM>

```

TD22057, 2017-09-02:
Could this be related to this comment in the openhab PLM docs about 3way switches: https://github.com/openhab/openhab1-addons/wiki/Insteon-PLM-Binding#3-way-switch-configurations-and-the-related-keyword
Looks like the code would need to understand this relationship and trigger a poll of the other device in order to get the updated status.

majortom64, 2018-01-31:
@rdagger I want to confirm that HA cannot control the hallway switch nor does it see state changes initiated by the hallway switch. If that is the case, it is not related to the issue above. If you cannot control the hallway switch directly from HA, the problem is most likely a communications issues or a problem with the link database on the switch itself.

How many Insteon devices do you have total in your system? How are you bridging the phases? How far is the hallway switch from the PLM? Have you watched the logs to see what messages (or error messages) are generated when you try to turn the switch on?

Once you have HA reliably controlling the switch and seeing its local button presses, you need to add two triggers based on the state change events for the two switches to keep them in sync.

rdagger, 2018-02-02:
@majortom64 I have 8 devices.  I'm using all dual band devices for bridging.  This is a very small home.  The hallway switch is about 12 feet from the PLM.  I don't see any errors in the logbook.

Stairs wall switch turns on both stairs and hallway lights (Stairs light status is updated in HA, but hallway light status is not updated in HA)

Hallway wall switch turns on both stairs and hallway lights (Hallway light status is updated in HA, but stairs light status is not updated in HA)

Stairs HA On command turns on both stairs and hallway (Hallway light status is not updated in HA)
Stairs HA Off command turns off both stairs and hallway (Hallway light status is not updated in HA)

Hallway HA On command does not work on either light.
Hallway HA Off command does not work on either light.

teharris1, 2018-03-23:
@rdagger you stated:

> Hallway HA On command does not work on either light.
Hallway HA Off command does not work on either light.

 did you ever get this working? Hopefully you have tried the latest version of HA and insteonplm. If it 
is not working in the latest HA it most likely is a signal or a  linking issue. 

Regarding you desire to have a change HA for Stairs to cascade to Halway, please note Insteon devices don't cascade messages to downstream devices. This is a design decision. There is a plan to address this via Home Assistant but it is further out in the roadmap. In the mean time, what you can do in HA is set up scenes. You need two `on` scense and two `off` scenes. Scene 1 `on` will notice when the hallway switch turns on and then turn on the stairs switch. Scene 2 `on` looks the change in the stairs switch and turns on the hallway switch. I have the same basic set up at my house. The `off` scenes are the same scenes just for the `off` command.

@majortom64 has put a request for this to be automatic #45 As I said, this in in the roadmap but it is not an easy issue to solve and has lots of technical challenges that may make it less than reliable.

rdagger, 2018-04-02:
@teharris1 I'm using the latest updated version of Hassbian.  Unfortunately, the problem persists.  I dumped the link tables from the PLM and they look correct.  I don't think it is a signal issue because everything is within 12 feet of each other.  The wall switches work as expected outside of HA.  Thanks for the info regarding scenes.

teharris1, 2018-04-03:
Can you post the link tables? Also, if you can put the system in debug mode (let me know if you need help) send me any lines that contain 'Processing message". You can use the following command at the OS level..Go to your config directory and type:

`cat home-assistant.log | grep -E "Processing message"`

rdagger, 2018-04-03:
Here are the link tables, thanks:
>>> plm.printLinkTable('Stairs')
000: AA 01 3x xx x6 FE 1C 01 : <PLM>
001: E2 01 4x xx x9 03 1C 01 : (Hallway)
002: AA 01 4x xx x9 FE 1C 01 : (Hallway)
003: E2 01 3x xx x6 03 1C 01 : <PLM>

>>> plm.printLinkTable('Hallway')
000: AA 01 3x xx x6 FE 1C 01 : <PLM>
001: AA 01 4x xx x4 FE 1C 01 : (Stairs)
002: E2 01 4x xx x4 03 1C 01 : (Stairs)
003: E2 01 3x xx x6 03 1C 01 : <PLM>

rdagger, 2018-04-03:
@teharris1 Here's the debug output:

pi@hassbian:/home/homeassistant/.homeassistant $ cat home-assistant.log | grep -E "Processing message"

2018-04-03 01:53:01 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x60, 'address': 39.XX.B6, 'category': 0x03, 'subcategory': 0x15, 'firmware': 0x9e, 'acknak': 0x06}
2018-04-03 01:53:02 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x69, 'acknak': 0x06}
2018-04-03 01:53:02 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x01, 'address': 40.XX.0E, 'linkdata1': 0x01, 'linkdata2': 0x20, 'linkdata3': 0x45}
2018-04-03 01:53:04 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:04 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x01, 'address': 40.XX.F4, 'linkdata1': 0x01, 'linkdata2': 0x20, 'linkdata3': 0x45}
2018-04-03 01:53:06 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:06 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x01, 'address': 40.XX.79, 'linkdata1': 0x01, 'linkdata2': 0x20, 'linkdata3': 0x45}
2018-04-03 01:53:07 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:07 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x01, 'address': 40.XX.A1, 'linkdata1': 0x01, 'linkdata2': 0x20, 'linkdata3': 0x45}
2018-04-03 01:53:09 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:09 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x01, 'address': 3D.XX.CE, 'linkdata1': 0x02, 'linkdata2': 0x2a, 'linkdata3': 0x45}
2018-04-03 01:53:11 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:11 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x01, 'address': 3D.XX.1C, 'linkdata1': 0x02, 'linkdata2': 0x39, 'linkdata3': 0x44}
2018-04-03 01:53:13 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:13 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x01, 'address': 3D.XX.7A, 'linkdata1': 0x02, 'linkdata2': 0x2a, 'linkdata3': 0x45}
2018-04-03 01:53:14 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:14 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x01, 'address': 3D.XX.C8, 'linkdata1': 0x02, 'linkdata2': 0x2a, 'linkdata3': 0x45}
2018-04-03 01:53:16 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:16 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 40.XX.F4, 'linkdata1': 0x00, 'linkdata2': 0x00, 'linkdata3': 0x00}
2018-04-03 01:53:17 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:17 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 40.XX.0E, 'linkdata1': 0x00, 'linkdata2': 0x00, 'linkdata3': 0x00}
2018-04-03 01:53:19 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:19 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 3D.XX.C8, 'linkdata1': 0x00, 'linkdata2': 0x00, 'linkdata3': 0x00}
2018-04-03 01:53:20 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:20 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 3D.XX.7A, 'linkdata1': 0x00, 'linkdata2': 0x00, 'linkdata3': 0x00}
2018-04-03 01:53:21 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:21 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 40.XX.79, 'linkdata1': 0x00, 'linkdata2': 0x00, 'linkdata3': 0x00}
2018-04-03 01:53:23 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:23 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 40.XX.A1, 'linkdata1': 0x00, 'linkdata2': 0x00, 'linkdata3': 0x00}
2018-04-03 01:53:24 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:24 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 3D.XX.CE, 'linkdata1': 0x00, 'linkdata2': 0x00, 'linkdata3': 0x00}
2018-04-03 01:53:25 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x06}
2018-04-03 01:53:25 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 3D.XX.1C, 'linkdata1': 0x00, 'linkdata2': 0x00, 'linkdata3': 0x00}
2018-04-03 01:53:27 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x6a, 'acknak': 0x15}
[... cut ...]
2018-04-03 02:05:42 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x50, 'address': 40.XX.79, 'target': 00.00.01, 'flags': 0xcf, 'cmd1': 0x13, 'cmd2': 0x00}
2018-04-03 02:05:42 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x50, 'address': 40.XX.79, 'target': 39.XX.B6, 'flags': 0x40, 'cmd1': 0x13, 'cmd2': 0x01}
2018-04-03 02:05:42 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x50, 'address': 40.XX.79, 'target': 13.02.01, 'flags': 0xcb, 'cmd1': 0x06, 'cmd2': 0x00}

teharris1, 2018-04-05:
The two devices showing manual updates are 40.XX.F4 and 40.XX.79. They both appear to be 2477D dimmer switches, is that correct? Which one is not working? Which one is hall and which one is stairs?

rdagger, 2018-04-12:
Yes 2477D.
.79 is the hallway (not working).
.F4 is the stairs.

teharris1, 2018-04-12:
The two important lines are:
`{'code': 0x50, 'address': 40.XX.F4, 'target': 00.00.01, 'flags': 0xcf, 'cmd1': 0x11, 'cmd2': 0x00}
 {'code': 0x50, 'address': 40.XX.79, 'target': 00.00.01, 'flags': 0xcf, 'cmd1': 0x11, 'cmd2': 0x00}`
Since F4 is working, 79 should also be working. It is very odd that it is not. These two messages tell me the link tables are fine and the insteonplm module is receiving the commands correctly. Since F4 is working and they are both the same device type, I have to assume the command is being processed correctly inside insteonplm.

Based on all the information you have sent, I would expect the following behavior:
1. Manually turn on/off F4 will turn on/off 79
2. Manually turn on/off F4 will tell HA to display F4 as on/off but not change 79
3. HA turn on/off F4 will turn the device on/off and display on HA as on/off but not change 79
4. Manually turn on/off 79 will turn on/off F4
5. Manually turn on/off 79 will tell HA to display 79 as on/off but not change F4
6. HA turn on/off 79 will turn the device on/off and display on HA as on/off but not change F4

Based on your description it seems 1, 2, 3 and 4 all are correct but 5 and 6 are not. Is that right?

rdagger, 2018-04-12:
Not quite.
1.  Correct
2.  Correct
3.  HA turn on/off F4 controls both devices correctly, but does not update HA display for 79
4.  Correct
5.  Correct
6.  HA turn on/off 79 does not turn on/off either light.

teharris1, 2018-04-12:
OK, from your description, can I assume F4 is attached to the load and 79 is not? That may explain what you are seeing. In other words, electronically turning on/off 79 will not change the load (i.e. the physical light) but electronically changing F4 will.

rdagger, 2018-04-12:
Yes, F4 and 79 were originally a 3-way switch.  I believe F4 is the one connected to the load.
I just noticed something about the LED indicator bars on the wall switches that corroborates your conclusion:
HA turn on/off F4 only changes the LED indicator on F4 despite effectively operating both lights.
HA turn on/off 79 only changes the LED indicator on 79.

teharris1, 2018-04-12:
Correct. That is the way INSTEON works. Electronically turning on F4 does not cascade the link to 79 and vice versa. I am working on a feature to have HA cascade the links but there are some issues with that. For now, you can use HA to do that with scenes and automation. Based on this, the system appears to be working as designed (even if it is not what is expected : ).  Can we close this issue?

rdagger, 2018-04-13:
Yes. Thanks for your help.

closed 2018-04-13 by teharris1