nugget/python-insteonplm #16: Add Support for FanLinc

majortom64, 2017-07-17:
Currently, there is no support for the FanLinc (it is basically two devices in a single package, an inline lamp module and an inline fan controller). 

According to http://madreporite.com/insteon/Insteon_device_list.htm it has a devcat of 0x01 and a sub cat of 0x2e and is an i2CS device.

nugget, 2017-07-30:
I don't have a way to test, but I believe it should be supported in version 0.7.5 (release imminent).  If you could test and let me know that would be great.

masterdka, 2017-12-07:
I have the fanlinc and I can see it found the light control portion of the module but did not see any options for the fan control.  Still new to the insteon devices with home assistant so I could be missing something.  I would be glad to help in any testing or feedback o get the fan portion working.

teharris1, 2017-12-09:
In order to support FanLink and a few other devices in Home Assistant we need to be able to control Group/Button states in the devices. This is done using the Light ON command (cmd1 0x11) message but using an ED message where D1 is the group/button to control. In the case of the FanLinc it is Group 2. This is a meaningful enhancement but if we  make it, the thermostat and other controls also fall in line. 

However, there is a key question that needs to be decided. Do you present the fan control as a separate device to Home Assistant or do you enhance Home Assistant to allow multiple controls in a single device? I don't know the HA side too well so others may have a better idea but it appears to me the easier thing is to present two devices to HA. Thoughts?

majortom64, 2017-12-10:
Given that there is no connection between the fan controller and the light controller other than their co-location in the single piece of hardware, I think presenting two devices makes more sense.

teharris1, 2017-12-22:
@majortom64 I added FanLinc support to my fork.  download [insteonplm](https://github.com/teharris1/python-insteonplm) and [home-assistant](https://github.com/teharris1/home-assistant) and give it a test. You don't need all of HA, just the four insteon_plm.py files and overwrite the current ones. This is a major refactoring of the insteonplm code so you may find other bugs. Please feel free to report any and all that you find. Thanks.

majortom64, 2018-01-28:
Both parts of the FanLinc work perfectly. Thanks! The instructions I found for which files I needed to change did not include copying components/fan/insteon_plm.py, so that took me a moment to figure out (finally starting to understand Home Assistant's structure).

Again, thanks for the work!

teharris1, 2018-01-28:
Awesome. Thanks for the feedback. I am making more changes so I may need help testing again soon if you don't mind.  Yes, I forget to update the instructions, sorry. Glad you figured it out.

teharris1, 2018-03-23:
@majortom64 have you tried this yet? Love to get your feedback. Thanks.

teharris1, 2018-04-07:
@majortom64 closing this per the note in #43

closed 2018-04-07 by teharris1