nugget/python-insteonplm #17: InsteonPLM Monitor crashes on launch.

majortom64, 2017-07-17:
Brand new installation (pip3 install insteonplm on python 3.5.3 on macOS Sierra using the Python.org macOS installer) and I get this when launches: 

INFO:insteonplm.tools:Are you ready to rumble?
INFO:insteonplm.tools:Relay Status
ERROR:asyncio:Task exception was never retrieved
future: <Task finished coro=<console() done, defined at /Users/example/Developer/InsteonPLM/lib/python3.5/site-packages/insteonplm/tools.py:11> exception=KeyError()>
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/Users/example/Developer/InsteonPLM/lib/python3.5/site-packages/insteonplm/tools.py", line 134, in console
    plm.status_request(addr)
  File "/Users/example/Developer/InsteonPLM/lib/python3.5/site-packages/insteonplm/protocol.py", line 765, in status_request
    device = self.devices[address.hex]
  File "/Users/example/Developer/InsteonPLM/lib/python3.5/site-packages/insteonplm/protocol.py", line 40, in __getitem__
    raise KeyError
KeyError
^CTraceback (most recent call last):
  File "./bin/insteonplm_monitor", line 11, in <module>
    load_entry_point('insteonplm==0.7.4', 'console_scripts', 'insteonplm_monitor')()
  File "/Users/example/Developer/InsteonPLM/lib/python3.5/site-packages/insteonplm/tools.py", line 145, in monitor
    loop.run_forever()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
    event_list = self._selector.select(timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/selectors.py", line 577, in select
    kev_list = self._kqueue.control(None, max_ev, timeout)
KeyboardInterrupt

There are 2 unknown devices in the system - a FanLinc and a 2334 KeypadLinc (in case that matters).

Thanks in advance. Happy to provide any additional information you need.

dakoellis, 2017-08-29:
Were you able to get anywhere with this?  I'm having the same issue and have 2 unknown devices as well:  Thermostat and ToggleLinc Relay

majortom64, 2018-01-28:
Using the refactored code from @teharris1, I now have this working.

teharris1, 2018-02-08:
@majortom64 closing this issue.

closed 2018-02-08 by teharris1