nugget/python-insteonplm #20: Add support for IP PLM for old Hub

djfjeff, 2017-08-11:
Hi,

Thank you for this component. I would like to know if it would be possible to add IP PLM support for the old Hub (2242). From what I understand (and checking the openhab component), it's the same protocol but over network instead of USB/RS232.

The reason is that the Insteon Local component in Home Assistant work by spamming the Hub to refresh it's state which is really not ideal. Also, the developer seems to have abandoned the project.

Thanks!

nugget, 2017-08-14:
I don't have a 2242 hub to develop against, but presumably this would not be too difficult to get going.  I looked at pricing for the older hubs and they're a bit more than I would want to spend to add support for a discontinued product that will only become more difficult to source in the future.

If someone has one and wants to take a shot at getting Pyserial to talk to a network socket I'd happily merge in the code.

teharris1, 2018-06-21:
@djfjeff I know this was requested a long time ago but if you are still interested, the code now supports the Hub. I am not sure it supports the version you are on but I would welcome feedback. If you are interested I will give you the download information.

teharris1, 2018-07-29:
Here is the code to download to test the hub:
[insteonplm](https://github.com/teharris1/python-insteonplm/archive/hub.tar.gz}
[Home Assistant](https://github.com/teharris1/home-assistant/archive/hub.tar.gz)
You need to uninstall the current `insteonplm` and install from the downloaded file, then you need the following files for Home Assistant:
- homeassistant/components/insteon_plm/*
- homeassistant/components/binary_sensor/insteon_plm.py
- homeassistant/components/sensor/insteon_plm.py
- homeassistant/components/light/insteon_plm.py
- homeassistant/components/switch/insteon_plm.py
- homeassistant/components/fan/insteon_plm.py
Replace the above files in your current version of HA with the ones listed above.

teharris1, 2018-07-30:
BTW, here is what a sample config looks like in `configuration.yaml`:
```
insteon_plm:
  host: xxx.xxx.xxx.xxx
  ip_port: 25105
  username: !secret hub_username
  password: !secret hub_password
  device_override:
    - address: 1a2b3c
      cat: 0x00
      subcat: 0x1b
    - address: 4d5e6f
      cat: 0x00
      subcat: 0x1c
```

Where xxx.xxx.xxx.xxx is the IP address of the hub. `ip_port` is optional and defaults to 25105.

TLMNicolas, 2018-09-03:
Ah, makes sense. I was hoping it wasnt in the latest, and thats why it wasnt working for me.
Ill make a new issue and put debugging info in there then and see if maybe I can figure out whats going on.

On a side note, how does the code determine if to talk to the 2242 or the 2245 hub?
Does it use the same API ?

Thanks for the terrific work by the way.

teharris1, 2018-09-30:
@djfjeff @TLMNicolas @wz2b Can we close this issue? I am confident that the 2242 hub is supported based on feedback from others but want to confirm before closing.

closed 2018-10-01 by teharris1