nugget/python-insteonplm #21: InsteonPLM gets stuck waiting for response.

Grant10k, 2017-08-13:
When it boots up, InsteonPLM scans all the links in the PLM and requests all their initial states, but at some point along the way either some device fails to respond or it doesn't understand the response and it gets stuck waiting forever. Any new command just gets added to the end of the queue. For example:

```
2017-08-13 02:28:20 DEBUG (MainThread) [insteonplm.protocol] turn_on '1d1f83' 2456D3
2017-08-13 02:28:20 DEBUG (MainThread) [insteonplm.protocol] Still waiting on last_command.
2017-08-13 02:28:20 DEBUG (MainThread) [insteonplm.protocol] Adding command to queue: 02621d1f830011ff
```

Any incoming data the PLM gets will report that there are a some command sitting in the queue.

```
2017-08-13 02:28:36 DEBUG (MainThread) [insteonplm.protocol] Received 5 bytes from PLM: b'76ac411101'
2017-08-13 02:28:36 DEBUG (MainThread) [insteonplm.protocol] Found a code 0x62 message which returns 9 bytes
2017-08-13 02:28:36 DEBUG (MainThread) [insteonplm.protocol] Wait for ACK/NAK on sent: b'02621d1f83001300' expecting rsize of 9
2017-08-13 02:28:36 DEBUG (MainThread) [insteonplm.protocol] Send queue contains 20 items
```

To see what the command was, I added a line to _send_hex() which outputs
```
2017-08-13 03:09:10 DEBUG (MainThread) [insteonplm.protocol] Still waiting on last_command.
2017-08-13 03:09:10 DEBUG (MainThread) [insteonplm.protocol] Which is :b'\x02j'
```
0x02j is 026A which is "Thermostat Get Zone Info", but I don't have a thermostat and Home Assistant doesn't list one (so InsteonPLM knows I don't either) so I'm thinking that's either not really the command being sent or that's not really the command to get thermostat zone info.

Grant10k, 2017-08-13:
I was wrong, `026A` is "Get next All Link Record". 

I'm thinking now that it finishes getting the links, and is then requesting one more, or it's getting some unusual data and then choking. This is the soonest point where it seems to stop getting the response it's expecting. 
```
2017-08-13 03:47:06 INFO (MainThread) [insteonplm.protocol] Requesting Next ALL-Link Record
2017-08-13 03:47:06 DEBUG (MainThread) [insteonplm.protocol] Sending 2 byte message: b'026a'
2017-08-13 03:47:06 DEBUG (MainThread) [insteonplm.protocol] Waiting for {'code': 87} timeout in 1 seconds
2017-08-13 03:47:06 DEBUG (MainThread) [insteonplm.protocol] Received 1 bytes from PLM: b'15'
2017-08-13 03:47:06 DEBUG (MainThread) [insteonplm.protocol] Send queue contains 0 items
```

Grant10k, 2017-08-18:
I can't figure out where there's a loop or something called so I can't check for old queue items, but I found something that seems to work. 

In `_send_hex()` if you write down the time the command was issued, and check it back in `_process_queue()` you can throw out the old command if it's still being waited on. 
```python
def _process_queue(self):
        self.log.debug('Send queue contains %d items', len(self._send_queue))
        if time.time() - self._wait_time > 1:
          self.log.debug('Tossing old command')
          self._last_command = None
          self._wait_for = {}
          self._buffer = bytearray()

        if self._clear_to_send() is True:
            command, wait_for = self._send_queue[0]
            self._send_hex(command, wait_for=wait_for)
            self._send_queue.remove([command, wait_for])
```
The idea is to reset everything so `_clear_to_send()` returns true and treats the old command like it's finished. I started at about 3 seconds, but had to reduce it down to 1 for anything to happen. I think it's because it has to come before the timeout function, which just seems to freeze everything up.

teharris1, 2017-09-01:
In general I think the wait_for concept needs to be rethought. It assumes the sent message and the received message are alsways going to be syncronous.  This is too often not the case. 

I have noticed in my system that it chokes when an unexpected message response appears in the queue. For example, if, during ALL-Link a motion sensor is triggered a message hits the message response queue but the system is wating for an All-Link ACK/NAK message which it does not find. The offending code is in 
```
def _wait_for_last_command(self):
        sentmessage = self._last_command
        rsize = self._rsize(sentmessage)
        self.log.debug('Wait for ACK/NAK on sent: %s expecting rsize of %d',
                       binascii.hexlify(sentmessage), rsize)
        if self._buffer.find(sentmessage) == 0:
```
which assumes that the last message sent is providing an ACK/NAK message responese and will be the next message peeled from the queue. .
 The simple solutin is to add an ELSE statement to this IF statement with the following:
```
        if self._buffer.find(sentmessage) == 0:
            ...
         else:
            self.log.debug('Could not find last message %s in buffer %s.', binascii.hexlify(sentmessage), binascii.hexlify(self._buffer))
            self._wait_for_recognized_message()    
```  

If you look at the log produced you can see that the "message" being searched for is two or three messages back so it is not found as the first message and the if statement failes. This causes a loop that never resolves. 

The message callback and insteon callback model works very well and I think it can replace any wait_for concept. All we need to do is add a sent message queue and match ACK/NAK messages against that queue when they come in to confirm message status.

Grant10k, 2017-09-01:
In addition, any command that needs a response should have a working timeout. If the network is busy or flaky I've seen a light that turns on a good 3 seconds after the switch is hit. 

I don't know what the maximum wait time should be, but if a command hasn't received a response in 5 seconds, it's probably never going to.

teharris1, 2018-03-23:
@Grant10k Have you tried the latest version of Home Assistant and insteonplm? I believe the issue you raised above is addressed. If not please let me know.

Grant10k, 2018-03-23:
The latest version of Home Assistant requires a python that's 0.0.1 higher than what ships with Ubuntu so I'm going to wait for that mess to sort itself out and try again later.

Grant10k, 2018-03-30:
Got it working, though I think the choice of minimum python version was an oversight. 

The insteon stuff is working with caveats. I removed several switches/controllers and replaced them with non-insteon devices so I don't have the exact same setup as when I last tested it. Is it fixed? Or was it just choking on an IOLinc that's now out of the picture? Can't say for sure, but it's working now.

It's working, but it's like...anecdotal evidence instead of scientific evidence.

closed 2018-03-30 by Grant10k