nugget/python-insteonplm #22: Add support for 2448A7 USB Stick

nugget, 2017-08-16:
Looking at the spec it appears that the package should be able to work with the single-band (wireless only) USB stick interface.

nugget, 2017-08-16:
Initial experiment seems very promising.  The code happily talks to the device just like it should.  There's no hardware button on the USB stick to initiate linking, though, which means the module will need to add support for software-initiated linking.

lizaoreo, 2017-09-19:
How are you adding devices to your 2448A?  It doesn't have a set button, so do you have to add it to HouseLink, then use HouseLink to add devices to it's known links or whatever, and then connect to HomeAssistant?

teharris1, 2018-05-06:
This capability had been added to insteonplm via the insteonplm_interactive command line too long and as of Home Assistant 0.69.0b0.

closed 2018-05-06 by teharris1