nugget/python-insteonplm #23: Documentation

wsmorgan, 2017-08-16:
If you don't mind me pestering you occasionally for additional info and clarification I can work on getting documentation setup. The biggest thing I would need to get started is a working code example.

teharris1, 2018-03-23:
@wsmorgan Is the offer still open? love to get some help with documentation.

wsmorgan, 2018-03-23:
@teharris1 I'm still willing to help, I don't have as much free time right now as I did but I can still put some time into it.

teharris1, 2018-03-30:
How can I get you started? Are you thinking user documentation or developer documentation? 

The best examples of running code are in Home Assistant and the tools.py module.

altersis, 2018-09-20:
Hi there, greetings from Canada! I'm also willing to help in the development/testing/documentation of insteonplm. Been an Insteon user for many years, using an ISY994. Unfortunately, the ISY development seems to have slowed down to snail's pace. Seeing how much progress has been made in HASS, I would like to help make this module a true option for people like me to move from the ISY. I've been testing it for the past couple of days, and have found that there is still a lot of work to do, specially on the testing of devices to ensure they work. I have wide variety of devices, like: Keypad lincs, IRLinc Receiver, Leak sensors, the old and new motion sensors, open/close sensors, IOLincs,  Wall Swtiches/Dimmers, Bulbs, mini remotes, etc.

altersis, 2018-09-20:
So far, I have tested the new motion sensor, an IOLinc and a IRLinc Receiver. My experience is as follows:

**New motion sensor (2844-222)**: As per the docs, used the override option. HASS created three binary_sensor entities: motion, light (should not be binary) and battery (should not be binary, I know this physical device can do Motion, Temperature, Luminace, Battery Level, Flag for battery power, Flag for motion inhibited or not, and a Flag for a tamper switch). They HASS entities do not display the correct state, but I can see from DEBUG level logging that the PLM is getting traffic from the sensor. 
`2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] Starting: data_received
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] Received 33 bytes from PLM: b'0250415cb92130a64113010250415cb9130101cb06000250415cb9130101cb0600'
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] Finishing: data_received
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] Total buffer: b'0250415cb92130a64113010250415cb9130101cb06000250415cb9130101cb0600'
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] Total buffer: b'0250415cb9130101cb06000250415cb9130101cb0600'
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] Total buffer: b'0250415cb9130101cb0600'
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] Buffer too short to have a message
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] Messages in queue: 3
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] RX: {'code': 0x50, 'address': 41.5C.B9, 'target': 21.30.A6, 'flags': 0x41, 'cmd1': 0x13, 'cmd2': 0x01}
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] RX: {'code': 0x50, 'address': 41.5C.B9, 'target': 13.01.01, 'flags': 0xcb, 'cmd1': 0x06, 'cmd2': 0x00}
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] RX: {'code': 0x50, 'address': 41.5C.B9, 'target': 13.01.01, 'flags': 0xcb, 'cmd1': 0x06, 'cmd2': 0x00}
2018-09-20 11:58:26 DEBUG (MainThread) [insteonplm.plm] Last item in self._recv_queue reached.`
Is there anything you recommend to try to get this sensor to update properly?
Thanks in advance!

teharris1, 2018-09-24:
@altersis Great to have you on board. Love to get the Motion Sensor II working better. Unfortunately there have never been any developer notes released for this device.  As for the comments above:
>As per the docs, used the override option. HASS created three binary_sensor entities: motion, light (should not be binary) and battery (should not be binary

The battery and light sensors of the 2842 (which is what is currently supported) only show battery low or light level reached and therefore are really just on/off sensors. Hence the reason they are binary. If the 2844 has light and battery levels that report a variable level then we can make that work no problem. I would just need to understand the device better. 

Also, to your comment above:
> I know this physical device can do Motion, Temperature, Luminace, Battery Level,

The 2842 only has Motion, Luminance and Battery Low. These are groups 0x01, 0x02 and 0x03 respectively. If you send me any lines that look like this from your homeassistant.log file that would help confirm the link groups:
`2018-09-08 00:04:48 DEBUG (MainThread) [insteonplm.plm] RX: {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x00, 'address': 27.C3.87, 'linkdata1': 0x01, 'linkdata2': 0x38, 'linkdata3': 0x42}`

Where `address` is the address of your Motion Sensor II. Also, if you can let the HASS system run in debug mode for about 2 days and run the following line at the command line that would help:
`cat homeassistant.log | grep -E '41.5C.B9'`

Sending that output will help me see what commands are coming from the device.

Finally, the idea of setting configuration of any device (i.e. ramp rate etc.) is the next big thing on this development pipe line. I will need a lot of help with that.

teharris1, 2018-09-24:
By the way, in the wiki, I setup the following page: [Tested Devices](https://github.com/nugget/python-insteonplm/wiki/Tested-Devices)

altersis, 2018-09-24:
Hi @teharris1, thanks for the info. I see. It's unfortunate, but the Insteon guys have not been very good at spreading knowledge about the motion sensor 2. I approached Insteon directly this morning an asked them to allow me to get access to the tech documents. Keep your finger crossed to see if we can get that. I linked one of the Motion Sensors to an ISY994, and found that the motion detector has more to offer: As you say it has Motion, Battery and Light binary sensors, and you can configure it to get actual numeric temperature in F, a boolean that indicates if it is battery or power supply powered plus Luminance and Battery levels expressed as a %. Strange for a Luminance sensor to do this, but that's what you an see in the ISY... unfortunately I could not find a place where it gets clarified what is the Luminance max numeric value to translate that to Lumens or something similar.  Take a look at this screen, hope it clarifies what I'm saying. 
[image]
I'll find time tomorrow to set one of the motion sensors again on my test Hass instance, capture that data for you and report back. Just noticed that I did not send you any comments on the IOLinc and IRLinc, but I guess it's better if we concentrate on just one device at a time. What do you think?

indec, 2019-07-11:
Hello! I just fired up Home Assistant on my Raspberry Pi for the first time, and my USB PLM just landed. I'm happy to write up a "new user guide" because to be honest I'm having a hard time with the first steps! If anyone's available for occasional n00b-question-answering, I'd gladly pave the way for those who come after me.

creeble, 2019-07-12:
@indec:
I'm in the same boat as you, I think.  Installed Hassbian on my Pi, installed what I believe is python-insteonplm (have a /usr/local/bin/insteonplm_monitor), have an Insteon 2413U PLM... and have no clue what to do next.

Let me know if you find any hints...

teharris1, 2023-02-10:
Closing this issue as the `insteonplm` module is no longer maintained. Please use the `pyinsteon` module instead.

closed 2023-02-10 by teharris1