nugget/python-insteonplm #25: Multiple Capabilities

teharris1, 2017-08-20:
I have multiple older devices that do not respond to 0x03 0x00 commands so they always come back as 0x01 0x00 devices (i.e. lights). I see that I can override this but with only one capability whereas your platform allows for multiple capabilities.  My suggestion is to change homeassistant/components/insteonplm.py starting at line 82 to:
        if isinstance(device['platform'], list):
          plm.protocol.devices.add_override(
              device['address'], 'capabilities', device['platform'])
        else:
          plm.protocol.devices.add_override(
              device['address'], 'capabilities', [device['platform']])

This will enable the following in the config file:
insteon_plm:
  port: /dev/insteon
  device_override:
    - address: aabbcc
      platform: 
        - light
        - dimmable

By the way, I am happy to contribute to this project. I have over 30 different insteon devices of all types but I have never programmed in Python (c# guy) and have never contributed to an opensource project. So I am bound to make mistakes and don't want to break your very well done project.

teharris1, 2017-08-22:
I submitted a pull request to the insteon_plm component on home-assistant but after thinking about this more I am not sure this is the best approach. I am thinking about a couple of things and wondering if there is a design concept that is being worked already. 

1) Shouldn't the "platform" key be saved for "light" where "dimmable" is a capability. Considering more complex devices and capabilities like "battery" which may drive measuring battery charge as an example. I am thinking that ipdb array should perhaps be [<platform>, [capability 1, capability 2, ...]]. 
2) For devices with multiple properties (i.e. multi button switches) in one, you need to be able to address the additional items. Perhaps the best way to do this is to create multiple entities. To do this perhaps the best option is to create an array of capabilities in ipdb. For example:
[
    [<platform>, [capability 1, capability 2,]], 
    [<platform>, [capability 1, capability 2]]
]
This would create two entities with two different entity_id's.  A working example would be an outlet model 2663-222. This model has a top and bottom outlet so the ipdb entry would look like:

Product(0x02, 0x39, None, 'On/Off Outlet', '2663-222', [['switch', [ ]. ['switch', [ ]]]

So we have a two elements in an array. Each element has a <platform> ('switch' is a scalar) and an array of capabilities (in this case [] is an empty array for each outlet since there no specific additional capabilities).

lanrongwen, 2017-08-30:
While not a direct fix for the multiple capabilities, It should now allow you to specify a device and it's class (the insteon device type hex code you can see from inside the insteon app) and use the ipdb to look up the appropriate device.

lanrongwen, 2017-08-30:
`def monitor():
    """Wrapper to call console with a loop."""
    devicelist = (
        {
            "address": "3c4fc5",
            "cat": 0x05,
            "subcat": 0x0b,
            "firmware": 0x00
        },
        {
            "address": "43af9b",
            "cat": 0x02,
            "subcat": 0x1a,
            "firmware": 0x00
        }
    )
    log = logging.getLogger(__name__)
    loop = asyncio.get_event_loop()
    asyncio.async(console(loop, log, devicelist))
    loop.run_forever()`

sample from my test harness.

lanrongwen, 2017-08-30:
(that didn't format well :) )

teharris1, 2017-09-01:
Are you saying that in the "device_override" section you would now be able to specify the 'cat', 'subcat' and 'firmware' hex codes directly rather than rely on Auto Discovery? I think that is an excellent idea and better than the one I suggested.

lanrongwen, 2017-09-03:
That's the intent.  I was able to test it in the test harness and it worked pretty good for me.  I haven't finished my testing within home assistant yet unfortunately...  :(  I'm not as knowledgeable on how HA works, but from what I can see, it should just work that way.

TD22057, 2017-09-11:
@teharris1 I wonder if this is what's happening to me in issue #31.  I have 3 devices paired and the monitor reports them all as lamplinc's (only 1 of them is).

lanrongwen, 2017-11-16:
I have the USB stick adapter, and everything in there comes up as lamplinc by default.  I don't think the discovery call is working correctly.

lanrongwen, 2017-11-23:
So I looked at the code that connects HA to this project, and it doesn't pass in the cat,subcat,firmware overrides...  :(

teharris1, 2017-11-23:
@TD22057 and @lanrongwen, lamp link is what every device registers as because the All-Link record returns device 0x00. I am working on a fix but trying to refactor the message handle process at the same time.

closed 2018-01-05 by teharris1