nugget/python-insteonplm #27: KeyError - When sensor (Door is open/closed)

jkzgit, 2017-08-23:
Hi
Any idea why i get this error when door sensor is triggered - open or close.
Running latest HomeAssistant on Raspbian with Insteon PLM.

2017-08-22 22:40:07 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback PLM._parse_insteon_standard(address: 2D.B...\xcf\x11\x01'))
Traceback (most recent call last):
  File "/usr/lib/python3.4/asyncio/events.py", line 120, in _run
    self._callback(*self._args)
  File "/srv/homeassistant/lib/python3.4/site-packages/insteonplm/protocol.py", line 508, in _parse_insteon_standard
    device = self.devices[msg.address.hex]
  File "/srv/homeassistant/lib/python3.4/site-packages/insteonplm/protocol.py", line 40, in __getitem__
    raise KeyError
KeyError

2017-08-22 22:40:08 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback PLM._parse_insteon_standard(address: 2D.B...e3kE\x11\x01'))
Traceback (most recent call last):
  File "/usr/lib/python3.4/asyncio/events.py", line 120, in _run
    self._callback(*self._args)
  File "/srv/homeassistant/lib/python3.4/site-packages/insteonplm/protocol.py", line 508, in _parse_insteon_standard
    device = self.devices[msg.address.hex]
  File "/srv/homeassistant/lib/python3.4/site-packages/insteonplm/protocol.py", line 40, in __getitem__
    raise KeyError
KeyError

nugget, 2017-08-23:
That indicates that the door sensor is not actually linked with the PLM.  The KeyError indicates that the sensor's hexid is not found in the PLM's ALL-Link database.

jkzgit, 2017-08-23:
Thanks, here are additional logs when HA connects to PLM.
Does it still imply the sensor is not linked to PLM and do you have a sample log when devices are correctly linked to PLM?

2017-08-22 22:37:51 INFO (MainThread) [homeassistant.setup] Setting up insteon_plm
2017-08-22 22:37:51 INFO (MainThread) [homeassistant.components.insteon_plm] Looking for PLM on /dev/ttyUSB0
2017-08-22 22:37:51 INFO (MainThread) [insteonplm.connection] Connecting to PLM on /dev/ttyUSB0
2017-08-22 22:37:51 INFO (MainThread) [insteonplm.protocol] New callback <function async_setup.<locals>.async_plm_new_device at 0x6e92d270> with {} (0 items already in list)
2017-08-22 22:37:51 INFO (MainThread) [homeassistant.setup] Setup of domain insteon_plm took 0.5 seconds.
2017-08-22 22:37:51 INFO (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=insteon_plm>
2017-08-22 22:37:51 INFO (MainThread) [homeassistant.setup] Setting up config
2017-08-22 22:37:51 INFO (MainThread) [insteonplm.protocol] Connection established to PLM
2017-08-22 22:37:51 INFO (MainThread) [insteonplm.protocol] Requesting PLM Info
2017-08-22 22:37:51 INFO (MainThread) [insteonplm.protocol] Requesting First ALL-Link Record
2017-08-22 22:37:51 INFO (MainThread) [homeassistant.setup] Setting up logbook
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.loader] Loaded config.core from homeassistant.components.config.core
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=config.core>
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.loader] Loaded config.group from homeassistant.components.config.group
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=config.group>
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.loader] Loaded config.hassbian from homeassistant.components.config.hassbian
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.loader] Loaded config.automation from homeassistant.components.config.automation
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=config.automation>
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.core] Bus:Handling <Event service_registered[L]: service=log, domain=logbook>
2017-08-22 22:37:52 INFO (MainThread) [insteonplm.protocol] PLM Info from 33.E3.6B: category:03 subcat:15 firmware:9e
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.setup] Setup of domain config took 0.2 seconds.
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=config>
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.setup] Setup of domain logbook took 0.1 seconds.
2017-08-22 22:37:52 INFO (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=logbook>
**2017-08-22 22:37:52 INFO (MainThread) [insteonplm.protocol] ALL-Link Record for 3F.7F.70: flags:a2 group:01 data:00/00/00**
2017-08-22 22:37:52 INFO (MainThread) [insteonplm.protocol] Auto Discovering device '3f7f70'.
2017-08-22 22:37:52 INFO (MainThread) [insteonplm.protocol] Requesting Next ALL-Link Record
**2017-08-22 22:37:52 INFO (MainThread) [insteonplm.protocol] ALL-Link Record for 2D.BC.06: flags:a2 group:01 data:00/00/00**
2017-08-22 22:37:52 INFO (MainThread) [insteonplm.protocol] Auto Discovering device '2dbc06'.
2017-08-22 22:37:52 INFO (MainThread) [insteonplm.protocol] Requesting Next ALL-Link Record
2017-08-22 22:37:52 INFO (MainThread) [insteonplm.protocol] ALL-Link database dump is complete
2017-08-22 22:37:52 INFO (MainThread) [insteonplm.protocol] Polling all devices in ALDB

nugget, 2017-08-23:
Are you using a full dual-band PLM or the RF-only USB Stick style?

jkzgit, 2017-08-23:
USB Dual-Band Adapter Model: 2413U
http://www.insteon.com/powerlinc-modem-usb
Thx

teharris1, 2018-03-23:
@jkzgit Have you tried the latest version of Home Assistant and insteonplm? I believe the issue you have above is address. If not, please let me know. THanks

teharris1, 2018-04-12:
@jkzgit Any update here. Would like to get you squared away.

closed 2018-05-19 by teharris1