nugget/python-insteonplm #31: Help adding smoke bridge support

TD22057, 2017-09-10:
I picked up an [Insteon Smoke Bridge](http://www.insteon.com/smoke-bridge/) to link wireless smoke detectors into my Home Assistant.  The bridge is supported by the [OpenHab PLM binding](https://github.com/openhab/openhab1-addons/wiki/Insteon-PLM-Binding) using category 0x10 , subcat 0x0A, product key F00.00.22.  The developer docs for the smoke bridge [can be found here](http://cache.insteon.com/developer/2982-222dev-022014-en.pdf).  

I don't have any problem doing the work to get this integrated myself, but I was hoping for some pointers as to how to proceed.   I looked at ipdb.py and see where to add the line, but I don't understand the model capability list and what that maps to (or what allowed values are for that list).

Thanks - Ted

TD22057, 2017-09-11:
FYI - I tried plugging in the smoke bridge, linking it to the PLM modem (sensor->modem), and ran the monitor to see what would happen.  I have a single LampLinc and that's it.  It gets the hex address right but identifies it as a lamplinc.  I tried adding a binary sensor definition to ipdb.py (it's not really a binary sensor but I figured that's a good place to start) and it didn't change anything (since the cat/subcat are reported wrong).  Is there any chance it's mis-reading the ID code or having some issue and using the ID code from the other device?

The smoke bridge address is 44.A3.79.

```$ ./bin/insteonplm_monitor --device /dev/insteon 
INFO:insteonplm.tools:Connecting to Insteon PLM at /dev/insteon
INFO:insteonplm.connection:Connecting to PLM on /dev/insteon
INFO:insteonplm.protocol:New callback <function console.<locals>.async_insteonplm_light_callback at 0x7f0457d11bf8> with {} (0 items already in list)
INFO:insteonplm.protocol:Connection established to PLM
INFO:insteonplm.protocol:Requesting PLM Info
INFO:insteonplm.protocol:Requesting First ALL-Link Record
INFO:insteonplm.protocol:PLM Info from 44.85.11: category:03 subcat:15 firmware:9e
INFO:insteonplm.protocol:ALL-Link Record for 3A.29.84: flags:e2 group:01 data:01/0e/43
INFO:insteonplm.protocol:Auto Discovering device '3a2984'.
INFO:insteonplm.protocol:New INSTEON Device 3A.29.84: LampLinc Dual Band 2-pin (01:0e)
WARNING:insteonplm.tools:New Device: {'address_hex': '3a2984', 'model': '2457D2', 'address': '3A.29.84', 'capabilities': ['light', 'dimmable'], 'firmware': 67, 'description': 'LampLinc Dual Band 2-pin', 'subcat': 14, 'cat': 1, 'product_key': None}
INFO:insteonplm.protocol:Requesting Next ALL-Link Record
INFO:insteonplm.protocol:ALL-Link Record for 3A.29.84: flags:a2 group:01 data:01/0e/43
INFO:insteonplm.protocol:Device '3a2984' is already added manually.
INFO:insteonplm.protocol:Requesting Next ALL-Link Record
INFO:insteonplm.protocol:ALL-Link Record for 44.A3.79: flags:a2 group:01 data:01/0e/43
INFO:insteonplm.protocol:Auto Discovering device '44a379'.
INFO:insteonplm.protocol:New INSTEON Device 44.A3.79: LampLinc Dual Band 2-pin (01:0e)
WARNING:insteonplm.tools:New Device: {'address_hex': '44a379', 'model': '2457D2', 'address': '44.A3.79', 'capabilities': ['light', 'dimmable'], 'firmware': 67, 'description': 'LampLinc Dual Band 2-pin', 'subcat': 14, 'cat': 1, 'product_key': None}
INFO:insteonplm.protocol:Requesting Next ALL-Link Record
INFO:insteonplm.protocol:ALL-Link database dump is complete
INFO:insteonplm.protocol:Requesting product data for 44.A3.79
INFO:insteonplm.protocol:Requesting product data for 3A.29.84
INFO:insteonplm.protocol:Polling all devices in ALDB
INFO:insteonplm.protocol:Requesting status for 44.A3.79
INFO:insteonplm.protocol:Requesting status for 3A.29.84
INFO:insteonplm.protocol:INSTEON standard 44.A3.79->44.85.11: cmd1:03 cmd2:ff flags:a0
INFO:insteonplm.protocol:INSTEON standard 3A.29.84->44.85.11: cmd1:03 cmd2:00 flags:20
ERROR:asyncio:Exception in callback PLM._parse_insteon_standard(target: 44.85...lse}, code: 80)
handle: <Handle PLM._parse_insteon_standard(target: 44.85...lse}, code: 80)>
Traceback (most recent call last):
  File "/usr/lib/python3.4/asyncio/events.py", line 120, in _run
    self._callback(*self._args)
  File "/opt/homeassistant/lib/python3.4/site-packages/insteonplm/protocol.py", line 525, in _parse_insteon_standard
    self._parse_product_data_response(msg.address, msg.userdata)
  File "/opt/homeassistant/lib/python3.4/site-packages/insteonplm/protocol.py", line 640, in _parse_product_data_response
    category = userdata[4]
IndexError: bytearray index out of range
INFO:insteonplm.protocol:INSTEON device status 44.A3.79 is at level 0xff
INFO:insteonplm.protocol:Device 44a379.onlevel changed: None->255"
INFO:insteonplm.protocol:INSTEON device status 3A.29.84 is at level 0x0
INFO:insteonplm.protocol:Device 3a2984.onlevel changed: None->0"
```

TD22057, 2017-09-11:
Another data point (that I don't understand).  I added an Insteon motion sensor (2842-2222) and paired it to the PLM modem to see if that would show up.  Then I ran the monitor again and got the same output.  The ID of the motion sensor showed up but the cat and subcat were reported as a dual band lamplinc.  So I'm thinking there is some kind of problem in the software at this point.  Not sure what it could be - any ideas?

teharris1, 2017-12-24:
@TD22057 my fork should support the Smoke Bridge. It should show up as a sensor with the following codes:
0: All clear
1: Smoke detected
2: CO2 detected
3: Test detected
4: Unknown message detected
5: All Clear detected
6: Low Battery
7: Sensor malfunction
10: Heartbeat received (0x0A: occurs every 24 hrs to ensure connectivity)
You need to download the [insteonplm ](https://github.com/teharris1/python-insteonplm)and [home-assistant ](https://github.com/teharris1/home-assistant) code and install it.  I cannot test it directly because I don't own one but I feel pretty good about the code.

teharris1, 2018-03-23:
@TD22057 Per previous threads, happy to collaborate with you if you want. Let me know.

closed 2018-03-23 by teharris1