nugget/python-insteonplm #33: Test Cases

lanrongwen, 2017-11-22:
I was thinking, if we are going to go through the hassle of a refactor we can probably add a test framework.  Any suggestions on a good one for python?

teharris1, 2017-12-02:
I am going to suggest we use unittest or pytest. Unittest is built in so likely always going to be supported. Unittest appears to be more widely used. I have been using pytest lately and it is very, very easy to use but I haven't tried to mock asyncio. I have only been developing unit tests.

teharris1, 2018-02-25:
@lanrongwen I would like to revisit this. My understanding has increased a lot since you posted this and I am going to suggest we use a lot of the ideas from HA. I have a PR almost ready to go that will introduce TOX into Travis CI for a bunch of lower level tests plus the unit tests. Love to get input.

closed 2018-06-05 by teharris1