nugget/python-insteonplm #34: Refactoring design decisions

teharris1, 2017-12-09:
The original code base is very good but has some challenges to implement more complex devices such as Thermostats or FanLinc devices. The goal of the refactoring effort is to remove these limitations and create a community driven architecture that is maintainable and extensible. Here are some jumping off questions to consider:
1) Is this a generic Insteon management tool or an extension for Home Assistant? The current design objective states it is generic but there are a number of HA specific ideas built in. 
2) What is the role of the PLM class? Currently all messages and commands are managed and processed by PLM. How should we separate concerns and define other classes that make the handling of messages and commands easier?
3) Do we employ Duck typing to messages or are they separate classes by message type? there are pros and cons here with meaningful design implications. Duck typing feels more Pythonesque and would simplify the creation of messages. However a single class for all message types may create a beast of a class that requires a lot of context specific logic.
4) How do we accept commands from control modules like Home Assistant? Should these commands come into a command module similar to the way PLM does it today? An alternative is to have a control module call a device specific class such as DimmableLightingControl that processes the Light ON command. Pros and cons of each especially when considering device state management. 
5) How do we handle complex devices such as the FanLinc where the standard message controls the light dimmer but the fan control requires access to an extended message to control group 2. Do we create a separate device class, use duck typing or force the control module (aka Home Assistant) to map to the Insteon methods where all this module does is make the commands available without any consideration for the upstream use case? 
These are just a few topics. Feel free to add more and contribute.

lanrongwen, 2017-12-14:
I think part of the question is do commands ever get reused.  My understanding was within a class like thermostat, it does reuse some commands depending upon what mode it's in...  I'm not sure without a deeper study of the protocol though...

teharris1, 2017-12-14:
If you want to see a consolidated list of the cmd1, cmd2 set it is in this file:
[constants.py](https://github.com/teharris1/python-insteonplm/blob/refactor/insteonplm/constants.py)
they are listed as tuples. From the spreadsheet you can see what commands are applicable to each device category and this list just shows the full set of commands independent of applicable device.

In the constants.py file if you see 'cmd2':None that means cmd2 is an input that is used to set the value level, for example 'cmd1':0x11, 'cmd2':None indicates that the method needs to replace 'cmd2' with a value.

teharris1, 2017-12-18:
OK, so I now have a reasonable baseline of a refactored [insteonplm](https://github.com/teharris1/python-insteonplm). It is based largely on the design of @nugget  but with some differences:
1) The function of the PLM class is to load the ALDB and to get the Device ID data (was not easy to figure out but I now have it correctly identifying all of my devices except two which will be fixed soon). The other function of the PLM class is to send and receive message. In the future it will need to perform All-Link functions but that is on the TODO list.
2) PLM.devices is the list of linked devices. Devices know what messages and commands to send and receive and how to handle them. There is (or will be) a device for each Insteon devcat. If a subcat requires unique behavior it can be subclassed or logic can be built into the base class depending on how complex the behavior change it. See the SwitchedLightingControl_2663_222 class for an example. (This class represents exactly how the FanLinc would work too with the exception that the fanlinc requires dimmable controls)
3) Messages are just information containers. They have no real logic other than parsing and some bit flag handling. I could be convinced to go back to Duck typing of messages but it was easier for me to break them out for debugging.

As for the design decsions above, I have assumed this is an Insteon generic module, not a Home Assistant extention. But I have considered how to connect control software like HA. There is a callback function (StateChangeSignal) that can work at the device level or at the state level (I believe state level is better but not 100% sure yet). 

The code is not wired up to HA yet so if you want to test it edit the Tools.py file and run it in insteonplm_monitor.

teharris1, 2017-12-21:
I now have testable code! There are a few known issues:

1. If a message comes into the PLM during startup the process may hang
2. Some battery operated devices may not be recognized because the default has changed from just using the ALDB data for device identification to using the Get ID Record message. Battery operated devices do not respond to request messages until they first send a message. Devcat 0x10 devices (motion sensors, leak sensors, door sensors) should all work but others may not.
3. I have not tested device overrides so someone please do :)

You need to download [insteonplm](https://github.com/teharris1/python-insteonplm) and [home-assistant](https://github.com/teharris1/home-assistant). From Home Assistant, you really only need to replace the four insteon_plm.py files located in components/, components/light, components/switch and components/binary_sensors.

nugget, 2017-12-21:
I just want to say that I'm tremendously enthused about this!

teharris1, 2017-12-21:
Thank you. I am very glad to help.

Another known issue.  HA does not automatically update the status of the devices. Have to decide if PLM will always pole the devices on startup (what I am leaning toward) or to make it an HA function to pole the device. 

BTW, I just added FanLinc support. I don't have one so I cannot test it. I suspect the controls will work but the status for the fan may not (i.e. HA will always show the fan as off).

nugget, 2017-12-21:
Having Home Assistant poll the devices violates their coding standards.  You can't have a Home Assistant status check initiate any network or device activity.  They're super-paranoid about blocking the asyncio loop (which isn't a bad thing).  I agree that having the PLM poll all known devices on startup as a best-effort attempt to learn the state of the universe.

I'd also toyed with the idea of keeping a state file on disk to be loaded on startup which would be very useful for the normal use case where an application is restarted quickly to apply code updates or os reboots.  Could potentially disregard old/stale states if the state file included metadata including the timestamp of the update for that item.

teharris1, 2017-12-21:
Good to know. That explains why you kept the device state in the PLM devices list. I mimicked what you did but I was not sure why it was that way. Trying to add polling to PLM now. Fingers crossed. 

Additional known issue (need to find a way to track them so they are not all bundled into this thread). Sometimes PLM finds a device and attempts to register it with HA but HA does not display it. Wierd.

teharris1, 2017-12-21:
OK, polling on startup appears to be working.

teharris1, 2017-12-21:
The device missing issue is an insteonplm issue not a Home Assistant issue. It appears an Get ID Request ACK message is coming back but no GET ID response. I will add a retry method to resubmit the Get ID request. The issue is intermittent.

teharris1, 2017-12-22:
Here are is the current status:
1) Auto discovery is working mostly. Still some issues with devices responding and I just had an issue with devices showing up twice in HA. Help debugging would be appreciated. 
2) Polling is working well. 
3) On/Off outlet 2663-222 controls the top and bottom outlets but the bottom status is wrong when it changes. Polling works but if you turn it off/on via Home Assistant it does not show the correct status.
4) Love to have someone test FanLinc. I expect similar behavior as #3 
5) Trouble with motion sensor not triggering. 
6) Have not tested device overrides.
these are the known issues. Help identifying/fixing would be great. Or find more :)

teharris1, 2017-12-24:
On/Off outlet 2663-222 is now working. Motion sensor also now working (and any other binary sensor).  FanLinc should then also work better. Support for sensors and Smoke Bridge added as well. I don't own a FanLinc or a Smoke Bridge so anyone who can test please do. This addresses most of the known issues. Let the testing begin!

dakoellis, 2017-12-31:
@teharris1 I was wondering if this is supposed to work with the 2413S PLM, or only the 2413U and 2412S?  When I try running insteonplm_monitor, I get 

`ERROR:asyncio:Task exception was never retrieved
future: <Task finished coro=<console() done, defined at /usr/lib/python3.6/site-packages/insteonplm/tools.py:11> exception=KeyError()>
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/insteonplm/tools.py", line 134, in console
    plm.status_request(addr)
  File "/usr/lib/python3.6/site-packages/insteonplm/protocol.py", line 801, in status_request
    device = self.devices[address.hex]
  File "/usr/lib/python3.6/site-packages/insteonplm/protocol.py", line 40, in __getitem__
    raise KeyError
KeyError
`

and it hangs there.

teharris1, 2017-12-31:
It should work with the 2413S. The error message you have indicates there is an issue with your install. They likely cause is the fact that I did not up-version the forked code. Try uninstalling and re-installing. with `pip3 uninstall insteonplm -y`, then reinstall.

dakoellis, 2017-12-31:
Tried uninstalling and reinstalling, and now I'm getting a different error:  

`ERROR:asyncio:Task exception was never retrieved`
`future: <Task finished coro=<console() done, defined at /home/user/python-insteonplm/insteonplm/tools.py:11> exception=TypeError('add_device_callback() takes 2 positional arguments but 3 were given',)>`
`Traceback (most recent call last):`
  `File "/home/user/python-insteonplm/insteonplm/tools.py", line 49, in console
    conn.protocol.add_device_callback(async_insteonplm_light_callback, criteria)`
  `File "/home/user/python-insteonplm/insteonplm/plm.py", line 139, in add_device_callback
    self.devices.add_device_callback(callback, criteria)`
`TypeError: add_device_callback() takes 2 positional arguments but 3 were given`

Is this still a system issue?  I just have no idea where to go from this point but I did try reinstalling asyncio and pyserial with no change

teharris1, 2017-12-31:
Sorry, I fixed that code. Once I had wired it up to Home Assistant I stopped testing with insteonplm_monitor. It should work now. By the way, hopefully you have edited tools.py to reflect your specific devices rather than my devices :)

dakoellis, 2017-12-31:
Since you said you've stopped testing with insteonplm_monitor, I tried running it in home assistant instead by replacing the 4 files you mentioned a few days ago, and got the following error (haven't enabled the light or switch component yet):  

`2017-12-31 12:47:19 ERROR (MainThread) [homeassistant.loader] Circular dependency detected: insteon_plm -> insteon_plm`
`2017-12-31 12:47:19 ERROR (MainThread) [homeassistant.setup] Setup failed for insteon_plm: Unable to resolve component or dependencies.`

Is it possible I'm getting these errors because I'm running it in a docker container, or did I miss something else?

dakoellis, 2017-12-31:
Since you said you've stopped testing with insteonplm_monitor, I tried running it in home assistant instead by replacing the 4 files you mentioned a few days ago, and got the following error (haven't enabled the light or switch component yet):  

`2017-12-31 12:47:19 ERROR (MainThread) [homeassistant.loader] Circular dependency detected: insteon_plm -> insteon_plm`
`2017-12-31 12:47:19 ERROR (MainThread) [homeassistant.setup] Setup failed for insteon_plm: Unable to resolve component or dependencies.`

Is it possible I'm getting these errors because I'm running it in a docker container, or did I miss something else?

edit: I realized I would need to update the insteonplm files as well

teharris1, 2017-12-31:
@dakoellis, I run in docker as well. Did you get it to work?

dakoellis, 2017-12-31:
I didn't.  Not sure if I updated the files correctly but I copied the insteon_plm.py files into the respective spots from your home-assistant and copied the entire insteonplm directory in and pip3 installed it.  Then I did a bunch of other things I can't remember but still no dice

teharris1, 2017-12-31:
Keep in mind, inside Docker you need to uninstall insteonplm as above then install the downloaded version. Hope that helps.

dakoellis, 2018-01-01:
still getting an error.  Looks like the discovery isn't working for me:

`Error while setting up platform insteon_plm`
`Traceback (most recent call last):`
 ` File "/usr/src/app/homeassistant/helpers/entity_component.py", line 171, in _async_setup_platform`
`    SLOW_SETUP_MAX_WAIT, loop=self.hass.loop)`
`  File "/usr/local/lib/python3.6/asyncio/tasks.py", line 358, in wait_for
    return fut.result()`
`  File "/usr/local/lib/python3.6/asyncio/futures.py", line 245, in result
    raise self._exception`
`  File "/usr/local/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)`
  `File "/usr/local/lib/python3.6/asyncio/coroutines.py", line 210, in coro
    res = func(*args, **kw)`
`  File "/usr/src/app/homeassistant/components/switch/insteon_plm.py", line 25, in async_setup_platform`
   ` for device in discovery_info:`
`TypeError: 'NoneType' object is not iterable`

teharris1, 2018-01-02:
@dakoellis Can you  upload your log file and your configuration.yaml (remove any sensitive information please).

dakoellis, 2018-01-02:
Sure I can do it tomorrow morning. For the log, should I set it to debug?

teharris1, 2018-01-02:
Yes, debug. You can set the debugging to this: 
`logger:
  default: error
  logs:
    insteonplm: debug
    homeassistant.components.light.insteon_plm: debug
    homeassistant.components.switch.insteon_plm: debug
    homeassistant.components.binary_sensor.insteon_plm: debug
    homeassistant.components.sensor.insteon_plm: debug`
Also, please check that your configuration.yaml Insteon configuration just has this:
`insteon_plm:
  port: /dev/DEVICENAME`

dakoellis, 2018-01-02:
OK I have my insteon_plm set on port /dev/ttyS0, and here's the output from the log:

    ❯ cat home-assistant.log 
    2018-01-02 06:51:22 INFO (MainThread) [insteonplm.connection] Connecting to PLM on /dev/ttyS0
    2018-01-02 06:51:22 DEBUG (MainThread) [insteonplm.aldb] Added new callback <function async_setup.<locals>.async_plm_new_device at 0x7fd38f944400> 
    2018-01-02 06:51:22 INFO (MainThread) [insteonplm.plm] Connection established to PLM
    2018-01-02 06:51:22 DEBUG (MainThread) [insteonplm.plm] Starting: _get_plm_info
    2018-01-02 06:51:22 INFO (MainThread) [insteonplm.plm] Requesting PLM Info
    2018-01-02 06:51:22 DEBUG (MainThread) [insteonplm.plm] Starting: send_msg
    2018-01-02 06:51:23 DEBUG (MainThread) [insteonplm.plm] Sending 2 byte message: 0260
    2018-01-02 06:51:23 DEBUG (MainThread) [insteonplm.plm] Ending: send_msg
    2018-01-02 06:51:23 DEBUG (MainThread) [insteonplm.plm] Ending: _get_plm_info
    2018-01-02 06:51:23 DEBUG (MainThread) [insteonplm.plm] Starting: _load_all_link_database
    2018-01-02 06:51:23 DEBUG (MainThread) [insteonplm.plm] Starting: _get_first_all_link_record
    2018-01-02 06:51:23 INFO (MainThread) [insteonplm.plm] Requesting ALL-Link Records
    2018-01-02 06:51:23 DEBUG (MainThread) [insteonplm.plm] Starting: send_msg
    2018-01-02 06:51:23 DEBUG (MainThread) [insteonplm.plm] Sending 2 byte message: 0269
    2018-01-02 06:51:23 DEBUG (MainThread) [insteonplm.plm] Ending: send_msg
    2018-01-02 06:51:23 DEBUG (MainThread) [insteonplm.plm] Ending: _get_first_all_link_record
    2018-01-02 06:51:23 DEBUG (MainThread) [insteonplm.plm] Ending: _load_all_link_database
    2018-01-02 06:51:23 ERROR (MainThread) [homeassistant.components.light] Error while setting up platform insteon_plm
    Traceback (most recent call last):
      File "/usr/src/app/homeassistant/helpers/entity_component.py", line 171, in _async_setup_platform
        SLOW_SETUP_MAX_WAIT, loop=self.hass.loop)
      File "/usr/local/lib/python3.6/asyncio/tasks.py", line 358, in wait_for
    return fut.result()
      File "/usr/local/lib/python3.6/asyncio/futures.py", line 245, in result
    raise self._exception
      File "/usr/local/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
      File "/usr/local/lib/python3.6/asyncio/coroutines.py", line 210, in coro
    res = func(*args, **kw)
      File "/usr/src/app/homeassistant/components/light/insteon_plm.py", line 28, in async_setup_platform
        for device in discovery_info:
    TypeError: 'NoneType' object is not iterable
    2018-01-02 06:51:23 ERROR (MainThread) [homeassistant.components.switch] Error while setting up platform insteon_plm
    Traceback (most recent call last):
      File "/usr/src/app/homeassistant/helpers/entity_component.py", line 171, in _async_setup_platform
    SLOW_SETUP_MAX_WAIT, loop=self.hass.loop)
      File "/usr/local/lib/python3.6/asyncio/tasks.py", line 358, in wait_for
    return fut.result()
      File "/usr/local/lib/python3.6/asyncio/futures.py", line 245, in result
    raise self._exception
      File "/usr/local/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
      File "/usr/local/lib/python3.6/asyncio/coroutines.py", line 210, in coro
    res = func(*args, **kw)
      File "/usr/src/app/homeassistant/components/switch/insteon_plm.py", line 25, in async_setup_platform
        for device in discovery_info:
    TypeError: 'NoneType' object is not iterable

dakoellis, 2018-01-02:
Had some more messages come in about 20 minutes later.  I put it in pastebin instead of here:

https://pastebin.com/Via9rZgs

teharris1, 2018-01-02:
Can you upload your configuration.yaml? The log indicates that lights/insteon_plm is being called even before the first all-link record comes back, which should not be possible because the ALDB is loaded first then the devices are setup in HA.  Is it possible you have Insteon configuration in your config file in other areas (i.e. lights: )?

nugget, 2018-01-02:
Yeah, that's almost certainly the problem.  You should just have a single root-level `insteon_plm` block in the Home Assistant configuration file.  insteon_plm is a first class HA "component" that will take care of the individual light/switch/etc platforms by itself, internally.

dakoellis, 2018-01-02:
Got it. So when I try that I don't get any INSTEON devices. Do I need overrides or is that just for when they show up incorrectly? I suppose the other issue could be with the serial port settings but I'm pretty sure they are set the same as they are in windows where I can read the ALDB fine

teharris1, 2018-01-02:
@dakoellis Have you linked your devices to your PLM? If not, that would explain why you don't get any insteon devices. Insteonplm currently only loads devices that are linked to your PLM. In the prior release, @lanrongwen  introduced a feature to allow HA to see devices that are not linked. I have not reintroduced that feature. I plan to introduce Linking as the next feature. But in the meantime it may make sense to reintroduce the unlinked devices. 

@dakoellis BTW, are you running this on docker in Windows? Not sure it matters but just wanted to know. Based on the debug messages you pasted to pastebin, you appear to be dropping bytes. Any time you see a debug message that reads "Trimming leading buffer garbage" there was a problem between the physical PLM receiving data and ascynio capturing it in the data input buffer. You will almost always get multiple lines in a row when this happens so only worry about groups of them not each individual message. In your case this may not be a big deal because of the previous error messages but something to keep an eye on. If it continues frequently you may have an issue at a port level. The fact that you are receiving data however indicates it is not too fubar.

dakoellis, 2018-01-02:
So I used windows with IonW-PLM to make sure the PLM was working and also used that to setup the links. That program can control my lights so I figured they were linked. Do I still need to link each device by holding down the buttons in this case? 

As for hass, I'm running it in Linux with docker. I also haven't gotten the garbage message since removing the other lines so that is probably the case with the garbage messages. Thanks so much for all of your help and the work you're doing on it! I'll be so happy to be able to get rid of this hub

dakoellis, 2018-01-03:
OK @teharris1 I did a factory reset on the PLM and manually linked everything using the set buttons, and now whenever I press a button on one of my switches I get the error shown in this new pastebin: https://pastebin.com/MzEYqEXA

also one more showing my configuration.yaml (sanatized): https://pastebin.com/dCWvcYnD

teharris1, 2018-01-03:
@dakoellis OK you found a good one now :)  First, can you try letting it start up without touching any devices and let me know how that goes? Also what is device 46.D0.E6? I assume device 43.6C.15 is your PLM, correct?

@nugget or @lanrongwen any ideas on why this would be?  When I dissect the inbound messages I see 4 messages:
1) 02 50 46d0e6 000001 cf 11 00 
2) 02 00 15 
3) 02 50 46d0e6 436c15 40 11 01
4) 02 50 46d0e6 110201 cf 06 00
The system is choking on the second one because there is no message type 0x00. Have either of you seen this before? I am going to code around this for now and treat an unknown message type as a buffer error but I don't believe that is what this is. It looks like a NAK of some sort but the message type makes no sense.

dakoellis, 2018-01-03:
In that last error log, I went 10 minutes without pressing anything, and I don't get anything after the 

`DEBUG (MainThread) [insteonplm.plm] Ending: _load_all_link_database`

46d0e6 is a 2477S switch, and yes 436c15 is my PLM.  Also, I'm still not seeing any of my switches/dimmers in HASS, in case that's unexpected, and I should have mentioned that I set the switches up as responders and controllers.  Is that what I should be doing?  I wasn't getting any messages just setting them as responders to the plm only.

teharris1, 2018-01-03:
Sorry, I should have noticed the time stamp on the messages. Your PLM is not responding at all to the `Get IM Info` (0x02 0x60) message which is VERY odd and it is not respond to the `Get First All-Link Record` (0x02 0x69) messages. At a minimum you should get a NAK message that the ALDB dump is complete (0x02 0x69 0x15). The fact that your PLM is not responding to a Get IM Info request is very odd. @nugget any ideas? This is either specific to the 2413s or the device is malfunctioning (but since you say it works with other software I don't thing we should assume that).

BTW, I did fix the issue with unknown message types (i.e. 0x00)

dakoellis, 2018-01-03:
Just to be safe, I went ahead and connected the plm to my windows computer to make sure everything was still working with it and I am still able to control the devices with it

teharris1, 2018-01-03:
@dakoellis Looking at some documentation it could be that the read time out on the serial port for serial devices may need to be 1s. I have changed connection.py to set the timeout=1000. It has no negative impact to my USB based device but it may help your serial device. Let me know.

teharris1, 2018-01-03:
@nugget The more I look at this the more convinced I am that the issue is serial.aio forcing `timeout = 0` and `write_timeout = 0` to create non-blocking functions. While I get this in concept I am wondering if this could be causing issues with commands sending and receiving? When I look at the [INSTEON_Modem_Developer's_Guide_20071012a](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjmr9jRjrzYAhXLMSYKHc_LA7oQFggnMAA&url=http%3A%2F%2Fcache.insteon.com%2Fpdf%2FINSTEON_Modem_Developer%27s_Guide_20071012a.pdf&usg=AOvVaw2vXDbkwNqDThUityZvTwiB) page 6, it clearly expects the write function to be uninterrupted and the read function will happen as fast as the modem can send. You seem to know a lot more about asyncio and serial.aio than I do. What are your thoughts? In the current code in plm.py line 87 I have set `self.transport.serial.timeout = 1` hoping that will fix the issue with the 2413s.

dakoellis, 2018-01-04:
@teharris1 Tried it a couple of times, but still not getting any devices populating and the initial startup debug log is the same.

teharris1, 2018-01-04:
@dakoellis I am at a loss. Very sorry. Will keep looking and will open an issue with supporting the 2413s. 

@nugget and @lanrongwen can either of you confirm the code is working? I am hoping to execute a pull to bring the code into this main branch.

dakoellis, 2018-01-04:
No problem.  You've done more to get it working than anything I've been able to find in the past few months lol

teharris1, 2018-01-04:
I created this tread [Support PLM 2413s](https://github.com/nugget/python-insteonplm/issues/36 ). Please post there so we can work there.

rstanley75, 2018-01-05:
@teharris1, I've installed [insteonplm](https://github.com/teharris1/python-insteonplm) and replaced all of the necessary files from [home-assistant](https://github.com/teharris1/home-assistant).  I'll try my best not to go into incoherent rambling.

From `configuration.yaml`:
```
insteon_plm:
  port: /dev/ttyS0
  device_override:
  - address: 146355
    platform: light
```

Also, from `configuration.yaml`:
```
logger:
  default: warn
  logs:
    homeassistant.components.insteon_plm: debug
    insteonplm.protocol: debug
```

From `home-assistant.log`:
```
2018-01-05 13:59:51 INFO (MainThread) [homeassistant.components.insteon_plm] Looking for PLM on /dev/ttyS0
2018-01-05 14:00:02 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:02 DEBUG (MainThread) [homeassistant.components.insteon_plm] <insteonplm.devices.securityHealthSafety.SecurityHealthSafety object at 0x7f11d722f400>
2018-01-05 14:00:02 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 3291d1 (32.91.D1) binary_sensor
2018-01-05 14:00:02 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:11 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:11 DEBUG (MainThread) [homeassistant.components.insteon_plm] <insteonplm.devices.securityHealthSafety.SecurityHealthSafety object at 0x7f11f11397f0>
2018-01-05 14:00:11 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 2c6eb8 (2C.6E.B8) binary_sensor
2018-01-05 14:00:11 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:23 ERROR (Recorder) [homeassistant.components.recorder.util] Error executing query: <insteonplm.devices.securityHealthSafety.SecurityHealthSafety object at 0x7f11d722f400> is not JSON serializable
2018-01-05 14:00:26 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:26 DEBUG (MainThread) [homeassistant.components.insteon_plm] <insteonplm.devices.dimmableLightingControl.DimmableLightingControl object at 0x7f11d58f3fd0>
2018-01-05 14:00:26 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 3524ad (35.24.AD) light
2018-01-05 14:00:26 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:30 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:30 DEBUG (MainThread) [homeassistant.components.insteon_plm] <insteonplm.devices.dimmableLightingControl.DimmableLightingControl object at 0x7f11d71d4e10>
2018-01-05 14:00:30 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 3a1faa (3A.1F.AA) light
2018-01-05 14:00:30 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:32 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:32 DEBUG (MainThread) [homeassistant.components.insteon_plm] <insteonplm.devices.dimmableLightingControl.DimmableLightingControl object at 0x7f11d4030780>
2018-01-05 14:00:32 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 3a1043 (3A.10.43) light
2018-01-05 14:00:32 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:34 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:34 DEBUG (MainThread) [homeassistant.components.insteon_plm] <insteonplm.devices.dimmableLightingControl.DimmableLightingControl object at 0x7f11d4045b00>
2018-01-05 14:00:34 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 35f428 (35.F4.28) light
2018-01-05 14:00:34 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:36 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:36 DEBUG (MainThread) [homeassistant.components.insteon_plm] <insteonplm.devices.dimmableLightingControl.DimmableLightingControl object at 0x7f11f0044898>
2018-01-05 14:00:36 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 35eaa1 (35.EA.A1) light
2018-01-05 14:00:36 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:38 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:38 DEBUG (MainThread) [homeassistant.components.insteon_plm] <insteonplm.devices.dimmableLightingControl.DimmableLightingControl object at 0x7f11d7197908>
2018-01-05 14:00:38 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 350df5 (35.0D.F5) light
2018-01-05 14:00:38 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-05 14:00:43 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
[... cut ...]
2018-01-05 14:15:14 DEBUG (MainThread) [homeassistant.components.insteon_plm] <insteonplm.devices.dimmableLightingControl.DimmableLightingControl object at 0x7eff999a1d68>
2018-01-05 14:15:14 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 35f428 (35.F4.28) light
2018-01-05 14:15:14 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
```

The outdoor temperature is at a blistering high of 2°F right now so if I can get 10 minutes I might attempt to address that sensor behind the strike plate before the temp begins its inevitable plummet back down to "Why do I live here?" levels.

teharris1, 2018-01-05:
Wow, great feedback. Thanks.

1) Device Override: The prior code mixed HA and Insteon concepts in overrides. I chose to go with Insteon overrides only (i.e. cat, subcat). But looking at your observation there is probably a good reason to go with HA overrides as well. In otherwords, this is a cat 0x02 device, so it is not dimmable by Insteon definition. A 'light' in HA is dimmable but can that can be disabled as a feature. I chose to go with cat 0x01 = light (i.e. dimmable) 0x02 = switch (i.e. on/off only). I will rethink that.

2)  Binary Sensor: I will take a look at that one. 

3) Strange behavior: Yes I have seen the same thing and I now think I know what that is (and it is an easy fix I believe).

rstanley75, 2018-01-06:
@teharris1, regarding the `ToggleLinc Relay 2466S`, I see this in the new `ipdb.py`:
```
84        Product(0x02, 0x0d, None, 'ToggleLinc Relay', '2466S', SwitchedLightingControl),
```
where in the original `ipdb.py` the entry was formatted as this:
```
59        Product(0x02, 0x0d, None, 'ToggleLinc Relay', '2466S', ['light']),
```

IIRC, I believe I was responsible for the `['light']` attribute having been added to the original `ipdb.py`, and that it was originally `['switch']`.  At the time it seemed like the right thing to do as the 2466S is, well, a relay with a toggle switch that's obviously intended to replace a light switch in an in-wall gang box.  But now that I think about it at its heart it's a ToggleLinc Relay and as such doesn't care what it's wired to.  It could switch a light, a wall outlet, a ceiling fan, a heater, or whatever you wanted as long as the relay meets or exceeds the requirements for the load.

So do we decide that we need to differentiate between relays *intended* as light switches and relays *intended* as, well, relays?  Or do we treat all relays as switches and leave it up to the end user if they want to read up on `device_override` so they can tell Home Assistant to treat it as a light or not?  More importantly, does the universe descend into chaos the moment we start down this slippery slope of defining the purpose of things based on our subjective opinions?

teharris1, 2018-01-06:
You have hit on my exact concern. Moving from switch to light and vise versa is easy but what if someone says "I want my motion sensor to look like a switch"? Now we either have to have the switch platform in HA account for all motion sensor differences (like the fact that you cannot turn it on/off). The permutations and impossible scenarios go crazy. I am inclined to keep it the way it is. Which means, you can override a device's cat, subcat and product_key 
```
insteon:
  port: /dev/ttyUSB01
  device_override:
  - address: a1b2c3
    cat: 0x02
    subcat: 0x09
    product_key: 0x00004
```
The intention is if discovery does not work correctly you can set the device definition in configuration. 

BTW, I believe I fixed item 3 above. For item 2, can you run in debug logging mode for insteonplm and send me the relevent lines surrounding the json serialization error? Thanks.

rstanley75, 2018-01-10:
@teharris1, sorry it's taken me so long to get back to you.  Things got real busy for a few days there and then I went to get back to it I made a dumb mistake and hosed my virtualenv.  I'm pretty sure I have things back where they're supposed to be though.

The phantom device issue from item 3 above seems to be fixed now.

I finally got behind that strike plate to address that hidden door sensor (I use it to detect if the deadbolt is latched) so I hope I can be of some help to figure out the json serialization error.  There's two hidden door sensors in this story: `3291d1` and `2c6eb8`.

This is the logger config from `configuration.yaml`:
```
logger:
#  default: error
#  logs:
#    requests.packages.urllib3.connectionpool: critical #for this item anything below error will not be logged
  default: fatal
  logs:
    homeassistant.components.recorder: warn
    homeassistant.components.insteon_plm: debug
    insteonplm: debug
```

Prior to messing with `3291d1`:
```
2018-01-10 10:00:27 DEBUG (MainThread) [insteonplm.plm] Starting: _get_next_all_link_recor
2018-01-10 10:00:27 INFO (MainThread) [insteonplm.plm] Requesting Next All-Link Record
2018-01-10 10:00:27 DEBUG (MainThread) [insteonplm.plm] Starting: send_msg
2018-01-10 10:00:27 DEBUG (MainThread) [insteonplm.plm] Sending 2 byte message: 026a
2018-01-10 10:00:27 DEBUG (MainThread) [insteonplm.plm] Ending: send_msg
2018-01-10 10:00:27 DEBUG (MainThread) [insteonplm.plm] Ending: _get_next_all_link_recor
2018-01-10 10:00:27 DEBUG (MainThread) [insteonplm.plm] Ending _handle_all_link_record_response
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Starting: data_received
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Received 13 bytes from PLM: b'026a060257e2013291d1101144'
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Total buffer: b'026a060257e2013291d1101144'
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Starting: _peel_messages_from_buffer
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Finishing: _peel_messages_from_buffer
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Messages in queue: 2
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Processing message 026a06
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.messagecallback] Trying callback key: 6a:None:None:06
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.messagecallback] Trying callback key: 6a:None:None:06
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.messagecallback] Trying callback key: 6a:None:None:06
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.messagecallback] Trying callback key: 6a:None:None:None
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] No call back found for message 026a06
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Processing message 0257e2013291d1101144
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.messagecallback] Trying callback key: 57:None:None:None
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Last item in self._recv_queue reached.
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Finishing: data_received
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Starting _handle_all_link_record_response
2018-01-10 10:00:28 INFO (MainThread) [insteonplm.plm] Found all link record for device 3291d1
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Product data: address 3291d1 cat: 10 subcat: 11 product_key: 44
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.aldb] New INSTEON Device '3291d1': Hidden Door Sensor (10:11)
2018-01-10 10:00:28 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-10 10:00:28 DEBUG (MainThread) [homeassistant.components.insteon_plm] <insteonplm.devices.securityHealthSafety.SecurityHealthSafety object at 0x7f50c93f4128>
2018-01-10 10:00:28 INFO (MainThread) [homeassistant.components.insteon_plm] New INSTEON PLM device: 3291d1 (32.91.D1) binary_sensor
2018-01-10 10:00:28 DEBUG (MainThread) [homeassistant.components.insteon_plm] Starting Home-Assistant async_plm_new_device
2018-01-10 10:00:28 INFO (MainThread) [insteonplm.plm] Device with id 3291d1 added to device list from ALDB data.
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Starting: _get_next_all_link_recor
2018-01-10 10:00:28 INFO (MainThread) [insteonplm.plm] Requesting Next All-Link Record
2018-01-10 10:00:28 DEBUG (MainThread) [insteonplm.plm] Starting: send_msg
2018-01-10 10:00:29 DEBUG (MainThread) [insteonplm.plm] Sending 2 byte message: 026a
2018-01-10 10:00:29 DEBUG (MainThread) [insteonplm.plm] Ending: send_msg
2018-01-10 10:00:29 DEBUG (MainThread) [insteonplm.plm] Ending: _get_next_all_link_recor
2018-01-10 10:00:29 DEBUG (MainThread) [insteonplm.plm] Ending _handle_all_link_record_response
2018-01-10 10:00:29 DEBUG (MainThread) [insteonplm.statechangesignal] Registered callback for state: Sensor
2018-01-10 10:00:30 DEBUG (MainThread) [insteonplm.plm] Starting: data_received
2018-01-10 10:00:30 DEBUG (MainThread) [insteonplm.plm] Received 13 bytes from PLM: b'026a060257a2013291d1101144'
2018-01-10 10:00:30 DEBUG (MainThread) [insteonplm.plm] Total buffer: b'026a060257a2013291d1101144'
2018-01-10 10:00:30 DEBUG (MainThread) [insteonplm.plm] Starting: _peel_messages_from_buffer
2018-01-10 10:00:30 DEBUG (MainThread) [insteonplm.plm] Finishing: _peel_messages_from_buffer
2018-01-10 10:00:30 DEBUG (MainThread) [insteonplm.plm] Messages in queue: 2
[... cut ...]
2018-01-10 11:46:51 ERROR (Recorder) [homeassistant.components.recorder.util] Error executing query: <insteonplm.devices.securityHealthSafety.SecurityHealthSafety object at 0x7f976fc29198> is not JSON serializable
```

If you'd like more detail I can share the full logs but these seemed to be the relevant parts.  The observation I'm making is that when both `3291d1` and `2c6eb8` are linked the json serialization error applies to `3291d1`.  When `2c6eb8` is the only hidden door sensor the json serialization error applies to that instead.  I don't know if it's relevant, but it seems `3291d1` is being discovered first in the cases where both sensors are linked.

teharris1, 2018-02-01:
I am ready to make a PR into this repo. TESTERS NEEDED!
@lanrongwen @nugget @dakoellis @rstanley75

closed 2018-02-01 by teharris1