nugget/python-insteonplm #35: I/O Linc

teharris1, 2018-01-03:
I have added I/O Linc 2450 to my refactored code. If someone can validate it and let me know if it works please. Thanks.

majortom64, 2018-01-28:
It does not seem to be working.

I think these are the right log entries:

2018-01-28 08:42:24 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.insteon_plm
2018-01-28 08:42:24 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.insteon_plm
2018-01-28 08:42:24 INFO (MainThread) [homeassistant.components.binary_sensor.insteon_plm] Registered 242e3f_2 with binary_sensor platform.
2018-01-28 08:42:24 INFO (MainThread) [homeassistant.components.binary_sensor.insteon_plm] Registered 242e3f_2 with binary_sensor platform.
2018-01-28 08:42:24 INFO (MainThread) [homeassistant.components.switch] Setting up switch.insteon_plm
2018-01-28 08:42:24 INFO (MainThread) [homeassistant.components.switch] Setting up switch.insteon_plm
2018-01-28 08:42:24 INFO (MainThread) [homeassistant.components.binary_sensor.insteon_plm] Sensor state for 242e3f_2 is 0
2018-01-28 08:42:24 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=binary_sensor.242e3f_2, old_state=None, new_state=<state binary_sensor.242e3f_2=off; INSTEON Address=XX.XX.XX, Description=I/O Linc Sensor, Model=2450, Category=07, Subcategory=00, Product Key / Firmware=00, Group=2, friendly_name=242e3f_2 @ 2018-01-28T08:42:24.817311-08:00>>
2018-01-28 08:42:24 INFO (MainThread) [homeassistant.components.binary_sensor.insteon_plm] Sensor state for 242e3f_2 is 0
2018-01-28 08:42:24 INFO (MainThread) [homeassistant.components.switch.insteon_plm] Registered 242e3f with switch platform.
2018-01-28 08:42:24 INFO (MainThread) [homeassistant.components.switch.insteon_plm] Registered 242e3f with switch platform.
2018-01-28 08:42:25 ERROR (MainThread) [homeassistant.components.switch] Error while setting up platform insteon_plm
Traceback (most recent call last):
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/helpers/entity_component.py", line 189, in _async_setup_platform
    SLOW_SETUP_MAX_WAIT, loop=self.hass.loop)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 358, in wait_for
    return fut.result()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/futures.py", line 245, in result
    raise self._exception
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/coroutines.py", line 210, in coro
    res = func(*args, **kw)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/switch/insteon_plm.py", line 29, in async_setup_platform
    InsteonPLMSwitchDevice(hass, device)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/switch/insteon_plm.py", line 43, in __init__
    self._device.lightOnLevel.connect(self.async_switch_update)
AttributeError: 'SensorsActuators_2450' object has no attribute 'lightOnLevel'
2018-01-28 08:42:25 ERROR (MainThread) [homeassistant.components.switch] Error while setting up platform insteon_plm
Traceback (most recent call last):
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/helpers/entity_component.py", line 189, in _async_setup_platform
    SLOW_SETUP_MAX_WAIT, loop=self.hass.loop)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 358, in wait_for
    return fut.result()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/futures.py", line 245, in result
    raise self._exception
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/coroutines.py", line 210, in coro
    res = func(*args, **kw)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/switch/insteon_plm.py", line 29, in async_setup_platform
    InsteonPLMSwitchDevice(hass, device)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/switch/insteon_plm.py", line 43, in __init__
    self._device.lightOnLevel.connect(self.async_switch_update)
AttributeError: 'SensorsActuators_2450' object has no attribute 'lightOnLevel'

majortom64, 2018-02-07:
I/O Link now works correctly. It recognizes both the switch and the sensor.

teharris1, 2018-02-08:
Closing this isse based on confirmation that the device works in the latest release.

closed 2018-02-08 by teharris1