nugget/python-insteonplm #36: Support PLM 2413s

teharris1, 2018-01-04:
Indications are that the 2413s is not supported. This thread will focus on debugging and resolving the issue.

teharris1, 2018-01-04:
Can you download this tool? [interceptty](https://github.com/geoffmeyers/interceptty)  I had trouble getting docker to see the front port but was able to use it outside docker using insteonplm_monitor.

teharris1, 2018-01-04:
For clarity, after you install interceptty you need to open two terminals in the first terminal you should enter:
`sudo interceptty /dev/ttyS0 /dev/newport`  where ttyS0 is your com port where the PLM is connected.
In the second window you should run:
`insteonplm_monitor --device /dev/newport -v`
I am looking for the output of hte first terminal which should look something like this:
```
 0x02
< 0x60 (`)
>       0x02
>       0x60 (`)
>       0x19
>       0x1a
>       0x60 (`)
>       0x03
>       0x20
>       0x38 (8)
>       0x06
< 0x02
< 0x69 (i)
>       0x02
>       0x69 (i)
>       0x06
>       0x02
>       0x57 (W)
>       0xe2
>       0x00
>       0x14
>       0x62 (b)
>       0x7a (z)
>       0x01
>       0x00
>       0x00
< 0x02
< 0x6a (j)
>       0x02
>       0x6a (j)
>       0x06
>       0x02
>       0x57 (W)
>       0xe2
>       0x00
>       0x2c (,)
>       0x38 (8)
>       0x32 (2)
>       0x01
>       0x00
>       0x00
```

dakoellis, 2018-01-04:
OK I ran it and this is what I got

    ❯ sudo interceptty /dev/ttyS0 /dev/plm
    < 0x02
    < 0x60 (`)
    < 0x02
    < 0x69 (i)

I had it running for a few minutes and didn't get anything else coming through, but a bunch of lines came in when my wife turned on a light, if that is helpful:

    > 	0x01
    > 	0x8d
    > 	0x0b
    > 	0x18
    > 	0x05
    > 	0xff
    > 	0xc8
    >  	0x8d
    > 	0x0b
    > 	0x18
    > 	0x09
    > 	0xc8
    >   	0x8d
    > 	0x0b
    > 	0x18
    > 	0x05
    > 	0xc8
    > 	0x8d
    > 	0x9b
    > 	0x7d (})
    > 	0x5c (\)
    > 	0xfc
    > 	0xc8
    > 	0x8d
    > 	0x0b
    > 	0x18
    > 	0x01
    > 	0xc8
    > 	0x8d
    > 	0x9b
    > 	0x7d (})
    > 	0x5c (\)
    > 	0xfc
    > 	0xc8
    > 	0x8d
    > 	0x5b ([)
    > 	0x19
    > 	0x05

teharris1, 2018-01-04:
That is exactly what would have been expected from our previous tests. It confirms that the PLM is not responding to message 0x02 0x60  or 0x02 0x69. As you can see from my example dump you should have received an immediate response from 0x02 0x60.  I am going to write a quick little program which does not use a asynchronous programming to see if that makes a difference.

teharris1, 2018-01-04:
[test2413s.tar.gz](https://github.com/nugget/python-insteonplm/files/1604967/test2413s.tar.gz)
@dakoellis can you download and untar (`tar -xvf test2413s.tar.gz`) this to your Linux box? Run 
`sudo interceptty /dev/ttyS0 /dev/plm` 
as above and in a separate terminal go to the directory where you downloaded this and cd to the test2413s directory. From there run:
`./test2413s --device /dev/plm --light xxxxxx --action` on (or off) 
and send the results from interceptty.

dakoellis, 2018-01-04:
here it is

```
❯ sudo interceptty /dev/ttyS0 /dev/plm
< 0x02
< 0x69 (i)
< 0x02
< 0x62 (b)
< 0x46 (F)
< 0xca
< 0x3c (<)
< 0x00
< 0x11
< 0xff
```

teharris1, 2018-01-04:
Did your light turn off or on? That is not good that interceptty did not catch anything. I am getting more confused about the problem. This program is a simple an I/O request as possible and still nothing from the PLM.

dakoellis, 2018-01-05:
nope nothing happened.  What are the chances that there's some kind of serial settings or driver issue going on?

I also tried this with both computers just to make sure something weird wasn't going on with the serial port on my main computer and got nothing.  Made sure with both that I was still receiving data when I pushed the switch on the wall as well

nugget, 2018-01-05:
This is along shot but it might be worth experimenting with the pyserial buffer settings.  I removed the larger write buffer last July before I figured out that the aio.py problem was actually a pyserial bug (see Issue #19) and never put the code back.

https://github.com/nugget/python-insteonplm/commit/9ea66b63560efe91128ed76e00a0c57f95dda1b7

It's *possible* that increasing the send/receive buffers might help here if this is a timing issue we're experiencing with the relatively slower RS232 as compared to USB.

teharris1, 2018-01-05:
@dakoellis 
[test2413s.tar.gz](https://github.com/nugget/python-insteonplm/files/1605386/test2413s.tar.gz)
Can you try this update? 

@nugget that may work. In my tar file attached I am just using serial not serial.aio for simplicity.

rstanley75, 2018-01-05:
Hey, I have a 2413s too!  And it's been acting up on me!  I thought it was just bad capacitors causing it to flake out on me but this whole thread has me wondering if what's being discussed here is the culprit.

First I did `./test2413s --device /dev/plm --light 3a1043 --action off` to turn off a LampLinc 2457D2, which gave me:
```
< 0x02
< 0x69 (i)
>       0x02
>       0x69 (i)
>       0x06
>       0x02
>       0x57 (W)
>       0xe2
>       0x01
>       0x14
>       0x63 (c)
>       0x55 (U)
>       0x02
>       0x0d
>       0x38 (8)
< 0x02
< 0x62 (b)
< 0x3a (:)
< 0x10
< 0x43 (C)
< 0x00
< 0x13
< 0x00
>       0x02
>       0x62 (b)
>       0x3a (:)
>       0x10
>       0x43 (C)
>       0x00
>       0x13
>       0x00
>       0x06
>       0x02
>       0x50 (P)
>       0x3a (:)
>       0x10
>       0x43 (C)
>       0x34 (4)
>       0xe6
>       0x2a (*)
>       0x26 (&)
>       0x13
>       0x00
```
Immediately after, I tried to send `./test2413s --device /dev/plm --light 3a1043 --action on` but the light didn't turn back on and nothing new was output from `interceptty`.

I restarted `interceptty` with the same parameters, then tried again with `./test2413s --device /dev/plm --light 3a1043 --action on`, and here was the output:
```
< 0x02
< 0x69 (i)
>       0x02
>       0x69 (i)
>       0x06
>       0x02
>       0x57 (W)
>       0xe2
>       0x01
>       0x14
>       0x63 (c)
>       0x55 (U)
>       0x02
>       0x0d
>       0x38 (8)
< 0x02
< 0x62 (b)
< 0x3a (:)
< 0x10
< 0x43 (C)
< 0x00
< 0x11
< 0xff
>       0x02
>       0x62 (b)
>       0x3a (:)
>       0x10
>       0x43 (C)
>       0x00
>       0x11
>       0xff
>       0x06
>       0x02
>       0x50 (P)
>       0x3a (:)
>       0x10
>       0x43 (C)
>       0x34 (4)
>       0xe6
>       0x2a (*)
>       0x26 (&)
>       0x11
>       0xff
```

I did this a few more times and the same thing happened where I had to restart `interceptty` after turning the lamp off to get it to accept another command, but not after turning it back on.  Normally in Home Assistant I can turn lamps on and off, but inevitably I'll it'll happen where I do that and then I can't control any more Insteon devices until I restart Home Assistant.

dakoellis, 2018-01-05:
@teharris1 here's my output:

```
❯ sudo interceptty /dev/ttyS0 /dev/plm
< 0x02
< 0x60 (`)
< 0x02
< 0x69 (i)
< 0x02
< 0x30 (0)
< 0x02
< 0x62 (b)
< 0x46 (F)
< 0xca
< 0x3c (<)
< 0x00
< 0x11
< 0xff
```
```
❯ ll /dev/ttyS0
crw-rw---- 1 root uucp 4, 64 Jan  4 19:14 /dev/ttyS0
```
with @rstanley75 having a lot more success, I'm starting to think it's a serial config issue or something.  @rstanley75 did you do anything to get the 2413S working initially?

teharris1, 2018-01-05:
@rstanley75 I have the same issue with needing to restart interceptty with my 2448U so I think that is "normal". Your output is exactly what I would expect. You are probably right to think it is a capacitor issue which is why I went to the USB stick. Hope your soldering skills are better than mine :) 

@dakoellis Good news / bad news. I don't think this is a problem with the 2413s based on the output from @rstanley75.

rstanley75, 2018-01-05:
@dakoellis, I didn't have to do anything special as far as serial-specific config goes.  I will say that my setup is a little more complicated than most in that I'm running it in a container on a Proxmox host and had to do some trickery to pass the device through to the container running Home Assistant.

teharris1, 2018-01-05:
@rstanley75 did you run the previous analysis from the Proxmox layer or the container? Not sure it matters but just curious. Also, there are two tar files in this thread. The first one runs at the recommended baud rate of 19200 where the second one takes it down to 4800. Which did you try?

FYI, in the second file I also sleep for a second after opening the port thinking there may be some driver issue with the serial port opening. The reason I was thinking that is no response comes from the commends sent to the PLM but responses come from devices after the PLM is connected for some period which is very hard to figure out.

rstanley75, 2018-01-05:
I wouldn't say my soldering skills are stellar, @teharris1, but I (and the parts) usually make it through unscathed.  My 2413S was purchased mid 2015 so it's probably about due.

I ran the analysis from inside the container using the second tar file but I'll give it a try with the first.

rstanley75, 2018-01-05:
Instead of downloading the first one I just changed `baudrate` to from `4800` to `19200` in `test2413s.py`.

`./test2413s --device /dev/plm --light 3a1043 --action off`:
```
< 0x02
< 0x30 (0)
>       0x02
>       0x30 (0)
>       0x15
< 0x02
< 0x62 (b)
< 0x3a (:)
< 0x10
< 0x43 (C)
< 0x00
< 0x13
< 0x00
>       0x02
>       0x62 (b)
>       0x3a (:)
>       0x10
>       0x43 (C)
>       0x00
>       0x13
>       0x00
>       0x06
>       0x02
>       0x50 (P)
>       0x3a (:)
>       0x10
>       0x43 (C)
>       0x34 (4)
>       0xe6
>       0x2a (*)
>       0x21 (!)
>       0x13
>       0x00
```
`./test2413s --device /dev/plm --light 3a1043 --action on` (after restarting `interceptty`):
```
< 0x02
< 0x30 (0)
>       0x02
>       0x30 (0)
>       0x15
< 0x02
< 0x62 (b)
< 0x3a (:)
< 0x10
< 0x43 (C)
< 0x00
< 0x11
< 0xff
>       0x02
>       0x62 (b)
>       0x3a (:)
>       0x10
>       0x43 (C)
>       0x00
>       0x11
>       0xff
>       0x06
>       0x02
>       0x50 (P)
>       0x3a (:)
>       0x10
>       0x43 (C)
>       0x34 (4)
>       0xe6
>       0x2a (*)
>       0x25 (%)
>       0x11
>       0xff
```

teharris1, 2018-01-05:
Exactly what I would expect so I think we can say the 2413s responds to the standard Insteon PLM configuration. 
@dakoellis I suggest you start looking at your Linux install. What distro are you on BTW?

dakoellis, 2018-01-05:
I'm running Arch.  Sorry to waste so much of your time with this since it looks like a config issue!

teharris1, 2018-01-05:
No worries.  I am going to close this issue then since this does not appear to be an Insteon specific problem.

teharris1, 2018-01-05:
@rstanley75 have you been able to try my refactored code? You need to download [insteonplm ](https://github.com/teharris1/python-insteonplm)and [home-assistant](https://github.com/teharris1/home-assistant). From Home Assistant, you really only need to replace the four insteon_plm.py files located in components/, components/light, components/switch, components/binary_sensors, components/fan and components/sensor.  It is a major revision and I am trying to get as many testers as I can before issuing a pull request.

rstanley75, 2018-01-05:
@teharris1, I haven't tried any of of your refactored code yet but I'll look into it now.

teharris1, 2018-01-08:
Closing per note above.

closed 2018-01-08 by teharris1