nugget/python-insteonplm #38: Add support for insteon link management.

majortom64, 2018-01-28:
One of Insteon's best features is that it can be configured to run with no controller, using device to devices links and groups. This is particularly valuable with KeyPadLinc - configuring bidirectional groups of links makes it so that all the button-LEDs remain in sync when one turns a light on or off via a linked KeyPadLinc without needing a controller's intervention.

There are several problems with manually creating these links:

1. the number of links needed gets very big quickly (2 KeyPadLincs require 4 links to have a button on each control the other, 3 KeyPadLincs require 6 links, _etc._) and remembering all that need to be done is hard.
2. there is no easy way to ensure that all the links are correctly set other than noticing that one button does not control the device as it should, or that the state gets out of sync.
3. it can involve a lot of walking (given that these linked devices do not need to be in the same room or even on the same floor, and that they have to be done one at a time, this is an actual issue).
4. some features cannot be supported directly (KeyPadLinc detached loads and self referential buttons are two examples).

@teharris1 had asked whether this software should just be a driver for Home Assistant or be more general. I think that really be able to use Insteon gear in this environment, one needs to be able to configure these links with (at a minimum) a command line tool. It would be great to have a GUI to do this, but that is not critical.

[This post](http://forums.indigodomo.com/viewtopic.php?p=123044#p123044) on the Indigo Forum explains the ideas better. 

As I mentioned, link management is particularly valuable with detached loads. Newer generation KeyPadLincs (_i.e._ all built after 2013, as I understand it) can detach control of the local load from button 1 (in 8-button mode) or buttons 1-2 and 7-8 (in 6-button mode). This is really nice for having the same button control the same light on multiple KeyPadLincs in the same space ([this](http://forums.indigodomo.com/download/file.php?id=5314) shows an example of how it would be used).

Here are the notes from the developer docs that explain it:

http://cache.insteon.com/developer/2334-xxx2486xdev-042013-en.pdf
Detach Load Notes:
- Detach Load is only supported through software. It cannot be turned on/off from the device.
- When Detach Load is turned On, the load will then be moved/controlled by Group 9
o Link to Group 9 using the SET Button.
o Link to Group 1 by using the ON or OFF Button. o Link to Buttons A-H using the A-H buttons.

- Control of the LEDs when Detach load is enabled
o To control the Group 1 LEDs, simply send the Keypad a Group 1 On/Off
command
o To control the other LEDs, use the Extended 2E Set LED Bit Mask command to
set those LEDs

-Getting status of the LEDs when Detach Load is enabled

o To get the Status of the 8 LEDs (Group 1-8), use the Read Data 2E command and
the Data 11 byte contains the LED State information.
o To get the Status of Group 9 (The Load), use the Standard status request
command Cmd1=19 Cmd2=00

teharris1, 2018-04-24:
@majortom64 I am going to separate detached load from All-Link management. I am also going to separate link checking from managing links. So for this request I see three separate requests:
1. Ability to create/delete/change links in a device ALDB
2. Ability to confirm links are set in both directions
3. Ability to set detached load and allow links to manage the load.

Did I miss anything?

majortom64, 2018-06-17:
@teharris1 I would add one thing to number 2:

Ability to automatically calculate all the needed links. What I mean by that is if I have three keypads and I want button b on all three to control and reflect the state of another switch (could be a keypad or a plain switch), I need to make sure they are all linked together:

1B -> 2B
2B -> 1B
1B -> 3B
3B -> 1B
2B -> 3B
3B -> 2B

Making sure the links are symmetrical is great, but what would be nice is the ability to say:

Link 1B 2B 3B 4B 5B

And have all the links needed happen automatically (to be clear, it should work with a mix of keypad linc and regular switch and the ability to pick any button as part of the linking). :-)

(P.S. Sorry it took me so long to respond. I thought I had and did not realize I had not until yesterday when I finally got my system reconfigured and running the current branch of everything.)

wz2b, 2018-09-04:
Using this sequence:

  connect
  load_aldb plm

I get this error:

```
ERROR:asyncio:Task exception was never retrieved
future: <Task finished coro=<Commander._read_line() done, defined at /srv/homeassistant/lib/python3.5/site-packages/insteonplm/tools.py:361> exception=Error('Odd-length string',)>
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/tools.py", line 366, in _read_line
    yield from self._exec_cmd(input)
  File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/tools.py", line 390, in _exec_cmd
    return_val = yield from func(arg)
  File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/tools.py", line 706, in do_load_aldb
    yield from self.tools.load_device_aldb(addr, clear)
  File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/tools.py", line 252, in load_device_aldb
    dev_addr = Address(addr)
  File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/address.py", line 17, in __init__
    self.addr = self._normalize(addr)
  File "/srv/homeassistant/lib/python3.5/site-packages/insteonplm/address.py", line 90, in _normalize
    normalize = binascii.unhexlify(addr.lower())
binascii.Error: Odd-length string
```
It's related to this ticket, which is still open, so I'm adding this note here, but will open a separate ticket if requested.

Separately, `load_aldb plm` never seems to actually finish.  I see it try (in a few cases it tells me I need to use a device override because it could not reach the device).

teharris1, 2023-02-10:
Closing this issue as the `insteonplm` module is no longer maintained. Please use the `pyinsteon` module instead.

closed 2023-02-10 by teharris1