nugget/python-insteonplm #39: Saving the results of auto-discovery.

majortom64, 2018-01-29:
It would be nice to be able to save the results of auto-discovery, so that Home Assistant would start faster. It could still do auto-discovery but at least would have all the devices previously discovered.

teharris1, 2018-01-30:
Agreed. Will put on the roadmap but it should be pretty easy.

teharris1, 2018-02-01:
This is complete in my latest version.

teharris1, 2018-02-08:
@majortom64 This should be working in the latest release. You should see a file in your HA config directory called `insteon_plm_device_info.dat`. This file saves the device info (cat, subcat, firmware) so that it does not need to ask the device again at startup. Please confirm we can close this issue.

closed 2018-03-23 by teharris1