nugget/python-insteonplm #40: Typo in the code....

majortom64, 2018-01-31:
I got this error:

2018-01-30 19:32:48 INFO (MainThread) [homeassistant.components.light.insteon_plm] Received update calback from PLM for 264bce
2018-01-30 19:32:49 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback PLM._handle_standard_or_extended_message_received(<insteonplm.m...t 0x105d73898>)
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/events.py", line 127, in _run
    self._callback(*self._args)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/insteonplm/plm.py", line 274, in _handle_standard_or_extended_message_received
    device.receive_message(msg)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/insteonplm/devices/dimmableLightingControl.py", line 194, in receive_message
    super().receive_message(msg)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/insteonplm/devices/dimmableLightingControl.py", line 72, in receive_message
    super().receive_message(msg)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/insteonplm/devices/devicebase.py", line 94, in receive_message
    callback(msg)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/insteonplm/devices/dimmableLightingControl.py", line 244, in _light_off_command_received
    device2.ligth_status_request()
AttributeError: 'DimmableLightingControl_2475F' object has no attribute **'ligth_status_request'**
2018-01-30 19:32:49 ERROR (MainThread) [homeassistant.core] Error doing job: Exception in callback PLM._handle_standard_or_extended_message_received(<insteonplm.m...t 0x105d73978>)
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/events.py", line 127, in _run
    self._callback(*self._args)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/insteonplm/plm.py", line 274, in _handle_standard_or_extended_message_received
    device.receive_message(msg)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/insteonplm/devices/dimmableLightingControl.py", line 194, in receive_message
    super().receive_message(msg)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/insteonplm/devices/dimmableLightingControl.py", line 72, in receive_message
    super().receive_message(msg)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/insteonplm/devices/devicebase.py", line 94, in receive_message
    callback(msg)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/insteonplm/devices/dimmableLightingControl.py", line 244, in _light_off_command_received
    device2.ligth_status_request()
AttributeError: 'DimmableLightingControl_2475F' object has no attribute **'ligth_status_request'**
2018-01-30 19:32:50 INFO (MainThread) [homeassistant.components.light.insteon_plm] Received update calback from PLM for 216a65

I would guess the two bolded parts are the problem. :-D

teharris1, 2018-02-01:
Try it now.

teharris1, 2018-02-08:
@majortom64 I am assuming this can be close, correct?

closed 2018-03-23 by teharris1