nugget/python-insteonplm #42: Bump version - refactored code meaningfully ready

teharris1, 2018-02-01:
My refactoring efforts are done I believe. I have a lot of documentation to write and there are still some open issues (small I think). But I am proposing that the latest code become version 0.8.0.0.  To test the latest code you need to download insteonplm and home-assistant. From Home Assistant, you really only need to replace the 5 insteon_plm.py files located in 
- components/insteon_plm.py, 
- components/light/insteon_plm.py, 
- components/switch/insteon_plm.py 
- components/binary_sensors/insteon_plm.py
- components/fan/insteon_plm.py

rstanley75, 2018-02-02:
I've been using your refactored code successfully until sometime last week when I experienced a motherboard failure.  Up until then it's been great.  All of my old glitching that I attributed to failing capacitors was gone and my PLM has been the most reliable it's ever been.  I should be able to resume testing after the weekend, I'm just waiting on a USB serial adapter to get around an incompatibility issue with ESXi and the new hardware.

majortom64, 2018-02-02:
I will install later tonight and let you know how it goes. :-)

teharris1, 2018-02-08:
@rstanley75, any luck retesting? Hopefully you got your issues fixed.

rstanley75, 2018-02-08:
@teharris1, I've been meaning to post.  It turns out my serial port was fine the whole time and getty had taken over the serial port to use as a serial console.  It was a simple fix, I just hadn't taken the right steps to troubleshoot the problem.

Anyway, for the most part so far so good with the refactored code.  The only issue I've run into is with my door sensors.

```
2018-02-07 22:20:03 WARNING (MainThread) [insteonplm.plm] Device 3291d1 found in the ALDB did not respond and is being removed from the list.
2018-02-07 22:20:03 WARNING (MainThread) [insteonplm.plm] If this device is still active you can add it to the device_override configuration.
2018-02-07 22:20:03 WARNING (MainThread) [insteonplm.plm] Device 2c6eb8 found in the ALDB did not respond and is being removed from the list.
2018-02-07 22:20:03 WARNING (MainThread) [insteonplm.plm] If this device is still active you can add it to the device_override configuration.
```

I think I'll have time later to play with `device_override` and see if I can resolve the issue in the manner the log messages suggest.

teharris1, 2018-02-08:
There is probably an easy fix for this in the code. What is the cat and subcat of the device?

teharris1, 2018-02-08:
Actually, can you put the insteonplm module in debug mode for logging and send me any lines that look like this:
`DEBUG:insteonplm.plm:Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 41.89.CF, 'linkdata1': 0x00, 'linkdata2': 0x00, 'linkdata3': 0x00}`
where address is the two addresses 32.91.D1 and 2C.6E.B8? Also, what software do you use for setting up device links to your PLM? Or do you just use the link buttons?

teharris1, 2018-02-08:
Given the positive feedback from @rstanley75 (the issues above is actually a design decision, not a bug) and from @majortom64 (FanLinc will be fixed tonight and not planning KPL for this release), I am going to merge the PR and bump the version. 

@nugget are you able to release the new version to PyPi?

nugget, 2018-02-08:
Absolutely, whenever you're ready to go.

teharris1, 2018-02-08:
Should be ready now.

rstanley75, 2018-02-08:
@teharris1, I use the link buttons to set up device links.  I link the device to the PLM and then I link again from the PLM to the device.

Here are the lines from the log that look like what you described:
```
2018-02-08 16:06:44 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x62, 'address': 32.91.D1, 'flags': 0x00, 'cmd1': 0x10, 'cmd2': 0x00, 'acknak': 0x06}
...
2018-02-08 16:06:46 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x62, 'address': 2C.6E.B8, 'flags': 0x00, 'cmd1': 0x10, 'cmd2': 0x00, 'acknak': 0x06}
...
2018-02-08 16:06:50 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x62, 'address': 32.91.D1, 'flags': 0x00, 'cmd1': 0x10, 'cmd2': 0x00, 'acknak': 0x06}
...
2018-02-08 16:06:52 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x62, 'address': 2C.6E.B8, 'flags': 0x00, 'cmd1': 0x10, 'cmd2': 0x00, 'acknak': 0x06}
...
2018-02-08 16:06:56 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x62, 'address': 32.91.D1, 'flags': 0x00, 'cmd1': 0x10, 'cmd2': 0x00, 'acknak': 0x06}
...
2018-02-08 16:06:58 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x62, 'address': 2C.6E.B8, 'flags': 0x00, 'cmd1': 0x10, 'cmd2': 0x00, 'acknak': 0x06}
```

That entire section of the log:
```
2018-02-08 16:06:40 DEBUG (MainThread) [insteonplm.devices.devicebase] Starting DeviceBase._receive_message
2018-02-08 16:06:40 DEBUG (MainThread) [insteonplm.devices.devicebase] Got Message ACK
2018-02-08 16:06:40 DEBUG (MainThread) [insteonplm.devices.devicebase] Message ACK with no callback
2018-02-08 16:06:40 DEBUG (MainThread) [insteonplm.devices.devicebase] Ending DeviceBase._receive_message
2018-02-08 16:06:40 DEBUG (MainThread) [insteonplm.devices.devicebase] Starting DeviceBase._receive_message
2018-02-08 16:06:40 DEBUG (MainThread) [insteonplm.devices.devicebase] Got Message ACK
2018-02-08 16:06:40 DEBUG (MainThread) [insteonplm.devices.devicebase] Message ACK with no callback
2018-02-08 16:06:40 DEBUG (MainThread) [insteonplm.devices.devicebase] Ending DeviceBase._receive_message
2018-02-08 16:06:40 DEBUG (MainThread) [insteonplm.plm] True
2018-02-08 16:06:42 DEBUG (MainThread) [insteonplm.plm] No new messages received.
2018-02-08 16:06:42 DEBUG (MainThread) [insteonplm.plm] Ending _get_from_send_queue
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Starting _handle_get_next_all_link_record_nak
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] All-Link device records found in ALDB: 2
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.devices.devicebase] Starting: devicebase.Id_Request
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Starting: send_msg
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Ending: send_msg
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.devices.devicebase] Ending: devicebase.Id_Request
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.devices.devicebase] Starting: devicebase.Id_Request
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Starting: send_msg
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Ending: send_msg
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.devices.devicebase] Ending: devicebase.Id_Request
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Ending _handle_get_next_all_link_record_nak
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Starting _get_from_send_queue
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Aquiring write lock
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] True
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Writing message: {'code': 0x62, 'address': 32.91.D1, 'flags': 0x00, 'cmd1': 0x10, 'cmd2': 0x00, 'acknak': None}
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Starting _get_from_send_queue
2018-02-08 16:06:43 DEBUG (MainThread) [insteonplm.plm] Ending _get_from_send_queue
2018-02-08 16:06:44 DEBUG (MainThread) [insteonplm.plm] Starting: data_received
2018-02-08 16:06:44 DEBUG (MainThread) [insteonplm.plm] Received 9 bytes from PLM: b'02623291d100100006'
2018-02-08 16:06:44 DEBUG (MainThread) [insteonplm.plm] Total buffer: b'02623291d100100006'
2018-02-08 16:06:44 DEBUG (MainThread) [insteonplm.plm] Starting: _peel_messages_from_buffer
2018-02-08 16:06:44 DEBUG (MainThread) [insteonplm.plm] Finishing: _peel_messages_from_buffer
2018-02-08 16:06:44 DEBUG (MainThread) [insteonplm.plm] Messages in queue: 1
2018-02-08 16:06:44 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x62, 'address': 32.91.D1, 'flags': 0x00, 'cmd1': 0x10, 'cmd2': 0x00, 'acknak': 0x06}
2018-02-08 16:06:45 DEBUG (MainThread) [insteonplm.plm] Calling method <function DeviceBase._receive_message at 0x6f2401457400>
2018-02-08 16:06:45 DEBUG (MainThread) [insteonplm.plm] Calling method <function DeviceBase._receive_message at 0x6f2401457400>
2018-02-08 16:06:45 DEBUG (MainThread) [insteonplm.plm] Last item in self._recv_queue reached.
2018-02-08 16:06:45 DEBUG (MainThread) [insteonplm.plm] Finishing: data_received
2018-02-08 16:06:45 DEBUG (MainThread) [insteonplm.devices.devicebase] Starting DeviceBase._receive_message
2018-02-08 16:06:45 DEBUG (MainThread) [insteonplm.devices.devicebase] Got Message ACK
2018-02-08 16:06:45 DEBUG (MainThread) [insteonplm.devices.devicebase] Message ACK with no callback
2018-02-08 16:06:45 DEBUG (MainThread) [insteonplm.devices.devicebase] Ending DeviceBase._receive_message
[... cut ...]
2018-02-08 16:15:10 DEBUG (MainThread) [insteonplm.plm] Finishing: data_received
2018-02-08 16:15:10 DEBUG (MainThread) [insteonplm.devices.devicebase] Starting DeviceBase._receive_message
2018-02-08 16:15:10 DEBUG (MainThread) [insteonplm.devices.devicebase] Ending DeviceBase._receive_message
2018-02-08 16:15:10 DEBUG (MainThread) [insteonplm.devices.devicebase] Starting DeviceBase._receive_message
2018-02-08 16:15:10 DEBUG (MainThread) [insteonplm.devices.devicebase] Ending DeviceBase._receive_message
```

teharris1, 2018-02-08:
Unfortunately these are not the lines I am looking for. The ones I am looking for have `code: 0x57` not `code: 0x62` and they would be toward the top of the log.  At the Linux level you can use:
`cat home-assistant.log | grep 0x57 | grep 1A.2B.3C` (substituting your device ID as appropriate)

rstanley75, 2018-02-08:
Oops, sorry about that!  I'll try again:
```
2018-02-08 16:06:19 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x01, 'address': 32.91.D1, 'linkdata1': 0x10, 'linkdata2': 0x11, 'linkdata3': 0x44}
2018-02-08 16:06:20 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 32.91.D1, 'linkdata1': 0x10, 'linkdata2': 0x11, 'linkdata3': 0x44}
2018-02-08 16:06:25 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x01, 'address': 2C.6E.B8, 'linkdata1': 0x10, 'linkdata2': 0x11, 'linkdata3': 0x43}
2018-02-08 16:06:26 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xa2, 'group': 0x01, 'address': 2C.6E.B8, 'linkdata1': 0x10, 'linkdata2': 0x11, 'linkdata3': 0x43}
```

teharris1, 2018-02-09:
@rstanley75 So now I am kicking myself because I had the code originally written for all cat 0x10 devices to use the ALDB data for identification. Unfortunately was too inconsistent but for this device (as well as motion sensors at least) I think this is consistent. Probably because they are battery operated (just a theory). I can make this work easy.

rstanley75, 2018-02-12:
@teharris1, my door sensors are working now.  Thanks!

teharris1, 2018-02-25:
@rstanley75 @majortom64 @nugget @lanrongwen Great news, the HA pull request was accepted today. I am impressed by their through code inspection and feel really good about the latest code in HA. (And learned a lot that can be applied here.) The refactored code will be in the next release. Thanks for all the help.

closed 2018-02-25 by teharris1