nugget/python-insteonplm #43: Problems running refactored code.

majortom64, 2018-02-06:
A few problems:

During setup:

2018-02-06 13:15:26 ERROR (MainThread) [homeassistant.components.switch] Error while setting up platform insteon_plm
Traceback (most recent call last):
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/helpers/entity_component.py", line 189, in _async_setup_platform
    SLOW_SETUP_MAX_WAIT, loop=self.hass.loop)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 358, in wait_for
    return fut.result()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/futures.py", line 245, in result
    raise self._exception
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/coroutines.py", line 210, in coro
    res = func(*args, **kw)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/switch/insteon_plm.py", line 26, in async_setup_platform
    _LOGGER.info('Registered %s with switch platform.', device.id)
AttributeError: 'dict' object has no attribute 'id'

While running:

2018-02-06 13:18:18 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/python-insteonplm/insteonplm/devices/devicebase.py", line 164, in _wait_for_direct_ACK
    callback(msg)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/python-insteonplm/insteonplm/states/sensor.py", line 196, in _status_message_received
    self._status_message_received)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/python-insteonplm/insteonplm/messagecallback.py", line 59, in remove
    cb = self._dict[msg]
KeyError: <insteonplm.messages.standardReceive.StandardReceive object at 0x1086f71d0>

teharris1, 2018-02-07:
@majortom64 Thanks for testing.  Please check your HA install. It appears you have the correct file in 
`/components/insteon_plm.py` but not `/components/switch/insteon_plm.py`. I fixed the second issue which is in the python-insteonplm code so you will need to download that again.

majortom64, 2018-02-07:
Did a git pull in teharris1/python/insteon_plm and teharris1/home-assistant, and this is what I get;

2018-02-07 00:33:53 ERROR (MainThread) [homeassistant.components.fan] Error while setting up platform insteon_plm
Traceback (most recent call last):
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/helpers/entity_component.py", line 189, in _async_setup_platform
    SLOW_SETUP_MAX_WAIT, loop=self.hass.loop)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 358, in wait_for
    return fut.result()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/futures.py", line 245, in result
    raise self._exception
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/coroutines.py", line 210, in coro
    res = func(*args, **kw)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 37, in async_setup_platform
    state_list.append(InsteonPLMBinarySensor( hass, device, stateKey, newnames, SUPPORT_SET_SPEED))
NameError: name 'InsteonPLMBinarySensor' is not defined

teharris1, 2018-02-07:
Should be fixed now.

majortom64, 2018-02-07:
Did a git pull and copied the sensor and fan PLM components.

2018-02-07 08:49:42 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/python-insteonplm/insteonplm/devices/devicebase.py", line 164, in _wait_for_direct_ACK
    callback(msg)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/python-insteonplm/insteonplm/states/sensor.py", line 196, in _status_message_received
    self._status_message_received)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/python-insteonplm/insteonplm/messagecallback.py", line 59, in remove
    cb = self._dict[msg]
KeyError: <insteonplm.messages.standardReceive.StandardReceive object at 0x105d61320>

majortom64, 2018-02-07:
Just noticed that the FanLinc does not appear and I received this error:

[homeassistant.components.fan.insteon_plm] Received update calback from PLM for device 216a65 state fanOnLevel
2018-02-07 09:55:19 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 233, in async_update_ha_state
    attr)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 329, in _attr_setter
    value = getattr(self, name)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 122, in supported_features
    return self._supported_features
AttributeError: 'InsteonPLMFan' object has no attribute '_supported_features'

teharris1, 2018-02-07:
I have addressed both of these now. Thanks so much for testing. I don't have a FanLinc or an I/O link so it is tough for me to check them.

majortom64, 2018-02-07:
I did a new git pull from your insteon-plm repo. Now I get this:

2018-02-07 14:00:25 INFO (MainThread) [homeassistant.components.fan.insteon_plm] Received update calback from PLM for device 216a65 state fanOnLevel
2018-02-07 14:00:25 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 233, in async_update_ha_state
    attr)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/helpers/entity.py", line 329, in _attr_setter
    value = getattr(self, name)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 122, in supported_features
    return self._supported_features
AttributeError: 'InsteonPLMFan' object has no attribute '_supported_features'

The FanLinc Fan no longer shows up.

teharris1, 2018-02-07:
Woops. Sorry. My commit did not take originally for some reason. It is there now.

majortom64, 2018-02-08:
That seems to have fixed my fan issues.

teharris1, 2018-02-08:
Great. So you can confirm both the fanlinc and I/o Linc work?

majortom64, 2018-02-08:
The UI does display the state of the IO link sensor and the IO switch can be controlled by the HASS UI.

Sorry, I realized I did not check if it actually controlled the fan. When I use the HASS UI to turn on the fan, I get this:

2018-02-08 12:15:36 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=fan, service=turn_on, service_data=entity_id=['fan.216a65_2'], service_call_id=4370625480-70>
2018-02-08 12:15:36 ERROR (MainThread) [homeassistant.core] Error executing service <ServiceCall fan.turn_on: entity_id=['fan.216a65_2']>
Traceback (most recent call last):
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/core.py", line 1010, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/__init__.py", line 218, in async_handle_fan_service
    yield from getattr(fan, method['method'])(**params)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 100, in async_turn_on
    self.async_set_speed(speed)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 112, in async_set_speed
    self.state.set_level(fanSpeed)
AttributeError: 'str' object has no attribute 'set_level'
2018-02-08 12:15:46 INFO (MainThread) [homeassistant.core] Bus:Handling <Event service_executed[L]: service_call_id=4370625480-69>

teharris1, 2018-02-08:
Great that the IO Link works.

FanLinc issue should be fixed now.

majortom64, 2018-02-09:
Now I get this: 

2018-02-08 17:55:47 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=fan, service=turn_on, service_data=entity_id=['fan.216a65_2'], service_call_id=4370625536-21>
2018-02-08 17:55:47 ERROR (MainThread) [homeassistant.core] Error executing service <ServiceCall fan.turn_on: entity_id=['fan.216a65_2']>
Traceback (most recent call last):
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/core.py", line 1010, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/__init__.py", line 218, in async_handle_fan_service
    yield from getattr(fan, method['method'])(**params)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 100, in async_turn_on
    self.async_set_speed(speed)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 112, in async_set_speed
    self._state.set_level(fanSpeed)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/python-insteonplm/insteonplm/states/dimmable.py", line 194, in set_level
    speed = self._value_to_fan_speed(val)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/python-insteonplm/insteonplm/states/dimmable.py", line 237, in _value_to_fan_speed
    return SPEED_MEDIUM
NameError: name 'SPEED_MEDIUM' is not defined
2018-02-08 17:55:57 INFO (MainThread) [homeassistant.core] Bus:Handling <Event service_executed[L]: service_call_id=4370625536-20>

teharris1, 2018-02-11:
@majortom64 I have fixed it in my repo. Can you please try again? Thanks so much.

majortom64, 2018-02-12:
The ceiling fan was on and the HASS UI should that. When I went to turn it off, I got:

2018-02-11 15:52:32 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=homeassistant, service=turn_off, service_data=entity_id=fan.216a65_2, service_call_id=4370629632-21>
2018-02-11 15:52:32 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=fan, service=turn_off, service_data=entity_id=['fan.216a65_2'], service_call_id=4370629632-22>
2018-02-11 15:52:32 ERROR (MainThread) [homeassistant.core] Error executing service <ServiceCall fan.turn_off: entity_id=['fan.216a65_2']>
Traceback (most recent call last):
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/core.py", line 1010, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/__init__.py", line 218, in async_handle_fan_service
    yield from getattr(fan, method['method'])(**params)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 103, in async_turn_off
    self.async_set_speed(SPEED_OFF)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 109, in async_set_speed
    self._state.off()
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/python-insteonplm/insteonplm/states/dimmable.py", line 203, in off
    self._send_method(ExtendedSend(self._address, COMMAND_LIGHT_OFF_0X13_0x00, self._udata), self._off_message_received)
NameError: name 'COMMAND_LIGHT_OFF_0X13_0x00' is not defined
2018-02-11 15:52:42 INFO (MainThread) [homeassistant.core] Bus:Handling <Event service_executed[L]: service_call_id=4370629632-21>

The HASS UI showed it go off for a few seconds, then switched it back to being on. It did not turn off the fan.

I then turned the fan off (from an Insteon switch), but the UI does not update.

Next, if I try to turn the switch off while the fan is off (but the UI thinks it is on), I get this:

2018-02-11 15:55:10 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=homeassistant, service=turn_off, service_data=entity_id=fan.216a65_2, service_call_id=4370629632-23>
2018-02-11 15:55:11 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=fan, service=turn_off, service_data=entity_id=['fan.216a65_2'], service_call_id=4370629632-24>
2018-02-11 15:55:11 ERROR (MainThread) [homeassistant.core] Error executing service <ServiceCall fan.turn_off: entity_id=['fan.216a65_2']>
Traceback (most recent call last):
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/core.py", line 1010, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/__init__.py", line 218, in async_handle_fan_service
    yield from getattr(fan, method['method'])(**params)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 103, in async_turn_off
    self.async_set_speed(SPEED_OFF)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/insteon_plm.py", line 109, in async_set_speed
    self._state.off()
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/python-insteonplm/insteonplm/states/dimmable.py", line 203, in off
    self._send_method(ExtendedSend(self._address, COMMAND_LIGHT_OFF_0X13_0x00, self._udata), self._off_message_received)
NameError: name 'COMMAND_LIGHT_OFF_0X13_0x00' is not defined
2018-02-11 15:55:21 INFO (MainThread) [homeassistant.core] Bus:Handling <Event service_executed[L]: service_call_id=4370629632-23>

and the state stays the same (it shows on but the fan is actually off)).

@teharris1 I can give you access to a machine if you would like to try this directly :-)

teharris1, 2018-02-12:
OK. I made the update.

majortom64, 2018-02-19:
2018-02-18 12:31:40 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=homeassistant, service=turn_on, service_data=entity_id=fan.216a65_2, service_call_id=4370431224-45>
2018-02-18 12:31:40 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=fan, service=turn_on, service_data=entity_id=['fan.216a65_2'], service_call_id=4370431224-46>
2018-02-18 12:31:40 ERROR (MainThread) [homeassistant.core] Error executing service <ServiceCall fan.turn_on: entity_id=['fan.216a65_2']>
Traceback (most recent call last):
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/core.py", line 1010, in _event_to_service_call
    yield from service_handler.func(service_call)
  File "/Users/majortomlocal/Developer/InsteonPLM-teharris/lib/python3.6/site-packages/homeassistant/components/fan/__init__.py", line 218, in async_handle_fan_service
    yield from getattr(fan, method['method'])(**params)
TypeError: 'NoneType' object is not iterable
2018-02-18 12:31:41 INFO (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=fan.216a65_2, old_state=<state fan.216a65_2=off; speed=off, speed_list=['off', 'low', 'medium', 'high'], INSTEON Address=21.6A.65, Description=FanLinc Dual Band, Model=2475F, Category=01, Subcategory=2e, Product Key / Firmware=00, Group=02, friendly_name=Master Bedroom Ceiling Fan, supported_features=1, homebridge_name=Master Bedroom Ceiling Fan @ 2018-02-18T12:06:16.884967-08:00>, new_state=<state fan.216a65_2=on; speed=high, speed_list=['off', 'low', 'medium', 'high'], INSTEON Address=21.6A.65, Description=FanLinc Dual Band, Model=2475F, Category=01, Subcategory=2e, Product Key / Firmware=00, Group=02, friendly_name=Master Bedroom Ceiling Fan, supported_features=1, homebridge_name=Master Bedroom Ceiling Fan @ 2018-02-18T12:31:41.681129-08:00>>

teharris1, 2018-03-23:
@majortom64 Hope all is well. Have you tried the latest version of HA and insteonplm? I believe the issues above is addressed. Can you confirm?

majortom64, 2018-03-25:
@teharris1 From which repo should I install? I was traveling and just returned. I am happy to test today.

teharris1, 2018-03-25:
It is in the production version. So just go with the 0.65.6 version of HA that is out right now.

majortom64, 2018-04-07:
The fan stuff seems to be working now (0.66). Still having problems with the KeyPadLincs.

teharris1, 2018-04-07:
That is great to hear about the FanLinc. 

Just FYI, I was so interested in the KPL from your description that I went and bought one. I have it installed in my office. Fascinating device. I now see why managing the link tables is so important with this device. Because of that I am working on All-Linking now. I had a break through with that a week or so ago where reliability is up to about 75-80%. (Not sure it can every be higher than that unfortunately) I think I can get link management done in a couple of weeks. After that I will be working on the KPL. 

I am going to close this issue since #44 is dedicated to the KPL.

closed 2018-04-07 by teharris1