nugget/python-insteonplm #44: Problems with device control and state display with refactored code.

majortom64, 2018-02-07:
I have been trying to understand exactly what is happening, but from what I can tell there are problems with controlling KeyPadLincs using the HASS UI when they have links to other devices.

I have a newly installed KPL with no links other than to the PLM. From the UI I can control it with no problem. If I turn it on, manually, it updates the UI (only seems to update the switch position, not the light icon but that is the same for all switches).

The SwitchLincs work fine from the UI and update the UI when controlled locally. When controlled by a linked button on a KPL, the UI does not update despite there being a link to the PLM from the button. (This goes to the link management issue #38 I raised earlier.)

If I try to control any of my KPLs that have links to other devices, I can change their state one time, but then the UI restores the original state (meaning that UI changes back to the original state, leaving the device in the new state) and from that point further I cannot control the device from the UI until I change its state locally. Changing its state locally, seems to reset it for a single use from the UI.

teharris1, 2018-02-07:
This is not surprising unfortunately. The KPL is not a switch but that is the way it currently appears because it is a cat 0x01 (dimmable) or cat 0x02 (on/off) device. Which means it sends to on/off commands like a 0x01 or 0x02 device but status updates are different. To make this work a new device will need to be created.

The other issue you are seeing, if I understand you correctly, is that state changes do not cascade along linked devices. In other words, if switchA controls switchB and switchB controls switchC, changes to switchA do not effect switchC. The PLM is like any other device. So the PLM can change the KPL the changes to the KPL don't cascade to the devices controlled by the KPL. This is by design in the Insteon model.

Hope that helps.

majortom64, 2018-02-08:
I am going to spilt the issues I raised in my first post, as they are unrelated. This is issue is that some of my KeyPadLincs cannot be controlled from the HASS UI while some can.

Here are the symptoms:

I have a KPL that shows up in the HASS interface. If I press switch in the interface to change its state, it changes for about 5 seconds and then returns to its original state. If the displayed state and the actual state agree, it controls the device. If they do not, then nothing happens.

For example:

**HASS interface** | **KPL controlled load** | **Result of pressing button in HASS UI**
------------------- | ------------------------ | ----------------------------------------
Shows **Off** | Device is **off** | UI switches to **on** for a few seconds, then returns to **off**. Device turns **on**.
Shows **off** | Device is **on** | UI switches to **on** for a few seconds, then returns to **off**. Device stays **on** (in other words, no change to device).
Shows **on** | Device is **on** | UI switches to **off** for a few seconds, then returns to **on**. Device turns **off**.
Shows **on** | Device is **off** | UI switches to **off** for a few seconds, then returns to **on**. Device stays **off** (in other words, no change to device).

It does not matter how I change the state of the device to match the displayed state in the interface, as long as they match, the UI will change the device state.

At first I thought it had to do with Insteon links between the KPL and other devices, but I discovered that it works for some devices with many links while it fails for some. I next thought it might be based on if a button on the KPL was set in one of the non-toggle modes, but that did not affect anything either. My final thought was that it had to do with a link between the KPL and the IO link that enabled it to show its state, but also was not it. I have no idea what the difference between the working and non-working ones is.

Here is the console output of a working switch after changing its state in the HASS UI:

[FP Overhead Working.txt](https://github.com/nugget/python-insteonplm/files/1708623/FP.Overhead.Working.txt)

Here is the console output of a failing switch where the device is in the same state as shown in the UI (meaning that the pressing the UI switch changes the device state):

[GR not working same state.txt](https://github.com/nugget/python-insteonplm/files/1708631/GR.not.working.same.state.txt)

Here is the console output of a failing switch where the device is in the opposite state as shown in the UI (meaning that the pressing the UI switch does not change the device state):

[GR not working oposite state.txt](https://github.com/nugget/python-insteonplm/files/1708633/GR.not.working.oposite.state.txt)

majortom64, 2018-06-17:
Running the current version of HASS from your repo, and the version of Insteon PLM (also from your repo), and I am still having this problem with some of my Keypad Linc switch.

I cannot tell the difference between those that work and those that do not.

The symptoms listed above are still the same, but 

**HASS interface** | **KPL controlled load** | **Result of pressing button in HASS UI**
------------------- | ------------------------ | ----------------------------------------
Shows **Off** | Device is **off** | UI switches to **on** for a few seconds, then returns to **off**. Device turns **on**.
Shows **off** | Device is **on** | UI switches to **on** for a few seconds, then returns to **off**. Device stays **on** (in other words, no change to device).
Shows **on** | Device is **on** | UI switches to **off** for a few seconds, then returns to **on**. Device turns **off**.
Shows **on** | Device is **off** | UI switches to **off** for a few seconds, then returns to **on**. Device stays **off** (in other words, no change to device).

Controlling the device from the switch itself, resets the UI.

Not sure what else to check, but happy to help debug it any way you want (including giving you access to the machine). :-D

teharris1, 2018-06-17:
@majortom64 I am having the same issue with my KPL. Can you check if you see the following message information:

`Processing message {'code': 0x50, 'address': 1A.2B.3C 'target': 4D.5E.6F, 'flags': 0xa0, 'cmd1': 0x11, 'cmd2': 0xfc}`

Where `1A.2B.3C` is your KPL. The import pieces are `'flags': 0xa0` and `'cmd2': 0xfc`, where 'cmd1' can be `0x11` or `0x13`. If you see this then it is an indication that the KPL ALDB is returning a error that the search of the ALDB is taking too long. I am not sure why this is happening but I think it is an Insteon issue.

majortom64, 2018-06-23:
I do not see any "Processing message" messages.

teharris1, 2018-06-23:
Check that you have the following in the logging section of the config file 
`insteonplm: debug`

Otherwise the lack of a message would be a linking issue.

teharris1, 2018-06-25:
@majortom64 I just pushed a new release to this site that fixes the issue I described above. This release 0.11.1 fixes the issue for me. You can download the code from `nugget/python-insteonplm` or you can just do a `pip3 uninstall insteonplm` then `pip3 install insteonplm`. This will upgrade you to 0.11.1. I am pushing an update to HA for this too.

majortom64, 2018-06-25:
Will try it in the morning. Thanks for all your hard work.

teharris1, 2018-06-26:
I am changing this from an enhancement to a bug. 
@majortom64 any update from your side?

majortom64, 2018-06-27:
This is what is in my logger section:

logger:
    default: info
    logs:
        homeassistant.components.insteon_plm: debug
	insteonplm.protocol: debug
# I added this next line today.
        insteonplm: debug 

Here is the output when I hit the switch and get that result:

2018-06-27 11:07:39 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=homeassistant, service=turn_off, service_data=entity_id=light.289ef5, service_call_id=4518009096-53>
2018-06-27 11:07:39 INFO (MainThread) [homeassistant.core] Bus:Handling <Event call_service[L]: domain=light, service=turn_off, service_data=entity_id=['light.289ef5'], service_call_id=4518009096-54>
2018-06-27 11:07:39 DEBUG (MainThread) [insteonplm.devices] Starting Device._send_msg
2018-06-27 11:07:39 DEBUG (MainThread) [insteonplm.devices] Ending Device._send_msg
2018-06-27 11:07:39 INFO (MainThread) [homeassistant.core] Bus:Handling <Event service_executed[L]: service_call_id=4518009096-54>
2018-06-27 11:07:39 DEBUG (MainThread) [insteonplm.devices] Starting Device._process_send_queue
2018-06-27 11:07:39 DEBUG (MainThread) [insteonplm.devices] Lock is locked from yeild from
2018-06-27 11:07:39 DEBUG (MainThread) [insteonplm.plm] Starting: send_msg
2018-06-27 11:07:39 DEBUG (MainThread) [insteonplm.plm] Ending: send_msg
2018-06-27 11:07:39 DEBUG (MainThread) [insteonplm.devices] Ending Device._process_send_queue
2018-06-27 11:07:39 DEBUG (MainThread) [insteonplm.plm] Aquiring write lock
2018-06-27 11:07:39 DEBUG (MainThread) [insteonplm.plm] Writing message: {'code': 0x62, 'address': 28.9E.F5, 'flags': 0x00, 'cmd1': 0x13, 'cmd2': 0x00, 'acknak': 0xNone}
2018-06-27 11:07:39 DEBUG (MainThread) [insteonplm.plm] Waiting for ACK or NAK message
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] Starting: data_received
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] Received 20 bytes from PLM: b'0262289ef5001300060250289ef548ee55a01300'
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] Total buffer: b'0262289ef5001300060250289ef548ee55a01300'
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] Starting: _peel_messages_from_buffer
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] Finishing: _peel_messages_from_buffer
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] Messages in queue: 2
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x62, 'address': 28.9E.F5, 'flags': 0x00, 'cmd1': 0x13, 'cmd2': 0x00, 'acknak': 0x06}
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.devices] Starting Device.receive_message
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.devices] Got Message ACK
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.devices] Look for direct ACK
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.devices] Ending Device.receive_message
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x50, 'address': 28.9E.F5, 'target': 48.EE.55, 'flags': 0xa0, 'cmd1': 0x13, 'cmd2': 0x00}
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.devices] Starting Device.receive_message
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.devices] Ending Device.receive_message
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] Last item in self._recv_queue reached.
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] Finishing: data_received
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] ACK or NAK received
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.plm] {'code': 0x62, 'address': 28.9E.F5, 'flags': 0x00, 'cmd1': 0x13, 'cmd2': 0x00, 'acknak': 0x06}
2018-06-27 11:07:40 DEBUG (MainThread) [insteonplm.devices] Starting Device._wait_for_direct_ACK
2018-06-27 11:07:40 INFO (MainThread) [homeassistant.core] Bus:Handling <Event service_executed[L]: service_call_id=4518009096-53>
2018-06-27 11:07:43 DEBUG (MainThread) [insteonplm.devices] No direct ACK messages received.
2018-06-27 11:07:43 DEBUG (MainThread) [insteonplm.devices] Releasing lock
2018-06-27 11:07:43 DEBUG (MainThread) [insteonplm.devices] Ending Device._wait_for_direct_ACK
2018-06-27 11:07:43 DEBUG (MainThread) [insteonplm.plm] No new messages received.

teharris1, 2018-06-27:
This feels like a signal or a link issue to me. Are you sure your links are good? The device responds to the message being sent but you did not get a direct ACK which means the device did not acknowledge the message. There are a few reasons that you would not receive an ACK. One is that the PLM does not have that device & group in the ALDB or the signal got lost. I cannot think of any other scenario.

teharris1, 2018-06-27:
Sorry, there is an ACK (sort of). It is the line that reads: `Processing message {'code': 0x50, 'address': 28.9E.F5, 'target': 48.EE.55, 'flags': 0xa0, 'cmd1': 0x13, 'cmd2': 0x00}` Let me look into this.