nugget/python-insteonplm #52: Leak sensor reporting 'wet' when it is dry

teharris1, 2018-03-20:
Leak sensor indicates the device is wet when it is dry Issue is the daily heartbeat is being seen as an 'on' command. Device needs to be rewritten to support groups 0x01 Dry  0x02 Wet and 0x04 Heartbeat.

teharris1, 2018-03-23:
Test code available: 

**Module insteonplm:**
[insteonplm/devices/securityHealthSafety.py](https://raw.githubusercontent.com/teharris1/python-insteonplm/leaksensor/insteonplm/devices/securityHealthSafety.py )
[insteonplm/states/sensor.py](https://raw.githubusercontent.com/teharris1/python-insteonplm/leaksensor/insteonplm/states/sensor.py)

**Module homeassistant:**
[homeassistant/components/insteon_plm.py](https://raw.githubusercontent.com/teharris1/home-assistant/leaksensor/homeassistant/components/insteon_plm.py)
[homeassistant/components/binary_sensor/insteon_plm.py](https://raw.githubusercontent.com/teharris1/home-assistant/leaksensor/homeassistant/components/binary_sensor/insteon_plm.py)
Please replace the current files with these files and restart HA. I have tested the code as best I can but I don't own one of these leak sensors so I cannot test if it works.

teharris1, 2018-03-27:
This has been reasonably tested.

@nugget I put a version in the version database. Can we push to pypi? Alternatively, can you give me admin access to pypi for this module?

nugget, 2018-03-27:
v0.8.3 pushed to PyPI.  Would be glad to add you to the project there, are you https://pypi.org/user/teharris/ ?

teharris1, 2018-03-30:
Updated in Home Assistant too.

closed 2018-03-30 by teharris1

teharris1, 2018-04-03:
@nugget Yes I am teharris on pypi.