nugget/python-insteonplm #65: Create interactive command line tool

teharris1, 2018-04-07:
Create an interactive command line tool that allows users to complete some basic tasks:
* Connect to PLM/Hub
* List devices
* Put the IM in All-Link mode as controller or responder
* Read a device All-Link database
* Write to a device All-Link database
* Test basic device commands (On/Off)

majortom64, 2018-04-08:
Two suggestions:
1) As part of the list devices (or maybe as a separate item): device info (device type, firmware version, etc.).
Send other Insteon commands (change/query settings, etc.).

2) Enable this via a server mode on the existing code, so that it can be done even while HA is active. Acknowledging that is much more work and might not be possible at all. :-)

teharris1, 2018-04-13:
The command line tool is in pretty good shape now. It performs the following functions:
 -  **connect** - Connect to the PLM (First thing you should do.  Type 'help connect' to get the options)
 -  **exit**: Exit the tool
 -  **help**: Display the help information
 -  **list_devices**: List all loaded devices
 -  **load_aldb**: Load the device All-Link Database (this may take up to 10 min per device)
 -  **on_off_test**: Simple On/Off test to confirm things are working
 -  **print_aldb**: Print a devices' All-Link Database (must use load_aldb first)
 -  **running_tasks**: List tasks running in the background (not real useful)
 -  **set_device**: Set the PLM device name (i.e. /dev/ttyUSB0)
 -  **set_log_level**: Set the logging level (i for information v for verbose)
 -  **set_workdir**: Set the directory to load saved device info from
 -  **start_all_linking**: Set the PLM to start All-Linking
 -  **write_aldb**: Write a record to a device All-Link database (BE CAREFUL!!!)
@majortom64 @nugget @yombo @lanrongwen  love to hear feedback!

majortom64, 2018-04-16:
From which repo should I install this?

teharris1, 2018-04-16:
It is in the main repo.

closed 2018-06-05 by teharris1