nugget/python-insteonplm #79: Windows support

takuy, 2018-06-09:
Just wondering - what is limiting the Window support? I'm currently using a similar project from years ago to control a few lights. (https://sourceforge.net/projects/pylights/)

Seems like they use the same Serial library, though I'd rather move to a maintained project, though the lack of Windows support is stopping that.

teharris1, 2018-06-10:
The issue is the serial.aio library that wraps pyserial in an asyncio.transport. Per the notes in the code:

> Windows event loops cannot wait for serial ports with the current implementation.

But this code has not been touched in almost a year. I am considering creating my asyncio.transport which may allow it to work. Not sure though. 

In the meantime, have you considered running [Docker on Windows](https://docs.docker.com/docker-for-windows/install/)? Docker is pretty lightweight vs running a virtual machine via hyper-v or other hypervisors.

takuy, 2018-09-07:
Great!

teharris1, 2018-09-07:
Per wz2b:

>This change caused things to break for me: 
```
transport, protocol = yield from create_serial_connection( 
                    self._loop, lambda: self.protocol,
                     self.device.upper(), baudrate=19200) 
```
>with this exception: 
serial.serialutil.SerialException: [Errno 2] could not open port /DEV/TTYUSB0: [Errno 2] No such file or directory: '/DEV/TTYUSB0' 
A previous version did some test about os.name == 'nt'  which only did the upper() if you are windows.   I think this needs to test whether or not it needs to do that conversion again.

This is a bug and needs to be fixed. it should go back to the `os.name == 'nt'` test you are referring to. Thanks for pointing that out.