nugget/python-insteonplm #95: Add support for 2441V thermostat adapter

wz2b, 2018-09-02:
I would like support for a Venstar thermostat using the 2441V adapter.  These devices are fairly well documented, and I would be happy to write and contribute the component but I'm not exactly sure how to hook into HASS.  If someone wants to get me started - especially how unsolicited ExtendedReceive messages get routed to the component then interpreted - then I would try and test it myself.

wz2b, 2018-09-02:
Delete.

wz2b, 2018-09-02:
I am trying to test the 2441V support but the device is not being detected.  Actually, it sort of is, but what is this?

2018-09-02 11:44:37 DEBUG (MainThread) [insteonplm.plm] Processing message {'code': 0x57, 'controlFlags': 0xe2, 'group': 0x00, 'address': 11.B3.17, '
linkdata1': 0x01, 'linkdata2': 0x00, 'linkdata3': 0x00}
2018-09-02 11:44:37 DEBUG (MainThread) [insteonplm.plm] Last item in self._recv_queue reached.
2018-09-02 11:44:37 DEBUG (MainThread) [insteonplm.plm] Finishing: _peel_messages_from_buffer
2018-09-02 11:44:37 DEBUG (MainThread) [insteonplm.plm] Found all link record for device 11b317
2018-09-02 11:44:37 DEBUG (MainThread) [insteonplm.plm] ALDB Data: address 11b317 data1: 01 data1: 00 data3: 00
2018-09-02 11:44:38 DEBUG (MainThread) [insteonplm.states] Registering callbacks for DimmableSwitch device 11.B3.17

That's the thermostat adapter (11.B3.17) but why does it think it's a DimmableSwitch?

I might just not have homeassistant configured correctly.  I just have
  climate:
by itself

teharris1, 2018-09-02:
Yes, None is an appropriate value. I am not sure why it was set to that. Please (please, please, please) go ahead and create the 2441v device. I would love to see how someone else can work with my code and create one of the more complex devices.  As for the HA side, here is a link to my personal repo where climate has been created.:
https://github.com/teharris1/home-assistant/tree/thermostat
The files you need are:
homeassistant/insteon/__init__.py
homeassistant/climate/insteon.py

wz2b, 2018-09-02:
Do you need to declare   climate: in the configuration file or not at all -
just the ALDB is all you need for it to find it and create a climate.*
device?

>

teharris1, 2018-09-02:
I assume you mean in the HA config. If so, then no. You only need the device in the IPDB associated to a python object type and you need the two HA files referenced above.

teharris1, 2018-09-02:
If you create your own device in `insteonplm` then you need to add that device to the `components/insteon/__init__.py` file. Here is a link to the section to add it to:
Import it here:
https://github.com/teharris1/home-assistant/blob/thermostat/homeassistant/components/insteon/__init__.py#L369
And link it to the `climate` platform here:
https://github.com/teharris1/home-assistant/blob/thermostat/homeassistant/components/insteon/__init__.py#L398

wz2b, 2018-09-02:
I will try to work it out.  I am running hassbian and I thought I had been
keeping it up to date but it's still using insteon_plm not insteon ...

wz2b, 2018-09-03:
My 2441V model doesn't have humidity, but the 2441TH one does.  I tried removing the humidity property but that caused it to fail to load because the HA InsteonThermostatDevice model doesn't have it.  What is the right way to handle this?  Just leave it in place, but it never gets an update?  There are some flags that are supposed to deal with this, SUPPORTED_FEATURE* but near as I can tell they only deal with the ability to set values, not retrieve them.

wz2b, 2018-09-03:
For reference: my specific thermostat reports itself as Product ID 0x05:0x03:0x95.  For some reason the 0x6E status message is reporting in degrees rather than half-degrees.  Not sure if that's just because it's old or what.  I can't find any documentation to support it reporting in degrees but cmd2 is definitely 0x49 which is decimal 73 (that's correct).  Not sure what's going on here yet.

It's strange to me to have a conversation in ticket comments but OK.  When I'm working on this I"m also on the HA discord #dev

wz2b, 2018-09-14:
Is it possible that the existing thermostat code doesn't work?  The current implementation says it's for a 2441TH but it doesn't match the cmd1=0x2E message format described in the 2441TH developer's guide, so I'm wondering where this came from.  It looks like it entered this codebase 18 days ago.  I'm trying to decide whether I should fix it to match the developer notes; or leave it alone because somebody else knows it to work.

craigjmidwinter, 2019-02-01:
I've got a 2441TH, I might try hacking around on the HASS side for the climate this weekend and see what I can get working from your branch @teharris1

teharris1, 2019-02-02:
@craigjmidwinter Have a look at this branch:
[https://github.com/teharris1/home-assistant/tree/thermostat](https://github.com/teharris1/home-assistant/tree/thermostat)
It works (roughly) with the current version of insteonplm and HA. But not great.

craigjmidwinter, 2019-02-06:
@teharris1 I poked around with that branch in my local install over the weekend, and it seems like it's pretty close! My thermostat is set to Celsius, and in order to be able to set the heatpoint and cool point, I needed to change the hardcoded units in the component, but after doing that, I had success setting the state at least. Getting the state from the component is another story though, the main temperature seems to always be in Farenheit, and the values that I get back for the heat point doesn't correlate to anything, from what I can tell. Didn't get to play with it too much as I've got a 3 week old that is commanding the bulk of my attention, but I might try to dig a bit deeper this weekend again if I have any spare time.

Since this discussion is trending to be more about the HASS component dev, perhaps this issue isn't the best place to discuss it, should we create a new issue on your HASS fork?

teharris1, 2023-02-10:
Closing this issue as the `insteonplm` module is no longer maintained. Please use the `pyinsteon` module instead.

closed 2023-02-10 by teharris1