nugget/python-insteonplm #99: Retrieving the ALDB (API question)

wz2b, 2018-09-04:
I'm attempting to write my own little tool for doing some maintenance of the ALDB, starting with the PLM.  It's not going super, and I'm looking for some help.  Based on that help, I'll try to contribute to the documentation and make it easier to understand.

What I'm doing involves Qt so I create the asyncio loop using quamash.  I don't think that affects my question, but I'm including this for completeness, as I'm a Java/Scala/C/C++ developer hacking my way through Python-land:

```app = QApplication(sys.argv)
loop = QEventLoop(app)
print(loop)
asyncio.set_event_loop(loop)```

My attempt to dump the ALDB starts with just printing to the console:

```class Controller:
  def __init__(self, loop):
    self.loop=loop
    print("Controller initialized")

  def connect(self):
    self.loop.create_task(self._connect())

  def load(self):
    self.loop.create_task(self._load())

  def dump(self):
     self.loop.create_task(self._dump())

  @asyncio.coroutine
  def _connect(self):
    self.conn = yield from insteonplm.Connection.create();
    print("Connected")

  @asyncio.coroutine
  def _load(self):
    print("Loading...")
    self.plm = self.conn.protocol
    self.plm.aldb.clear()
    yield from asyncio.sleep(1, loop=self.loop)
    self.plm.read_aldb()

    yield from asyncio.sleep(5, loop=self.loop)

    while self.plm.aldb.status == ALDBStatus.LOADING:
      print(str(self.plm.aldb.status))
      yield from asyncio.sleep(1, loop=self.loop)

    print(str(self.plm.aldb.status))
    print("Done.");

  @asyncio.coroutine
  def _dump(self):
    for i in self.plm.aldb:
      record=self.plm.aldb[i]
      print(record)```

The problem is I get very few records - sometimes zero, sometimes two.  There should be something on the order of 80 links.  This is all with 0.13.1.

Am I doing something obviously wrong here?

wz2b, 2018-09-05:
I figured it out.  It was polling the aldb by itself as soon as I connected.  I really don't want it to do that, at least not unless it can somehow give me a callback telling me when it's done ....

teharris1, 2018-09-30:
I have started working on some changes to the ALDB which will allow for direct writes to the modem.  The ALDB of the modem works different than the ALDB of other devices. This is, in my opinion, a problem with Insteon that it does not have a meaningful API that allows it to keep DB changes in sync and it does not have a real method to ensure you don't break things when you write to the ALDB. 

Bottom line, writing to he ALDB is a high risk situation and requires a lot of testing. I am also good to remove the ALDB load function on startup and make that a function of the front end. This only means that the consumer of the application will need to know how to load devices into the `Modem.devices` list. 

The current method is kind of an idiot proof method of ensuring any device in the `devices` list is in the modem's database. If you add a device to the `devices` list that is not in the modem's database then you will get inconsistent results. This is a limitation of Insteon, not the library.

wz2b, 2018-10-01:
Just to explain what I'm trying to do here ...

I really want the ability to capture the entire ALDB (from the modem) to a readable file (JSON, XML, CSV, whatever) ... be able to edit it to fix/remove bogus stuff ... then replace the whole thing.  To do this, you'd have to write any record from the NEW list not in the IM, then delete any record in the IM that's not in the NEW list.  But alternatively you could just wipe out the entire ALDB on the IM and start over.  That would take longer, I think, but maybe not if you can accomplish this by just setting the high watermark on the first record - I'm not sure.  Whatever's safer.

The field devices are trickier because there are direct links (switches to loads for example) that don't involve the IM.

I really want to have the ability to write some kind of lint checker for these links.  I want for example to be able to find half-links; find things in the IMDB that aren't in the field devices; and detect things like a field device has a direct link to an appliance module that's no longer on the network (which slows/chokes things like crazy).  I'd like to never have to run HouseLinc again (how I currently do all this stuff).

So I look forward to seeing where you're going with this.

teharris1, 2020-01-20:
@wz2b Hope you are well. Not sure if you have been following pyinsteon but I have made a lot of progress. Here is the underlying architecture. All layers communicate via a pub/sub model. I like the new architecture much better. I appreciate the feedback regarding the complexity of the insteonplm object model. 
[image]

wz2b, 2020-01-25:
Hi Tom!

> @wz2b <https://github.com/wz2b> Hope you are well. Not sure if you have
been following pyinsteon but I have made a lot of progress. Here is the
underlying architecture. All layers communicate via a pub/sub model. I like
the new architecture much better. I appreciate the feedback regarding the
complexity of the insteonplm object model.

So it's funny, you sent this e-mail while I was in Las Vegas, so I marked
it to respond after I returned and for some odd reason it ended up in the
wrong place and never notified me.  This morning I woke up and it popped
into my head that I hadn't gotten back to you.  Strange.

I haven't been paying attention.  I have been off in the world of golang,
and my organization got a big contract for the government (having to do
with large vehicle electrification) that has occupied me.  But I am still
interested.  I just need to get back into the groove.  The second odd
coincidence here is that I had a keypadlinc fail on me.  I replaced it, but
now I'm trying to figure out the least painful way to re-link everything,
since it has a lot of buttons that have to be linked individually.  So what
you are writing about here is completely on topic.  The main thing I'm
really looking for, still, is

- Capture the ALDB of the entire network (every device) to a file
- Be able to edit it by hand etc
- Be able to tell some piece of software "ok, now go put it all back / full
synchronization / delete records that shouldn't be there" .. being able to
sync just one device is good too but if you're linking usually there's at
least 2 things needing updating

So in my case, at least, a fancy GUI to do all this isn't even necessary,
as long as there's a file I can just edit somewhere.  That file can be
pretty much anything, though - including python.

So that's a long-winded way of me saying I'll go download / update
everything today and try to catch myself back up to where you are, then
figure out how I can contribute, either in software, thought, or at least
just helping test.

Thanks very much for reaching out.  Sorry I disappeared for so long.

-C