nugget/python-insteonplm #111: Possible bug in with Keypad 2334_222 6 button map

wonderslug, 2018-10-10:
Looks like the current mapping for the 6 button mode on the 2334_222 is

`{3: 'C', 4: 'D', 5: 'E', 6: 'F', 7: 'G'}`

But the naming convention is like in the owners manual PDF

[image]

I am wondering if it should be the following on the button mapping for 6?

`{3: 'A', 4: 'B', 5: 'C', 6: 'D'}`

It also showing a group id 7 map, does that do anything on a 2334_222 in 6 mode?

wonderslug, 2018-10-11:
Sure I was just wondering if the group id of 7 on the 2334_222 when it was in a 6 button configuration did something specific since it was listed in the dict of button labels.

teharris1, 2018-10-11:
Closing due to PR being merged. Will be released in 0.15.0.

closed 2018-10-11 by teharris1