nugget/python-insteonplm #114: Add support for PLM all-link groups

mstovenour, 2018-11-04:
All-link groups are sometimes called Insteon scenes.  Insteon calls these all-link groups and responders react to all-link group broadcasts.  Most all Insteon devices can be all-link group controllers and it is common to link Insteon devices to each other using this capability.  Implementing scene control for all types of controllers is a large task since it requires knowledge of the ALDB entries from all devices.  It looks like a lot of work is underway to implement this capability (#45), but I suspect it will take quite a while because there are a few layers to that onion.

This enhancement request is specific to just PLM controller groups.  If a PLM is linked into a fully meshed responder/controller all-link group, then the PLM can update all the responders in the scene (all-link group).  Likewise, if any other controller updates the scene, the PLM will be notified.  Effectively the PLM can be modeled as a 254-button controller from HASS.  insteonplm will need specific changes since interacting with the all-link controller groups requires different PLM messages than interacting with a true multi-button controller.

Prior to getting full scene capabilities; an interim step could be to expose the PLM controller groups to HASS as switch devices and then hide all the associated devices in the UI.  HASS would track the scene state as the device state.  From a coding perspective adding these PLM controller groups as switches is all that is needed since hiding devices in the UI is already supported.  Once full scene updates can be tracked, the underlying devices would not need to be hidden.  My initial idea is to modify the PLM ALDB auto-discovery so that only responder entries are discovered as bonafide devices.  Any controller entries would be added to a generic switch device as they are discovered.  E.g. if there are 5 controller all-link groups (0x10-0x14) a switch device would be added for each one using the PLM ID (e.g. switch.1fdc78_10, switch.1fdc78_11, …, switch.1fdc78_14)

I really see this enhancement as orthogonal to adding support for ALDB and full tracking of device scenes.  HASS still should ultimately be enhanced to support PLM controller groups.

I would be happy to provide as much assistance coding this capability as desired.  I plan to start prototyping now but opened this request so that we could discuss how likely this change is to be incorporated if I create it.  If this enhancement seems like a good idea, I would also like to discuss implementation details to make sure the code changes fit into the overall architecture of the project.

I implemented the i2cs support in Misterhouse a few years back and got deep into the Insteon protocol back then.  That knowledge is a bit rusty but I’m quickly coming back up to speed.

teharris1, 2018-11-05:
This would be a great addition to the library. Yes. How can I help?

mstovenour, 2018-11-05:
I suppose for now you've already helped by not shooting down my plan to add the plm scenes as a multi-button device.

teharris1, 2018-11-05:
:) happy to have helped.

One thing to consider with home assistant, is the new device concept. For this I would see the PLM as a device. Note that there is a difference between an entity and a device. In HA today all you see are entities. But there is a concept to link them to a device.

teharris1, 2018-11-08:
Did Misterhouse did you ever implement the Hub 2245? This is the v2 hub (aka 2014 Hub) that does not have a PLM. This `insteonplm` module does work with it for all tasks except database management. It can put the Hub in link mode but it cannot perform database managment via command 0x6f (All-Link Database Management). The command is constantly NAKing. Any insight?

mstovenour, 2018-11-17:
Someone added Hub support but I don't recall the issue you are describing.  From what I recall it was a very minor change so I suspect it was the hub version with the integrated PLM.  Also, none of the documentation I have is new enough to cover that version of the hub.

teharris1, 2019-07-30:
@mstovenour you can now trigger All-Link groups with this library. It was added to the `Modem` class with the methods `trigger_group_on` and `trigger_group_off`.

teharris1, 2019-07-30:
Just saw that you are a Home Assistant user. Here is the link to documentation in HA.
https://www.home-assistant.io/components/insteon/#insteon-scenes