nugget/python-insteonplm #211: cannot easily delete aldb record from PLM

rpdrewes, 2019-12-22:
The del_aldb command works to delete records from non-PLM devices, by setting inuse flag to 0, but it seems to have no effect on the PLM. The records still stay inuse 1 when deleted.

I have seen references in other Issues that the aldb on the PLM is different than the aldbs on the non-PLM devices, but I'm at a loss as to how to delete records from the PLM without using button-unlinking or a full factory reset of the PLM (followed by relinking the desired devices).

Am I missing something?
Thanks

teharris1, 2020-01-20:
The method for doing his is to use the PLM `manage_aldb_record` method. There is no way to directly manage an ALDB record in the modem.  What this method requires is to set the `control_code` which determines the action to take on a link (not a record directly). Have a look at the doc string at the top of the method : https://github.com/nugget/python-insteonplm/blob/7a8f77455572231474b3277781fa8272e0117e70/insteonplm/plm.py#L261

teharris1, 2020-02-21:
@rpdrewes hopefully this answers your question. Let me know if it does not.